taligentx/dscKeybusInterface #2: PC1616/PC1832/PC1864 series support - data logs needed

taligentx, 2018-05-03:
Feel free to post logs as captured by the KeybusReader example for these panels in this issue to help decode any differences to the PC1555MX panel (multiple partitions, for one).

MagnusPer, 2018-05-04:
Great Work!! Have just downloaded and tested the KeyBusReader sketch on my NodeMCU, my goal is to integrate to [Homey](https://www.athom.com/en/).

**Alarm system - PowerSerie - PC1515**
Atttached log for Zone B 0x2D command
276.76: 00101101 0 10000000 00000011 10000001 11000111 00000001 11111001 [0x2D] Unrecognized command [CRC OK]
[log_PowerSerie-832_PC1515.txt](https://github.com/taligentx/dscKeybusInterface/files/1974723/log_PowerSerie-832_PC1515.txt)

Example of decoding Zone B - https://github.com/stagf15/DSC_Panel/blob/master/DSC.cpp

taligentx, 2018-05-06:
@MagnusPer - thanks for the logs, I've updated the library to support zones 9-32 but requires confirmation that this is functional.  If you or another is willing, it would be good to see logs with all zones closed, system armed, a zone 9-32 opened, trip the alarm, disarm the alarm, then close the zone.

MagnusPer, 2018-05-06:
@taligentx - I will tripp an alarm for zone 9 when my familiy not is in the house:-)  I will try to take the log during the next comming days

MagnusPer, 2018-05-09:
@taligentx 
**Alarm system - PowerSerie 832 - PC5015 - Panel LCD5500**
Zone 9 tripped - Alarm

[Log_PowerSerie832_PC5015-ALARM_zone9_KeybusReader.txt](https://github.com/taligentx/dscKeybusInterface/files/1988971/Log_PowerSerie832_PC5015-ALARM_zone9_KeybusReader.txt)
[Log_PowerSerie832_PC5015-ALARM_zone9_Status.txt](https://github.com/taligentx/dscKeybusInterface/files/1988972/Log_PowerSerie832_PC5015-ALARM_zone9_Status.txt)

MagnusPer, 2018-05-09:
@taligentx 
**Alarm system - PowerSerie 832 - PC5015 - Panel LCD5500**
Zone 10 tripped - Fire Alarm

[Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_KeybusReader.txt](https://github.com/taligentx/dscKeybusInterface/files/1988979/Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_KeybusReader.txt)
[Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_Status.txt](https://github.com/taligentx/dscKeybusInterface/files/1988981/Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_Status.txt)

taligentx, 2018-05-09:
Updated for zone 9-16 alarm on a zone expander and added fire status.

@MagnusPer Can you open zone 9 and then trip fire zone 10 (so they are both open at the same time)?  Also if you're up for it, it'd be very useful to see zone 16 configured on the zone expander, open/closed and in alarm/restored.  I can see how zones 9-15 are indicated, but zone 16 is unclear.  This should allow using the Arduino/esp8266 as a virtual zone expander for zones 9-16.

sj-louw, 2018-05-23:
@taligentx 

I have a DSC Power Series 1864 panel with multiple partitions, zone expanders, etc in use. Im willing to wire up my Arduino to the panel with the KeybusReader sketch, to capture some logs.

Is there any specific events that you need to be captured, that I can attach, to try make the 18xx panel support as feature complete as possible?

Thanks

taligentx, 2018-05-23:
Hi @sj-louw,

Thanks for your offer!  I've updated the list of needed data and removed the request for multiple partitions and zone expander data - looking at logs isn't going to be enough to decode these.  If all goes well I'll have a PC1864 panel for testing by next week.

It'd be great if you could load the Status sketch and log the output for partition 1:
* Arm partition 1 in stay mode, then disarm
* Arm partition 1 in away mode, open a zone in partition 1 and let the alarm sound, then disarm

If you're up for changing the zone expander configuration, zones 33-64 are still unknown, having data from KeybusReader for at least 1 zone in each zone group (33-40, 41-48, 49-56, 57-64) would be a good place to start decoding these.

sj-louw, 2018-05-28:
@taligentx 

Had some trouble getting the circuit right. I am reading from panel, but still busy with testing to verify my circuit... Panel went in alarm after seemingly working for some hours over the weekend. While in alarm it intermittently lost comms to main keypad, until I disconnected the Arduino circuit. Now I have, from what I understand a "General Supervisory" trouble code :) Nothing else works. Will try to get it sorted and then post the logs.

taligentx, 2018-05-31:
@sj-louw It does seem like there is some kind of wiring issue - there is quite a bit of invalid data in the log, for example:

```
  155.29: 00100111 0 00010000 [0x27] [CRC Error]
  155.40: 01100111 0 [0x67] Unrecognized command [No CRC or CRC Error]
  155.41: 01100011 1 [0x63] [CRC Error]
  155.47: 00110111 [0x37] Unrecognized command [No CRC or CRC Error]
  155.49: 00000001 1 [0x01] Unrecognized command [CRC OK]
  155.57: 00110000 0 00011000 11100000 00011000 [0x30] Unrecognized command [No CRC or CRC Error]
  155.64: 00000000 0 11100001 00000000 11000000 00110001 11000000 [0x00] Unrecognized command [No CRC or CRC Error]
  155.73: 11000111 1 [0xC7] Unrecognized command [No CRC or CRC Error]
```

Check the wiring and if you're still seeing logs with invalid data like the above please open a new issue with a description of the Arduino model, pins used, and the exact wiring to help debug it.   Thanks!

In other news, I've picked up a PC1864 panel but it seems there's either a problem with the panel or the older PC series keypads are not fully compatible (the installer code is accepted to reach the *8 programming menu but the panel errors out entering any programming section).  I'll keep an eye out for reasonably priced PC5500/RFK5500/RFK5564 keypads, though looking at the eBay sales history it may be some time.

For now, I'll be relying on contributed data logs to continue developing the panel decoding so all additional info is welcome!

sj-louw, 2018-05-31:
OK, thanks. As per our conversation in the other threads, I will post more findings as we progress.

On a side note, can you not try making your own PC-LINK cable and program the panel via the DLS software? This way you might get away with not buying a keypad :) I have made my own PC-LINK cable a while back, and it works nicely.

taligentx, 2018-05-31:
@sj-louw I actually did give DLS 5 a try with an FTDI USB/serial board but wasn't able to get a connection, even after going into `*6 + 5` to enable the DLS window.  Can you provide details on your process to connect?  Hopefully it doesn't require the installer `*8 + 499` command to initiate the connection.

jvitkauskas, 2018-06-02:
PC1832: held fire button, disarmed with code 1234

```
 315.28: 01110111 1 11111111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Fire alarm
  315.30: 00011100 0 [0x1C] Verify keypad Fire/Auxiliary/Panic
  315.30: 01110111 1 [Keypad] Fire alarm
  315.33: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
  315.38: 10111011 0 00100000 00000000 11011011 [0xBB] Bell: on
  315.45: 10100101 0 00011000 00011000 00100000 00011000 01001110 11111111 01011010 [0xA5] 06/01/2018 00:06 | Keypad Fire alarm
  315.49: 10000111 0 00000001 00000011 10001011 [0x87] Panel output:Unrecognized command: Add to 0x87 | PGM1 on | PGM2 on
  315.58: 11100110 0 00011010 01000000 00000001 00000000 10011001 00000000 00000000 00000000 11011010 [0xE6] Unknown command
  315.61: 01110101 0 10000000 11110101 [0x75] Partition 1 | Beep pattern: solid tone
  315.67: 11001110 0 00100000 00011000 01001110 00000000 00000000 01010100 [0xCE] Unrecognized command [CRC OK]
  315.76: 00000101 0 00000000 00010111 01010000 11000111 01010000 11000111 01010000 11000111 [0x05] Partition 1 | Lights: none | Unrecognized command: 0x17 | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
  315.90: 11100110 0 00011000 00000001 00010000 00000000 00000000 00000000 00000000 00001111 [0xE6] Unknown command
  315.95: 10000111 0 00000001 11110011 01111011 [0x87] Panel output:Unrecognized command: Add to 0x87 | PGM1 on | PGM2 on
  316.04: 11101011 0 00000000 00011000 00011000 00100000 00011000 00000000 01001110 11111111 10100000 [0xEB] Unrecognized command [CRC OK]
  316.19: 10100101 0 00011000 00011000 00100000 00011000 01010010 11111111 01011110 [0xA5] 06/01/2018 00:06 | Keypad Fire alarm restored
  316.26: 11001110 0 00100000 00011000 01010010 00000000 00000000 01011000 [0xCE] Unrecognized command [CRC OK]
  316.43: 11101011 0 00000000 00011000 00011000 00100000 00011000 00000000 01010010 11111111 10100100 [0xEB] Unrecognized command [CRC OK]
  316.84: 00000101 0 11010100 00010001 01010000 11000111 01010000 11000111 01010000 11000111 [0x05] Partition 1 | Lights: Memory Trouble Fire Backlight | Partition in alarm | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
  316.88: 10000111 0 00000001 00000011 10001011 [0x87] Panel output:Unrecognized command: Add to 0x87 | PGM1 on | PGM2 on
  317.67: 11111111 1 00000101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] 1
  317.88: 10000111 0 00000001 11110011 01111011 [0x87] Panel output:Unrecognized command: Add to 0x87 | PGM1 on | PGM2 on
  318.04: 11111111 1 00001010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] 2
  318.44: 11111111 1 00001111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] 3
  318.95: 11111111 1 00010001 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] 4
  319.01: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
  319.05: 10000111 0 00000001 00000011 10001011 [0x87] Panel output:Unrecognized command: Add to 0x87 | PGM1 on | PGM2 on
  319.23: 10000111 0 00000001 00000000 10001000 [0x87] Panel output:Unrecognized command: Add to 0x87 | PGM1 off | PGM2 off
  319.27: 10111011 0 00000000 00000000 10111011 [0xBB] Bell: off
  319.36: 11100110 0 00011010 01000000 00000000 00000000 00011001 00000000 00000000 00000000 01011001 [0xE6] Unknown command
  319.39: 01110101 0 00000000 01110101 [0x75] Partition 1 | Beep pattern: off
  319.47: 00000101 0 10010100 00000011 00010000 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1 | Lights: Memory Trouble Backlight | Zones open | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
  332.83: 11100110 0 00101100 00001000 00000000 00000000 00000000 00000000 00011010 [0xE6] Unknown command
```

jvitkauskas, 2018-06-02:
PC1832: pressed buttons left, right, right, left

```  403.07: 11111111 1 11110111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Left arrow
  404.02: 11111111 1 11110111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Left arrow
  404.57: 11111111 1 11110111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Left arrow
  405.13: 11111111 1 11110111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Left arrow
```

Seems like there's no difference between left and right in logs?

jvitkauskas, 2018-06-02:
PC1832: idle, after configuring the time.

```
 1862.96: 10100101 0 00011000 00011000 01010001 01110000 00000000 00000000 10010110 [0xA5] 06/02/2018 17:28 | Timestamp
 1872.83: 00000101 0 00000000 00010111 00010000 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1 | Lights: none | Unrecognized command: 0x17 | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
 1879.06: 11111111 1 11111111 11111111 11111111 11111111 11111111 11110111 11111111 11111111 [Keypad] Unrecognized key: 0xFF
 1879.14: 00011011 0 00010000 11000111 00010000 11000111 00010000 11000111 00010000 11000111 [0x1B] Unrecognized command [No CRC or CRC Error]
 1879.15: 11111111 1 11111111 11111111 11111110 11111111 11111111 11111111 11111111 11111111 [Keypad] Unknown Keybus notification
 1879.21: 11100110 0 00101100 00000001 00000000 00000000 00000000 00000000 00010011 [0xE6] Unknown command
 1879.39: 11100110 0 00011010 01000000 00000000 00000000 00010001 00000000 00000000 00000000 01010001 [0xE6] Unknown command
 1879.42: 01110101 0 00000000 01110101 [0x75] Partition 1 | Beep pattern: off
 1879.87: 11100110 0 00011000 00000001 00010000 00000000 00000000 00000000 00000000 00001111 [0xE6] Unknown command
 1892.83: 11100110 0 00101100 00000010 00000000 00000000 00000000 00000000 00010100 [0xE6] Unknown command
 1952.80: 11100110 0 00101100 00000100 00000000 00000000 00000000 00000000 00010110 [0xE6] Unknown command
 2012.85: 11100110 0 00101100 00001000 00000000 00000000 00000000 00000000 00011010 [0xE6] Unknown command
 2072.81: 11100110 0 00001011 00000000 11110001 [0xE6] Unknown command
 2102.96: 10100101 0 00011000 00011000 01010001 10000000 00000000 00000000 10100110 [0xA5] 06/02/2018 17:32 | Timestamp
 2132.79: 11100110 0 00101100 00010000 00000000 00000000 00000000 00000000 00100010 [0xE6] Unknown command
 2192.84: 11100110 0 00101100 00100000 00000000 00000000 00000000 00000000 00110010 [0xE6] Unknown command
 2252.84: 11100110 0 00101100 01000000 00000000 00000000 00000000 00000000 01010010 [0xE6] Unknown command
 2312.79: 11100110 0 00001101 00000000 11110011 [0xE6] Unknown command
 2342.98: 10100101 0 00011000 00011000 01010001 10010000 00000000 00000000 10110110 [0xA5] 06/02/2018 17:36 | Timestamp
 2372.79: 11100110 0 00101100 10000000 00000000 00000000 00000000 00000000 10010010 [0xE6] Unknown command
 2432.84: 11100110 0 00101100 00000001 00000000 00000000 00000000 00000000 00010011 [0xE6] Unknown command
```

I still have a yellow triangle (trouble) flashing on my keypad and I have the following troubles:

Service Req.
AC Loss
Phone Line

jvitkauskas, 2018-06-02:
PC1832: idle, after configuring the time & disabling phone line

```
 2882.92: 00010110 0 00001110 01000001 11110111 01011100 [0x16] Unknown panel | Zone wiring: DEOL | Interval 4m
 2883.00: 00000101 0 00000000 00010111 00010000 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1 | Lights: none | Unrecognized command: 0x17 | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
 2895.41: 11111111 1 11111111 11111111 11111111 11111111 11111111 11110111 11111111 11111111 [Keypad] Unrecognized key: 0xFF
 2895.49: 00011011 0 00010000 11000111 00010000 11000111 00010000 11000111 00010000 11000111 [0x1B] Unrecognized command [No CRC or CRC Error]
 2895.50: 11111111 1 11111111 11111111 11111110 11111111 11111111 11111111 11111111 11111111 [Keypad] Unknown Keybus notification
 2895.56: 11100110 0 00101100 00000001 00000000 00000000 00000000 00000000 00010011 [0xE6] Unknown command
 2895.74: 11100110 0 00011010 01000000 00000000 00000000 00010001 00000000 00000000 00000000 01010001 [0xE6] Unknown command
 2895.88: 11100110 0 00011000 00000001 00010000 00000000 00000000 00000000 00000000 00001111 [0xE6] Unknown command
 2895.91: 01110101 0 00000000 01110101 [0x75] Partition 1 | Beep pattern: off
 2912.82: 11100110 0 00101100 00000010 00000000 00000000 00000000 00000000 00010100 [0xE6] Unknown command
 2972.78: 11100110 0 00101100 00000100 00000000 00000000 00000000 00000000 00010110 [0xE6] Unknown command
 3002.86: 00101101 0 00000000 00010111 00010000 11000111 00000000 00011011 [0x2D] Partition 1 | Lights: none | Unrecognized command: 0x17 | Partition 2: disabled | Zones 9-16 open: none
 3032.75: 11100110 0 00001011 00000000 11110001 [0xE6] Unknown command
 3062.97: 10100101 0 00011000 00011000 01010001 11000000 00000000 00000000 11100110 [0xA5] 06/02/2018 17:48 | Timestamp
 3092.77: 11100110 0 00101100 00001000 00000000 00000000 00000000 00000000 00011010 [0xE6] Unknown command
 3152.80: 11100110 0 00101100 00010000 00000000 00000000 00000000 00000000 00100010 [0xE6] Unknown command
 3212.82: 11100110 0 00101100 00100000 00000000 00000000 00000000 00000000 00110010 [0xE6] Unknown command
 3272.77: 11100110 0 00001101 00000000 11110011 [0xE6] Unknown command
 3302.93: 10100101 0 00011000 00011000 01010001 11010000 00000000 00000000 11110110 [0xA5] 06/02/2018 17:52 | Timestamp
 3332.80: 11100110 0 00101100 01000000 00000000 00000000 00000000 00000000 01010010 [0xE6] Unknown command
 3392.77: 11100110 0 00101100 10000000 00000000 00000000 00000000 00000000 10010010 [0xE6] Unknown command
 3452.77: 11100110 0 00101100 00000001 00000000 00000000 00000000 00000000 00010011 [0xE6] Unknown command
```

I still have a yellow triangle (trouble) flashing on my keypad and I have the following troubles:

Service Req. *
AC Loss

jvitkauskas, 2018-06-02:
PC1832: Holding chime button to turn OFF door chime:
```
 4327.55: 11111111 1 10111011 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Door chime configuration
 4327.63: 11100110 0 00011010 01000000 00000000 00000000 00010001 00000000 00000000 00000000 01010001 [0xE6] Unknown command
 4327.66: 01111111 0 00000001 10000000 [0x7F] Beep: long beep
 4327.81: 11100110 0 00011000 00000001 00010000 00000000 00000000 00000000 00000000 00001111 [0xE6] Unknown command
 4327.89: 00000101 0 10010100 10100100 00010000 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1 | Lights: Memory Trouble Backlight | Door chime disabled | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
 4329.79: 00000101 0 10010100 00000011 00010000 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1 | Lights: Memory Trouble Backlight | Zones open | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
```

Holding chime button to turn ON door chrime:
```
 4378.15: 11111111 1 10111011 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Door chime configuration
 4378.24: 11100110 0 00011010 01000000 00000000 00000000 00010001 00000001 00000000 00000000 01010010 [0xE6] Unknown command
 4378.27: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
 4378.41: 11100110 0 00011000 00000001 00010000 00000000 00000000 00000000 00000000 00001111 [0xE6] Unknown command
 4378.49: 00000101 0 10010100 10100011 00010000 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1 | Lights: Memory Trouble Backlight | Door chime enabled | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
 4380.78: 00000101 0 10010100 00000011 00010000 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1 | Lights: Memory Trouble Backlight | Zones open | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
```

taligentx, 2018-06-02:
@jvitkauskas Excellent thanks for the detailed logs!  It's very strange that the left and right arrow keys are producing the same data.  It's possible that the arrow keys only act on the microcontroller within the keypad itself and the keybus message is just a notification that an arrow key was pressed.  I'll be reviewing the rest of the logs as well.

As far as decoding progress:  I did get DLS functional but the DLS access code had been changed on my 1864.  I did a little more digging discovered that the `6666` installer code that was accepted by the panel is some sort of dummy or factory code.  Fortunately, the well known 6321 code worked so I have access to the panel.  This was frustrating enough that I'll be adding functionality to the library to support an unlocker sketch to brute force the installer code on any panel, including handling keypad lockout (either by automatically waiting for the lockout to expire or trigger an external relay to cycle the panel power).  Without keypad lockout it should be possible to unlock a panel in a few hours.

I'm wrapping up decoding partition 2 and after a bit more verification I will be pushing out a new branch for multiple partitions for testing.  Partition 2 was relatively easy because the first PowerSeries panels only supported up to 2 partitions, so the original protocol is designed for only 2 partitions - the 0xA5 command only has 2 bits for the partition number, and the status command 0x05 has 4 bytes, 2 bytes per partition.  The current generation PowerSeries uses 8 bytes for command 0x05 for 4 partitions, and introduces a new command, 0x1B for partitions 5-8.

Partitions 3-8 will require decoding the 0xE6 command - this is new to the current generation PowerSeries so the decoding will be from scratch.  It's already very different from the rest of the commands in that the length is variable.

So, still work to be done but soon the library should be ready to work with partitions 1-2 and zones 1-32.

jvitkauskas, 2018-06-03:
I've also got Envisalink EVL4. Maybe we can use it as an oracle for unknown commands?

taligentx, 2018-06-07:
I've mostly decoded 0xE6 and 0xEB and have pushed a new [develop branch](https://github.com/taligentx/dscKeybusInterface/tree/develop) for feedback and testing,  supporting partitions 2-8, zones 33-64, and writing to partitions 1 and 2.  Please note that the library methods have changed and sketches will need to be updated with the new methods - I was hoping to avoid changing the public interface but supporting multiple partitions requires it.

Next up will be decoding writing to partitions 3-8 if the PC series keypads I have available will support it - otherwise logs of keypads assigned to these partitions and writing keys would be great.

@jvitkauskas  Thanks for the notes on DLS!  If it's useful in the future, I came across a way to connect to the panel with DLS without going into the panel installer programming.  With the cable unplugged from the panel, start the connection from DLS and then plug the cable into the panel while the connection is pending.  The panel will automatically detect the connection attempt and connect to DLS.  

Also, thanks for the offer for the Envisalink output!  From their TPI document, it seems to cover the same topics as the IT-100 - a few of these are on the todo list (command 1-4 output, writing partitions 3-8, zone tamper/fault 33-64) and a few I can't test (thermostat, wireless, etc).  If you come across any output from the Envisalink that does not match the output in KeybusReader please let me know, there are still quite a few command sequences that are unknown (though none that should affect the core functionality of tracking alarm status and controlling the panel).

taligentx, 2018-06-14:
Update on writing to partitions 3-8 - the PC series keypads I have do not support these partitions, logs would be useful if this is something that should be implemented.

jvitkauskas, 2018-06-16:
Hi, unfortunately I cannot arm the system, because i don't have it installed and I don't have AC power source and siren connected.

However, I have configured my PC1832 for 4 partitions:
* Partition 1: Zones 1, 5, 6, 7, 8
* Partition 2: Zone 2
* Partition 3: Zone 3
* Partition 4: Zone 4

Here I am providng logs of me shorting zones and restoring them afterwards.

Zone 1 / Partition 1:
```
 1978.09: 11100110 0 00001101 00000000 11110011 [0xE6] Zones 49-56 open: none
 1980.09: 11100110 0 00001111 00000000 11110101 [0xE6] Zones 57-64 open: none
 1982.32: 00100111 0 00000000 00010111 10010000 00000011 00000001 11010010 [0x27] Partition 1: none - Unrecognized data17 | Partition 2: Trouble Backlight - Zones open | Zones 1-8 open: 1 
 1982.42: 11100110 0 00001001 00000000 11101111 [0xE6] Zones 33-40 open: none
 1982.63: 10100101 0 00011000 00011010 00010110 00011101 10001100 11111111 10010101 [0xA5] 2018.06.16 22:07 | Zone fault: 1
 1982.69: 11001110 0 00100000 00011101 10001100 00000000 00000000 10010111 [0xCE] Unrecognized data
 1982.87: 11101011 0 00000000 00011000 00011010 00010110 00011100 00000001 10001100 11111111 11011011 [0xEB] 2018.06.16 22:07 | Zone fault: 1
 1982.95: 11111111 1 11111111 11111111 11111111 11111011 11111111 11111111 11111111 11111111 [Keypad] Zone status notification
 1983.05: 11010101 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0xD5] Keypad zone query
 1983.05: 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 11111111 11000011 [Keypad] Slot 8 zone open
 1984.64: 00100111 0 00000000 00010111 10010000 00000011 00000000 11010001 [0x27] Partition 1: none - Unrecognized data17 | Partition 2: Trouble Backlight - Zones open | Zones 1-8 open: none
 1984.71: 10100101 0 00011000 00011010 00010110 00011101 01101100 11111111 01110101 [0xA5] 2018.06.16 22:07 | Zone fault restored: 1
 1984.78: 11001110 0 00100000 00011101 01101100 00000000 00000000 01110111 [0xCE] Unrecognized data
 1984.96: 11101011 0 00000000 00011000 00011010 00010110 00011100 00000001 01101100 11111111 10111011 [0xEB] 2018.06.16 22:07 | Zone fault restored: 1
```

Zone 2 / Partition 2:
```
 2018.14: 00100111 0 00000000 00010111 00000000 00010111 00000010 01010111 [0x27] Partition 1: none - Unrecognized data17 | Partition 2: none - Unrecognized data17 | Zones 1-8 open: 2 
 2018.30: 10100101 0 00011000 00011010 00010110 00011101 10001101 11111111 10010110 [0xA5] 2018.06.16 22:07 | Zone fault: 2
 2018.36: 11001110 0 00100000 00011101 10001101 00000000 00000000 10011000 [0xCE] Unrecognized data
 2018.54: 11101011 0 00000000 00011000 00011010 00010110 00011100 00000001 10001101 11111111 11011100 [0xEB] 2018.06.16 22:07 | Zone fault: 2
 2020.28: 00100111 0 00000000 00010111 00000000 00010111 00000000 01010101 [0x27] Partition 1: none - Unrecognized data17 | Partition 2: none - Unrecognized data17 | Zones 1-8 open: none
 2020.34: 10100101 0 00011000 00011010 00010110 00011101 01101101 11111111 01110110 [0xA5] 2018.06.16 22:07 | Zone fault restored: 2
 2020.41: 11001110 0 00100000 00011101 01101101 00000000 00000000 01111000 [0xCE] Unrecognized data
 2020.59: 11101011 0 00000000 00011000 00011010 00010110 00011100 00000001 01101101 11111111 10111100 [0xEB] 2018.06.16 22:07 | Zone fault restored: 2
```

Zone 3 / Partition 3:
```
 2050.04: 00100111 0 00000000 00010111 00000000 00010111 00000100 01011001 [0x27] Partition 1: none - Unrecognized data17 | Partition 2: none - Unrecognized data17 | Zones 1-8 open: 3 
 2050.20: 10100101 0 00011000 00011010 00010110 00100001 10001110 11111111 10011011 [0xA5] 2018.06.16 22:08 | Zone fault: 3
 2050.26: 11001110 0 00100000 00100001 10001110 00000000 00000000 10011101 [0xCE] Unrecognized data
 2050.44: 11101011 0 00000000 00011000 00011010 00010110 00100000 00000001 10001110 11111111 11100001 [0xEB] 2018.06.16 22:08 | Zone fault: 3
 2052.09: 01100100 0 00000100 01101000 [0x64] Partition 1 | Beep: 2 beeps
 2052.12: 01101001 0 00000100 01101101 [0x69] Partition 2 | Beep: 2 beeps
 2052.17: 11100110 0 00011001 00000100 00000100 00000111 [0xE6] Partition 3 | Beep: 2 beeps
 2052.22: 11100110 0 00011001 00001000 00000100 00001011 [0xE6] Partition 4 | Beep: 2 beeps
 2053.96: 00100111 0 00000000 00010111 00000000 00010111 00000000 01010101 [0x27] Partition 1: none - Unrecognized data17 | Partition 2: none - Unrecognized data17 | Zones 1-8 open: none
 2054.07: 10100101 0 00011000 00011010 00010110 00100001 01101110 11111111 01111011 [0xA5] 2018.06.16 22:08 | Zone fault restored: 3
 2054.21: 11001110 0 00100000 00100001 01101110 00000000 00000000 01111101 [0xCE] Unrecognized data
 2054.39: 11101011 0 00000000 00011000 00011010 00010110 00100000 00000001 01101110 11111111 11000001 [0xEB] 2018.06.16 22:08 | Zone fault restored: 3
```

Zone 4 / Partition 4:
```
 2110.75: 00100111 0 00000000 00010111 00000000 00010111 00001000 01011101 [0x27] Partition 1: none - Unrecognized data17 | Partition 2: none - Unrecognized data17 | Zones 1-8 open: 4 
 2110.82: 10100101 0 00011000 00011010 00010110 00100101 10001111 11111111 10100000 [0xA5] 2018.06.16 22:09 | Zone fault: 4
 2110.88: 11001110 0 00100000 00100101 10001111 00000000 00000000 10100010 [0xCE] Unrecognized data
 2111.09: 11101011 0 00000000 00011000 00011010 00010110 00100100 00000001 10001111 11111111 11100110 [0xEB] 2018.06.16 22:09 | Zone fault: 4
 2112.08: 01100100 0 00000100 01101000 [0x64] Partition 1 | Beep: 2 beeps
 2112.11: 01101001 0 00000100 01101101 [0x69] Partition 2 | Beep: 2 beeps
[... cut ...]
 2113.71: 00100111 0 00000000 00010111 00000000 00010111 00000000 01010101 [0x27] Partition 1: none - Unrecognized data17 | Partition 2: none - Unrecognized data17 | Zones 1-8 open: none
 2113.78: 10100101 0 00011000 00011010 00010110 00100101 01101111 11111111 10000000 [0xA5] 2018.06.16 22:09 | Zone fault restored: 4
 2113.85: 11001110 0 00100000 00100101 01101111 00000000 00000000 10000010 [0xCE] Unrecognized data
 2114.06: 11101011 0 00000000 00011000 00011010 00010110 00100100 00000001 01101111 11111111 11000110 [0xEB] 2018.06.16 22:09 | Zone fault restored: 4
```

jvitkauskas, 2018-06-16:
And here is idle log:

```
 2899.75: 11111111 1 11111111 11111111 11111111 11111111 11111111 11110111 11111111 11111111 [Keypad] Unrecognized key: 0xFF
 2899.83: 11111111 1 11111111 11111111 11111110 11111111 11111111 11111111 11111111 11111111 [Keypad] Unknown Keybus notification
 2899.90: 11100110 0 00101100 00000001 00000000 00000000 00000000 00000000 00010011 [0xE6] Partition 1 | Enabled zones 33-64: none
 2900.08: 11100110 0 00011010 01000000 00000000 00000000 00010001 00000000 00000000 00000000 01010001 [0xE6] 0x1A: Unrecognized data
 2900.22: 11100110 0 00011000 00000001 00010000 00000000 00000000 00000000 00000000 00001111 [0xE6] Partition 1 | Status lights flashing: Trouble | Zones 33-64 flashing: none
 2900.25: 01110101 0 00000000 01110101 [0x75] Partition 1 | Beep pattern: off
 2912.11: 11100110 0 00101100 00000010 00000000 00000000 00000000 00000000 00010100 [0xE6] Partition 2 | Enabled zones 33-64: none
 2972.07: 11100110 0 00001011 00000000 11110001 [0xE6] Zones 41-48 open: none
 3002.23: 10100101 0 00011000 00011010 00010110 01100000 00000000 00000000 01001101 [0xA5] 2018.06.16 22:24 | Timestamp
 3032.10: 11100110 0 00101100 00000100 00000000 00000000 00000000 00000000 00010110 [0xE6] Partition 3 | Enabled zones 33-64: none
 3032.17: 11100110 0 00101011 00000100 00000100 00000000 00000000 00000000 00011001 [0xE6] Partition 3 | Enabled zones  1-32: 3 
 3062.23: 00010110 0 00001110 01000001 10110111 00011100 [0x16] Unknown panel | Zone wiring: DEOL | Interval 4m
 3092.06: 11100110 0 00101100 00001000 00000000 00000000 00000000 00000000 00011010 [0xE6] Partition 4 | Enabled zones 33-64: none
 3092.13: 11100110 0 00101011 00001000 00001000 00000000 00000000 00000000 00100001 [0xE6] Partition 4 | Enabled zones  1-32: 4 
 3152.10: 11100110 0 00101100 00010000 00000000 00000000 00000000 00000000 00100010 [0xE6] Partition 5 | Enabled zones 33-64: none
 3212.02: 11100110 0 00001101 00000000 11110011 [0xE6] Zones 49-56 open: none
 3242.26: 10100101 0 00011000 00011010 00010110 01110000 00000000 00000000 01011101 [0xA5] 2018.06.16 22:28 | Timestamp
 3272.10: 11100110 0 00101100 00100000 00000000 00000000 00000000 00000000 00110010 [0xE6] Partition 6 | Enabled zones 33-64: none
 3332.10: 11100110 0 00101100 01000000 00000000 00000000 00000000 00000000 01010010 [0xE6] Partition 7 | Enabled zones 33-64: none
 3392.05: 11100110 0 00101100 10000000 00000000 00000000 00000000 00000000 10010010 [0xE6] Partition 8 | Enabled zones 33-64: none
 3452.02: 11100110 0 00001111 00000000 11110101 [0xE6] Zones 57-64 open: none
 3482.22: 10100101 0 00011000 00011010 00010110 10000000 00000000 00000000 01101101 [0xA5] 2018.06.16 22:32 | Timestamp
 3512.09: 11100110 0 00101100 00000001 00000000 00000000 00000000 00000000 00010011 [0xE6] Partition 1 | Enabled zones 33-64: none
 3572.05: 11100110 0 00101100 00000010 00000000 00000000 00000000 00000000 00010100 [0xE6] Partition 2 | Enabled zones 33-64: none
 3632.09: 11100110 0 00101100 00000100 00000000 00000000 00000000 00000000 00010110 [0xE6] Partition 3 | Enabled zones 33-64: none
 3632.16: 11100110 0 00101011 00000100 00000100 00000000 00000000 00000000 00011001 [0xE6] Partition 3 | Enabled zones  1-32: 3 
 3662.18: 00100111 0 00000000 00010111 00000000 00010111 00000000 01010101 [0x27] Partition 1: none - Unrecognized data17 | Partition 2: none - Unrecognized data17 | Zones 1-8 open: none
 3662.33: 11001110 0 00000010 00000001 00000000 00000000 00000000 11010001 [0xCE] Unrecognized data
 3662.41: 10100101 0 00011000 00011010 00010110 10001000 11101000 11111111 01011100 [0xA5] 2018.06.16 22:34 | Panel AC power failure
 3662.47: 11001110 0 00100000 10001000 11101000 00000000 00000000 01011110 [0xCE] Unrecognized data
 3662.64: 11101011 0 00000000 00011000 00011010 00010110 10001000 00000000 11101000 11111111 10100010 [0xEB] 2018.06.16 22:34 | Panel AC power failure
 3692.06: 11100110 0 00001001 00000000 11101111 [0xE6] Zones 33-40 open: none
 3722.22: 10100101 0 00011000 00011010 00010110 10010000 00000000 00000000 01111101 [0xA5] 2018.06.16 22:36 | Timestamp
 3752.09: 11100110 0 00101100 00001000 00000000 00000000 00000000 00000000 00011010 [0xE6] Partition 4 | Enabled zones 33-64: none
 3752.16: 11100110 0 00101011 00001000 00001000 00000000 00000000 00000000 00100001 [0xE6] Partition 4 | Enabled zones  1-32: 4 
 3812.09: 11100110 0 00101100 00010000 00000000 00000000 00000000 00000000 00100010 [0xE6] Partition 5 | Enabled zones 33-64: none
 3872.09: 11100110 0 00101100 00100000 00000000 00000000 00000000 00000000 00110010 [0xE6] Partition 6 | Enabled zones 33-64: none
 3932.06: 11100110 0 00001011 00000000 11110001 [0xE6] Zones 41-48 open: none
 3962.22: 10100101 0 00011000 00011010 00010110 10100000 00000000 00000000 10001101 [0xA5] 2018.06.16 22:40 | Timestamp
 3992.07: 11100110 0 00101100 01000000 00000000 00000000 00000000 00000000 01010010 [0xE6] Partition 7 | Enabled zones 33-64: none
 4052.09: 11100110 0 00101100 10000000 00000000 00000000 00000000 00000000 10010010 [0xE6] Partition 8 | Enabled zones 33-64: none
 4112.09: 11100110 0 00101100 00000001 00000000 00000000 00000000 00000000 00010011 [0xE6] Partition 1 | Enabled zones 33-64: none
 4172.01: 11100110 0 00001101 00000000 11110011 [0xE6] Zones 49-56 open: none
 4202.24: 10100101 0 00011000 00011010 00010110 10110000 00000000 00000000 10011101 [0xA5] 2018.06.16 22:44 | Timestamp
[... cut ...]
 6932.06: 11100110 0 00101011 00001000 00001000 00000000 00000000 00000000 00100001 [0xE6] Partition 4 | Enabled zones  1-32: 4 
 6991.99: 11100110 0 00101100 00010000 00000000 00000000 00000000 00000000 00100010 [0xE6] Partition 5 | Enabled zones 33-64: none
 7051.96: 11100110 0 00001101 00000000 11110011 [0xE6] Zones 49-56 open: none
 7082.17: 10100101 0 00011000 00011010 00010111 10000000 00000000 00000000 01101110 [0xA5] 2018.06.16 23:32 | Timestamp
 7112.04: 11100110 0 00101100 00100000 00000000 00000000 00000000 00000000 00110010 [0xE6] Partition 6 | Enabled zones 33-64: none
```

jvitkauskas, 2018-06-17:
Here is me configuring keypad for partition 3 slot 8:
```
    6.49: 11111111 1 00101000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: *
    6.55: 01011101 0 00000000 00000000 00000000 00000000 00000000 01011101 [0x5D] Partition 1 | Status lights flashing: none | Zones 1-32 flashing: none
    6.64: 11100110 0 00011000 00000001 00000000 00000000 00000000 00000000 00000000 11111111 [0xE6] Partition 1 | Status lights flashing: none | Zones 33-64 flashing: none
    6.72: 00001010 0 10000000 10011110 00000000 00000000 00000000 00000000 00000000 00101000 [0x0A] Partition 1: Backlight - Enter * function code | Zone lights: none
    6.80: 11100110 0 00100000 10000000 10011110 00000000 00000000 00000000 00000000 10000000 10100100 [0xE6] Status lights: Backlight - Enter * function code | Zone lights: none
    6.89: 00000101 0 10000000 10011110 10010000 00000011 10010000 00000011 10010000 00000011 [0x05] Partition 1: Backlight - Enter * function code | Partition 2: Trouble Backlight - Zones open | Partition 3: Trouble Backlight - Zones open | Partition 4: Trouble Backlight - Zones open
    6.91: 11111111 1 00100010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 8
    6.97: 00000101 0 10000000 10110111 10010000 00000011 10010000 00000011 10010000 00000011 [0x05] Partition 1: Backlight - Enter installer code | Partition 2: Trouble Backlight - Zones open | Partition 3: Trouble Backlight - Zones open | Partition 4: Trouble Backlight - Zones open
    8.58: 11111111 1 00010110 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 5
    8.90: 11111111 1 00010110 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 5
    9.23: 11111111 1 00010110 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 5
   10.13: 11111111 1 00010110 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 5
   10.16: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
   10.47: 00010110 0 00001110 01000001 10100111 00001100 [0x16] Unknown panel | Zone wiring: DEOL | Unrecognized data
   10.53: 01011101 0 00100000 00000000 00000000 00000000 00000000 01111101 [0x5D] Partition 1 | Status lights flashing: Program | Zones 1-32 flashing: none
   10.64: 11100110 0 00011000 00000001 00100000 00000000 00000000 00000000 00000000 00011111 [0xE6] Partition 1 | Status lights flashing: Program | Zones 33-64 flashing: none
   10.72: 00001010 0 10000010 11100100 00000000 00000000 00000000 00000000 00000000 01110000 [0x0A] Partition 1: Armed Backlight - *8: Main menu | Zone lights: none
   10.80: 11100110 0 00100000 10000010 11100100 00000000 00000000 00000000 00000000 10000000 11101100 [0xE6] Status lights: Armed Backlight - *8: Main menu | Zone lights: none
   10.88: 00000101 0 10000010 11100100 10010000 11000111 10010000 11000111 10010000 11000111 [0x05] Partition 1: Armed Backlight - *8: Main menu | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
   10.95: 10100101 0 00011000 00011010 00101010 10010101 10101101 00000000 01000011 [0xA5] 2018.06.17 10:37 | Enter *8 programming
   11.30: 11101011 0 00000000 00011000 00011010 00101010 10010100 00000001 10101101 00000000 10001001 [0xEB] 2018.06.17 10:37 | Enter *8 programming
   15.87: 11111111 1 00000000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 0
   16.20: 11111111 1 00000000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 0
   16.63: 11111111 1 11111111 00000000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 2 | Key: 0
   16.66: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
   16.81: 00001010 0 10000001 11111000 00000000 00000000 00000000 00000000 00000000 10000011 [0x0A] Partition 1: Ready Backlight - Keypad programming | Zone lights: none
   16.90: 11100110 0 00100000 10000001 11111000 00000000 00000000 00000000 00000000 10000000 11111111 [0xE6] Status lights: Ready Backlight - Keypad programming | Zone lights: none
   16.98: 00000101 0 10000001 11111000 10010000 11000111 10010000 11000111 10010000 11000111 [0x05] Partition 1: Ready Backlight - Keypad programming | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
   18.45: 11111111 1 00000000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 0
   18.48: 01100100 0 00001000 01101100 [0x64] Partition 1 | Beep: 4 beeps
   18.56: 00001010 0 10000001 11100101 00000000 00000000 00000000 00000000 00000000 01110000 [0x0A] Partition 1: Ready Backlight - Keypad slot assignment | Zone lights: none
   18.65: 11100110 0 00100000 10000001 11100101 00000000 00000000 00000000 00000000 10000000 11101100 [0xE6] Status lights: Ready Backlight - Keypad slot assignment | Zone lights: none
   18.73: 00000101 0 10000001 11100101 10010000 11000111 10010000 11000111 10010000 11000111 [0x05] Partition 1: Ready Backlight - Keypad slot assignment | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
   20.94: 11111111 1 11111111 11110111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 2 | Key: Left/right arrow
   22.51: 11111111 1 11110111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: Left/right arrow
[... cut ...]
  334.37: 00000101 0 00000000 00010111 00000000 00010111 10010000 00000111 00000000 00010111 [0x05] Partition 1: none - Unrecognized data17 | Partition 2: none - Unrecognized data17 | Partition 3: Trouble Backlight - Failed to arm | Partition 4: none - Unrecognized data17
  336.74: 00000101 0 00000000 00010111 00000000 00010111 10010000 00000011 00000000 00010111 [0x05] Partition 1: none - Unrecognized data17 | Partition 2: none - Unrecognized data17 | Partition 3: Trouble Backlight - Zones open | Partition 4: none - Unrecognized data17
```

jvitkauskas, 2018-06-17:
Here's keys for partition 3:

| Key  | Bytes |
| ------------- | ------------- |
| Left/right  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 11110111 11111111  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |

jvitkauskas, 2018-06-17:
**Partition 4**

Everything in this reply was done with keypad configured to partition 4.

Key codes:

| Key | Bytes |
| ------------- | ------------- |
| Left  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 11110111 11111111  |
| Right  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 11110111 11111111  |
| 1 | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00000101 11111111  |
| 2  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00001010 11111111  |
| 3  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00001111 11111111  |
| 4  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00010001 11111111  |
| 5  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00010110 11111111  |
| 6  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00011011 11111111  |
| 7  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00011100 11111111  |
| 8  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00100010 11111111  |
| 9  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00100111 11111111  |
| 0 | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00000000 11111111  |
| *  | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00101000 11111111  |
| # | 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00101101 11111111  |

jvitkauskas, 2018-06-17:
I've also tried configuring keypad to partition 4 slot 1 and key codes do not seem to differ.

jvitkauskas, 2018-06-17:
@taligentx where are you located? I might be able to get you a keypad

taligentx, 2018-06-18:
@jvitkauskas Thanks to your logs I've added partitions 3-8 to virtual keypad in the develop branch.  On the decoding side, I've found that writing to partitions 5-8 uses the same bit positions as partitions 1-4 but sends the key during a different status command.

Also thanks for the very generous offer for the keypad!  I'll be in touch.

taligentx, 2018-07-17:
Added zones 33-64 tamper and fault decoding after picking up a zone expander, this completes the basic decoding for partitions 1-8 and zones 33-64.  Thanks all for the valuable feedback and logs!

closed 2018-07-17 by taligentx