taligentx/dscKeybusInterface #3: Great job

Quiqui64, 2018-05-03:
Great job, I will be trying to interface this with Smartthings when return from travel.

Thank you,

taligentx, 2018-05-04:
Hi @Quiqui64,

Yes, the panel DC ground (Aux -) must be connected to the Arduino and esp8266 DC ground to provide a reference for the clock and data lines - see the wiring diagram in the examples and README.md.

As for SmartThings, if it works out that would be a good example to add to this library (if you'd like to share the code).  The [SmartThings MQTT Bridge](https://github.com/stjohnjohnson/smartthings-mqtt-bridge) may be a suitable way to integrate (see the Status-MQTT-Homebridge example) - or do you have another method in mind?

adrianmihalko, 2018-05-04:
Nice project, too bad I already bought an Envisalink and this still missing a proper Homekit integration.

taligentx, 2018-05-04:
Hi @adrianmihalko - do you mean this library is missing proper HomeKit integration or the Envisalink?  For the Envisalink, there is [dustindclark/homebridge-envisalink](https://github.com/dustindclark/homebridge-envisalink).

taligentx, 2018-05-08:
Closing as the ground wiring question is resolved.

closed 2018-05-08 by taligentx