taligentx/dscKeybusInterface #4: No ArmedStay detected

MagnusPer, 2018-05-04:
Great work!! Have done some initial test using the Arduino Status Sketch on NodeMCU, there is **no status ArmedStay** detected only Exit delay in progress and  Partition disarmed

Alarm system: Powerserie 832 - PC5015
[Log_powerSerie832_PC5015-NoArmedStayDetection_KeybusReader.txt](https://github.com/taligentx/dscKeybusInterface/files/1974806/Log_powerSerie832_PC5015-NoArmedStayDetection_KeybusReader.txt)
[Log_powerSerie832_PC5015-NoArmedStayDetection_Status.txt](https://github.com/taligentx/dscKeybusInterface/files/1974807/Log_powerSerie832_PC5015-NoArmedStayDetection_Status.txt)

taligentx, 2018-05-04:
Hi @MagnusPer,

Thanks for the logs!  (also, I've added an option in KeybusReader to hide keypad digits for posting logs publicly if needed in the future)

It seems that the PC5015 panel does not send as many messages in the 0xA5 command as the PC1555MX.  I've updated the library to use the 0x27 command to detect the arming status, see if this resolves the issue.

Unrelated questions to improve the decoding:
* Is your panel configured to use double end of line resistors?
* What message does the keypad LCD display when you open a zone and KeybusReader shows `[0x05] Status lights: Ready Backlight | Unrecognized command, byte 3: 0x02`?

Next up will be adding the other zone groups for zones 9+.

MagnusPer, 2018-05-05:
Verified OK - Thanks for thr quick update
[Log_powerSerie832_PC5015-NoArmedStayDetection_Status_OK.txt](https://github.com/taligentx/dscKeybusInterface/files/1976951/Log_powerSerie832_PC5015-NoArmedStayDetection_Status_OK.txt)

I return soon with the urealated answers...

MagnusPer, 2018-05-06:
@taligentx 

> Is your panel configured to use double end of line resistors?

All sections (doors and motion detectors ) are configured with double line resistors, exception is the fire detctor that shall be configured by the manual with single line resistor.

> What message does the keypad LCD display when you open a zone and KeybusReader shows [0x05] Status lights: Ready Backlight | Unrecognized command, byte 3: 0x02?

**Alarm System Status**  - Disarmed, no doors open, only motion detectors reporting OPEN/RESTORED while walking around. 
**LCD Display Status** - Ready

Sitting and watching the KeybusReader output I see the command comes and go while the system is in ready state. Sorry but I do not have any idea what that message means, It's atleast not triggered by arming or disarming.

Add an additional log from my observation 
[Log_PowerSerie832_PC5015-Unknown-0x05-Byte3_KeybusReader.txt](https://github.com/taligentx/dscKeybusInterface/files/1977932/Log_PowerSerie832_PC5015-Unknown-0x05-Byte3_KeybusReader.txt)

Maybe the answer is in other projects  such https://github.com/sjlouw/dsc-alarm-arduino/blob/master/DSCPanel/DSC.cpp

taligentx, 2018-05-06:
Figured it out - 0x02 seems to be a flag for stay/away zones open, while 0x03 is a flag for the other zones.   Thanks for the links to the other projects, I've actually looked them over previously and used what I could, but in some cases (like keypads) the data was different than my results (I suspect this is due to whether the data is being read with a sufficient delay).  Thanks for the logs, they're very useful!

closed 2018-05-06 by taligentx