taligentx/dscKeybusInterface #5: Continuous status - Zone Restored 9

MagnusPer, 2018-05-06:
In the last release when you added the support for Zone B - Section 9-16 the sections are continuously reported - **Zone restored: 9** 

**Info:** My alarm system has an expansion card for additional 8 sections (9-16)

[Log_PowerSerie832_PC5015-ContinuesRestoredZone-9_KeybusReader.txt](https://github.com/taligentx/dscKeybusInterface/files/1977895/Log_PowerSerie832_PC5015-ContinuesRestoredZone-9_KeybusReader.txt)
[Log_PowerSerie832_PC5015-ContinuousRestoredZone-9_Status.txt](https://github.com/taligentx/dscKeybusInterface/files/1977896/Log_PowerSerie832_PC5015-ContinuousRestoredZone-9_Status.txt)

taligentx, 2018-05-06:
I've updated the Status example to handle the zone change flags, let's see if this resolves the issue.

MagnusPer, 2018-05-07:
Issue resolved, Thanks for the update!
Now the Zone 9 returns to restored only one time.

[Log_PowerSerie832_PC5015-ContinuesRestoredZone-9_KeybusReader_OK.txt](https://github.com/taligentx/dscKeybusInterface/files/1980696/Log_PowerSerie832_PC5015-ContinuesRestoredZone-9_KeybusReader_OK.txt)
[Log_PowerSerie832_PC5015-ContinuousRestoredZone-9_Status_OK.txt](https://github.com/taligentx/dscKeybusInterface/files/1980700/Log_PowerSerie832_PC5015-ContinuousRestoredZone-9_Status_OK.txt)