taligentx/dscKeybusInterface #6: Zone Alarm not reported

MagnusPer, 2018-05-09:
Zone 9 in alarm state but not reported in status sketch.
```
Exit delay in progress
Zone open: 7
Zone restored: 7
Partition armed away
Zone open: 9
05/09/2018 19:00 | Partition in alarm
Zone open: 7
Partition disarmed
05/09/2018 19:00 | Zone alarm restored: 9
Zone restored: 7
Zone restored: 9
```
[Log_PowerSerie832_PC5015-ALARM_zone9_KeybusReader.txt](https://github.com/taligentx/dscKeybusInterface/files/1988940/Log_PowerSerie832_PC5015-ALARM_zone9_KeybusReader.txt)
[Log_PowerSerie832_PC5015-ALARM_zone9_Status.txt](https://github.com/taligentx/dscKeybusInterface/files/1988941/Log_PowerSerie832_PC5015-ALARM_zone9_Status.txt)

taligentx, 2018-05-09:
Interesting - this may be due to the zone expander.  Command 0xA5 has a block of 32 commands that seems likely for zone alarm (immediately after is another block of 32 for the zone alarm restored).  In your logs, there is an entirely new command in 0xA5 instead.  See if this commit is able to decode the zone alarm now.

taligentx, 2018-05-14:
Is your zone expander configured for zones 9-12 or zones 9-16?  The PC5108 expander has many different zone configurations so I wouldn't be surprised if it takes some more data to work it out.

MagnusPer, 2018-05-14:
Since I only used zone 9-12 I haven't thought about this before. 
Right now the jumpers on the [PC5108 expander](http://cms.dsc.com/download.php?t=1&id=12878) are configured as
Zone 9-12  => J1 - on, J2 - off, J3 - on
Zone 13-16 => J4 - off, J5 - off, J6 - on

I will try to move and tripp one detector to the last Zone 8 on the expander board to help you out, but it will take some days, the family starts to loose the temper ;-) they do not have the same view on what is important!

taligentx, 2018-05-14:
That's understandable, I've had a PC1555MX panel on the bench so testing hasn't been a problem - a live environment is another story.  If the problem is noise, disconnecting the siren at the panel shouldn't affect the zone data.

If you have a limited opportunity to make changes, you may want to setup zone 12 instead.  The data for 13-16 could be different since they are in a different group on the expander.  To make the library useful for your environment, I would suggest:

* Connect zone 12
* Start data capture
* Setup zone 12 in programming
* Have zone 9 and 12 open at the same time
* Arm the system
* Trip zone 9, wait several seconds, then trip zone 12
* Close zone 9, wait several seconds, close zone 12.
* Disarm

For zones 13-16, the same procedure would need to be repeated but using zone 13 and zone 16 instead of 9 and 12.  I think this will only happen if someone else is already using these zones on an expander and is able to send the data because making lots of changes to a live environment is difficult.

taligentx, 2018-05-14:
On second thought, forget about zone 12.  I looked over the data again and your idea to setup zone 16 would be more useful:
* Connect zone 16
* Start data capture
* Setup zone 16 in programming
* Have zone 9 and 16 open at the same time
* Arm the system
* Trip zone 9, wait several seconds, then trip zone 16
* Close zone 9, wait several seconds, close zone 16
* Disarm

I also just pushed a fix for displaying the zone expander status message correctly.

taligentx, 2018-05-17:
Looks good, let's keep this issue open until we can confirm the rest of the zones on the expander.

taligentx, 2018-05-17:
I updated the zone expander status decoding again, let's see if the KeybusReader decoding is fixed now.  This is just to decode the data coming directly from the expander, it doesn't affect the Status examples because they use the data coming from the panel.