taligentx/dscKeybusInterface #7: Fire alarm not detected

MagnusPer, 2018-05-09:
Fire Alarm (Zone 10) not reported in status sketch.

```
Zone open: 10  // Fire alarm zone activated
Zone open: 2
Zone open: 1
...
...
Zone restored: 2
Zone restored: 10 // Fire alarm zone restored
```
[Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_Status.txt](https://github.com/taligentx/dscKeybusInterface/files/1988954/Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_Status.txt)
[Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_KeybusReader.txt](https://github.com/taligentx/dscKeybusInterface/files/1988955/Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_KeybusReader.txt)

taligentx, 2018-05-09:
Thanks for the logs, I've added fire status tracking, let me know if there are any issues.

MagnusPer, 2018-05-14:
Works fine!

```
Zone restored: 2
Zone open: 10         // Fire zone open
Fire status on        // Fire status ON
Zone open: 2
...
...
Zone restored: 2
Zone restored: 10      // Fire zone restored
Zone open: 2
Fire status restored   // Fire status OFF
Zone restored: 1
Zone open: 1
```
[Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_Status_OK.txt](https://github.com/taligentx/dscKeybusInterface/files/2001846/Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_Status_OK.txt)
[Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_KeybusReader_OK.txt](https://github.com/taligentx/dscKeybusInterface/files/2001848/Log_PowerSerie832_PC5015-FIRE_ALARM_zone10_KeybusReader_OK.txt)

taligentx, 2018-05-14:
Great, thanks for the update!

closed 2018-05-14 by taligentx