taligentx/dscKeybusInterface #8: [Enhancement] Init value for Fire and Alarm status

MagnusPer, 2018-05-15:
Provide an init value for at least Fire and Alarm at startup, meaning let the status flag be true once.
- dsc.partitionAlarmChanged
- dsc.fireStatusChanged

The logic as of today provides updates only if there has been a status change. In the Homey app there is an absence of the status in the Fire and Tamper alarm device since nothing is provided.

taligentx, 2018-05-15:
Ah, just saw your edit.  Sounds good.

By the way, if you get the Homey integration working to your satisfaction and think others would benefit (and would like to share the code), I can add your example to the library examples.

closed 2018-05-15 by taligentx

MagnusPer, 2018-05-15:
No problem to share my example of integration to Homey when finished