taligentx/dscKeybusInterface #10: Only one Zone Alarm reported

MagnusPer, 2018-05-17:
Only one zone is reported to be in alarm state, its seems it's only the first tripped zone that is reported

```
Partition armed away
Zone open: 9
05/17/2018 16:06 | Zone alarm: 9
05/17/2018 16:06 | Partition in alarm
Zone open: 7                              // Zone configured as door entrance delay
Zone restored: 7
Zone open: 1                              // Motion detection - sholud also be Zone alarm state ??
Zone open: 2                              // Motion detection - sholud also be Zone alarm state ??
Zone restored: 1
Zone open: 5                              // Motion detection - sholud also be Zone alarm state ??
Zone restored: 2
Zone open: 2
Zone open: 1
Zone restored: 5
Partition disarmed
Zone restored: 1
Zone restored: 2
05/17/2018 16:07 | Zone alarm restored: 9
Zone open: 7
Zone restored: 9
```
[Log_PowerSerie832_PC5015-ALARM_OnlyOneZone_Status.txt](https://github.com/taligentx/dscKeybusInterface/files/2013616/Log_PowerSerie832_PC5015-ALARM_OnlyOneZone_Status.txt)
[Log_PowerSerie832_PC5015-ALARM_OnlyOneZone_KeybusReader.txt](https://github.com/taligentx/dscKeybusInterface/files/2013617/Log_PowerSerie832_PC5015-ALARM_OnlyOneZone_KeybusReader.txt)

taligentx, 2018-05-17:
When zone 7 was tripped, the panel started an entrance delay and it will wait until that delay is over before it sends zone alarm commands - you disarmed it before the delay was over, so the panel didn't send any further zone alarm commands, even for the motion detectors.  If you can try again, wait for the entrance delay to finish - you should see the zone 7 light flashing on the keypad once it is in the alarm state, and the microcontroller should detect it as well.

I've seen the same behavior on the PC1555MX - tripping a delay zone after an instant zone results in the panel starting the entrance delay and ignoring stay/away zones until the entrance delay is over, even though the alarm is tripped from the initial instant zone.  I'm not sure why DSC set it up this way - it seems that once the alarm is tripped, all zones should trip instantly without entrance delays.  Perhaps this is changed in the newer panels.

MagnusPer, 2018-05-22:
Followed our instruction in #6 

1. zone 9 and 16 open at the same time
2. Close zone 9 and 16
3. Arm the system
4. Trip zone 9, wait several seconds, then trip zone 16
5. Close zone 9, wait several seconds, close zone 16
6. Disarm

Still only one Zone Alarm reported
```
Exit delay in progress
Zone restored: 7
Partition armed away
Zone open: 9
05/22/2018 16:23 | Zone alarm: 9
05/22/2018 16:23 | Partition in alarm
Zone open: 16
Zone restored: 16
05/22/2018 16:23 | Zone alarm restored: 16
Zone open: 11
Zone restored: 11
05/22/2018 16:23 | Zone alarm restored: 11
Zone restored: 9
05/22/2018 16:24 | Zone alarm restored: 9
Zone open: 7
Zone restored: 7
Partition disarmed
```
[Log_PowerSerie832_PC5015-ALARM_Zone-9-16_KeybusReader.txt](https://github.com/taligentx/dscKeybusInterface/files/2027108/Log_PowerSerie832_PC5015-ALARM_Zone-9-16_KeybusReader.txt)
[Log_PowerSerie832_PC5015-ALARM_Zone-9-11-16_Status.txt](https://github.com/taligentx/dscKeybusInterface/files/2027110/Log_PowerSerie832_PC5015-ALARM_Zone-9-11-16_Status.txt)

taligentx, 2018-05-22:
Thanks!  Zones 1-32 should report correctly now.  This is actually two different issues - the first test you showed is because the alarm was disarmed before entrance delay was over.  The second is because the 0xA5 command decoding is actually a bit simpler than I originally thought.

For anyone looking at the code, the 0xA5 status commands are separated into different groups based on byte 5, bits 0 & 1 and not related at all to the value of byte 7.

MagnusPer, 2018-05-23:
Yippee - last issue from my side solved!!

[Log_PowerSerie832_PC5015-ALARM_OnlyOneZone_Status_OK.txt](https://github.com/taligentx/dscKeybusInterface/files/2032324/Log_PowerSerie832_PC5015-ALARM_OnlyOneZone_Status_OK.txt)

Thank you so much for good cooperation and quick updates from your side!
At the moment, I have no open issues or anything to report. My feeling compared to other projects I've tested this is by far the most stable.

Information - I have notified and linked to this project on a number of different home automation sites, I hope by this that you will have more people coming in...

taligentx, 2018-05-23:
Great to hear!  Is zone 2 a delay zone?  It didn't get reported as being in alarm before the system was disarmed.

Thanks very much for your detailed feedback and logs, it's been essential as I had hit the limits of what I could test with the hardware I had available.  It will be interesting to hear if there are any stability or performance issues - so far the top priority has been decoding the Keybus protocol.

And thanks for spreading the word!

MagnusPer, 2018-05-23:
Correct - Zone 2 is a delay zone

taligentx, 2018-05-23:
Perfect, closing this one out.

closed 2018-05-23 by taligentx