taligentx/dscKeybusInterface #9: Wiring for arming and disarming alarm.

feder-ico, 2018-05-17:
Hi,
Sorry to open an issue just for a question buy I didn´t know how to contact you.
In regards of wiring... I want to arm the alarm with Home Asisstant. Do I need to make the connection with the transistor as indicated for Virtual keypad or not?

In case I need it: I see that the normal connection and the Virtual Keypad connection both uses the same green DSC wire. To use the Virtual Keypad should I replace the 10k resistor wiring with the 1k+transistor connection or should I connect both things to the same wire (leaving both connections at the same time)

Thanks in advance for your answer.

Best regards.

taligentx, 2018-05-17:
Hi @feder-ico,

Yes, to send commands to the system like arm/disarm you will need both connections:  data `dscReadPin` with two resistors for the voltage divider as well as the virtual keypad`dscWritePin` with the NPN transistor and 1k ohm base resistor (almost any NPN should do, I pulled one at random from spare parts).  They are both connected to the DSC Green wire but operate independently.

Which panel are you using with Home Assistant?

feder-ico, 2018-05-17:
Hi taligentx,
Thanks for your fast answer.
At first, I thought only reading the alarm  and show states on Home Assistant with "MQTT Alarm Control Panel" and binaries sensors, but now I changed my mind an I would like to arm or disarm the sistem it as well. 
Since I am doing the wiring now I haven´t implemented any keypad yet. Which one do you suggest?

I will try to put it to work this weekend.

Thanks again and kind regards.

taligentx, 2018-05-17:
Sorry, I'm not sure what you mean about implementing a keypad - do you already have a DSC system or will you be purchasing one?

I was curious because it'd be great to see data from more panels, especially the PC1616/PC1832/PC1864 series.

taligentx, 2018-05-17:
Ah, thanks for the clarification - the virtual keypad in this library refers to the ability to send keys to the DSC panel from the microcontroller (the keys currently available are listed in `README.md`).  You can fully control the alarm from other software connecting to this interface - it's up to the other software to tell this interface what to do.

So if there is a keypad or number/text input component for Home Assistant that can connect to this interface (over MQTT, etc), you can setup a keypad - the sketch would just need to be updated to receive the keys from the software and then write them to the security system (quite simple to do).

I haven't looked into this yet as it seems that most remote control software for security systems are only interested in arming/disarming and getting alarm and zone states, but it's certainly possible - do you know if there is a HASS component for number/text input?

Your HASS setup looks great, and it would be very useful to get logs from the 1832 - you can test the Status-MQTT-HomeAssistant sketch, though I wouldn't be surprised if the code needs to be adjusted, the PC1616/PC1832/PC1864 panels are a newer generation than the panels I've tested with so far.

If you can, please load the KeybusReader sketch (you can set `dsc.hideKeypadDigits = true;` to hide your access codes) and if you see commands being decoded in the serial monitor, save a log of the data from the following:
* Open a zone in partition 1, then close the zone
* Open a zone in partition 2, then close the zone
* Arm in stay mode, then disarm
* Arm in away mode, open a zone in partition 1 and let the alarm sound, then disarm
* Arm in away mode, open a zone in partition 2 and let the alarm sound, then disarm

Thanks!

feder-ico, 2018-05-18:
Thanks for the clarification about the Virtual Keypad. Home Assistant has a keyboard component but I think it is not useful for this since it offers just limited keys for media playing: https://www.home-assistant.io/components/keyboard/

Tomorrow I will load the status-mqtt sketch and send you some log (after calling the monitor service to prevent them phone me every time I trigger an event :) )
Bear in mind that I can arm or diarm each partition separately, so I will assume when you say "arm in stay mode" you mean any zone.

I am glad of being able to help you providing logs infomation.

Thanks and regards.

taligentx, 2018-05-18:
Sounds great.  To clarify, I'll need the logs from the KeybusReader sketch - this contains all of the Keybus data.

Thanks for the heads-up about arming the partition - I've adjusted the data needed from the KeybusReader sketch:
* Unarmed - start the sketch and let the system sit unarmed for 5 minutes (this collects data from some commands that run at 4 minute intervals).  
* Open a zone in partition 1, then close the zone
* Open a zone in partition 2, then close the zone
* Arm partition 1 in stay mode, then disarm
* Arm partition 1 in away mode, open a zone in partition 1 and let the alarm sound, then disarm
* Arm partition 2 (the mode doesn't matter), open a zone in partition 2 and let the alarm sound, then disarm

The Status-MQTT sketch will help you see if you can connect HASS through MQTT but it's likely that there will be errors in the decoding for the 1832 -  the above logs from KeybusReader will be essential to see how the panel sends its commands.

Thanks for testing this and the feedback, once I can confirm that the 1832 is decoded properly then the library should be functional enough to release in the Arduino Library Manager and Platform.io library.

feder-ico, 2018-05-21:
Hi @taligentx 
Find enclosed the logs for you to analyze. I hope you find it useful.
In Home Assisnant sensors seems to work fine but alarm states fails.

Just in case you need to know more about my system, the configuration is as follows:
- Partition 1
  - Zone 1: 2 margetics sensors for 2 doors
  - Zone 2: 2 IRP
  - Zone 3: 3 magnetics for windows
  - Zone 4: 3 magnetics for windows
  - Zone 5: 2 IRP
  - Zone 6: 2 glass break detectors
  - Zone 7: 2 panic buttons
- Partition: 2
  - Zone 8: 1 magnetic
- Partition: 3
  - Nothing

Let me know if I can provide further help.

[dsc.txt](https://github.com/taligentx/dscKeybusInterface/files/2020958/dsc.txt)

feder-ico, 2018-05-22:
Sensors are working fine but alarm states doesn´t.
I usually use only Partition 1, Partition 2 is always armed and I never use Partition 3 because I don´t have any sensor.

So, when alarm (Partition 1) is armed, "pending" payload is received on mqtt broker and after that nothing else happends and panel stays on "pending" state forever. 
You can see the log below.

2018-05-22 08:52:47 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
2018-05-22 08:52:48 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'
2018-05-22 08:52:51 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
2018-05-22 08:52:53 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'
2018-05-22 08:52:56 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
2018-05-22 08:52:59 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'
2018-05-22 08:53:01 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
2018-05-22 08:53:03 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'
**I armed here**
2018-05-22 08:53:05 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get: b'pending'
2018-05-22 08:53:09 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
2018-05-22 08:53:12 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'
2018-05-22 08:53:13 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
2018-05-22 08:53:15 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'
2018-05-22 08:53:17 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
2018-05-22 08:53:19 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'
2018-05-22 08:55:30 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
2018-05-22 08:55:31 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'
2018-05-22 08:55:34 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
**I dissarmed here**
2018-05-22 08:55:36 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'
2018-05-22 08:55:38 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
2018-05-22 08:55:40 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'
2018-05-22 08:55:41 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'1'
2018-05-22 08:55:43 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dsc/Get/Zone2: b'0'

taligentx, 2018-05-22:
I've adjusted the alarm status decoding to use two different commands to set the alarm status - when you get a chance, I'd like to see the log of the regular `Status` example sketch (from the latest commit) when partition 1 is armed/tripped/disarmed to see if this is an issue with getting the status or with sending to MQTT.

If all goes well, I should have a PC1864 panel soon to test with multiple partitions as it will require quite a bit of testing.

sj-louw, 2018-05-28:
@feder-ico

For a virtual keypad this would work nicely in Homeassistant:
https://community.home-assistant.io/t/yet-another-take-on-an-alarm-system/32386
https://github.com/gazoscalvertos/Hass-Custom-Alarm

[image]

taligentx, 2018-05-31:
@sj-louw Do you know if the keypad component can be used independently?  I've only looked through it superficially - the project looks good but seems to be a standalone security system and not designed to integrate with an existing system.

sj-louw, 2018-05-31:
@taligentx 

The keypad component seems to be integrated in the software to be used as a custom component for Homeassistant. Looking at the installation / config guide, it seems to be configureable via standard Homeassistant methods (https://github.com/gazoscalvertos/Hass-Custom-Alarm/blob/master/guidance/configuration.md), in this case also using MQTT. This made me think that if we can decode all the functions as per screenshot above (meaning read and write to/from panel), this component can be fully configured in Homeassistant to talk two way with a DSC alarm systems. Essentially making an old DSC system a networked smart alarm. :) Another bonus with homeassistant is that after its configured and functional, its very simple to enable homekit (homebridge in the background, without any extra config) support in Homeassistant. It then provides you the ability to use the alarm via Homeassistant web interface or if you prefer, via Apple Homekit.

taligentx, 2018-06-07:
@feder-ico I've pushed a new [develop branch](https://github.com/taligentx/dscKeybusInterface/tree/develop) with support for multiple partitions - let me know if this branch resolves the issue.

feder-ico, 2018-06-07:
Hi Nikhil,
I am still on leave. My trip got longer than expected and I will be flying back home on june 16, so until then I won´t be able to test.
I apologize and promise I will test that as soon as I get home and I will also send the remaining logs.

feder-ico, 2018-06-20:
Hi Nikhil,
Finally, I came back home :)
I have loaded the new sketch and now statuses seems to work ok.
The only thing I could not get to work is the one " alarm_control_panel:" per partition on Home Assistant. Only one works (the last one).

taligentx, 2018-06-25:
Hi @feder-ico - welcome back!  There was a copy/paste error in the example `configuration.yaml`, there should only be a single listing of `alarm_control_panel` with all partitions listed in the same section.  I've released a new version with the fixes (and also added trouble status to HomeAssistant).

Also, the library is now in the Arduino IDE Library Manager and PlatformIO Library Registry so future updates can be notified and installed directly from within each IDE (it may take a day or so for the new version 0.4 to be shown).

feder-ico, 2018-06-26:
Hi @taligentx , Thanks for the update. Al partitions are working fine now.

taligentx, 2018-06-26:
@feder-ico Great to hear!

closed 2018-06-26 by taligentx