taligentx/dscKeybusInterface #12: [Question] - Options to log arm/disarm action

MagnusPer, 2018-05-19:
I trying to activate/disarm the alarm from the Homey. I have followed your description in terms of hardware and software setup, but the only thing that happends when i trigg the armStay is the LCD display Ready LED is toggle on/off. (so something happens...)

I need now to debug - I would like via the KeybusReader sketch to trigger a message to be sent to the system i.e to verify the hardware connection is ok. Is there a possiblity to add such feature? or do you have another idea?

taligentx, 2018-05-20:
The KeybusReader and Status sketches already have Serial input, you can type the keys directly:

```
void loop() {

  // Reads from serial input and writes to the Keybus as a virtual keypad
  if (Serial.available() > 0 && dsc.writeReady) {
      dsc.write(Serial.read());
  }
```

KeybusReader is the one you'll want to use for debugging - you can try entering digits 0-9 and see if they are recognized correctly in the serial output.

MagnusPer, 2018-05-20:
You've thought about everything, I did not look enough on the code :-)
The wiring is working after test with diffferent commands both from Homey and diom the Keybus Reader sketch.

However I have not yet succceed to use the s and w command only the `accessCode` sems to work to both set the system in Arm Stay and Disarm. When press the Function Button A on the LCD display the text states "**Function not available**". I need to figure out if there is some sort of setting in the programming of the system allowing to use the Buttons on the LCD display.

The first 2 Arm Stay is initiated by the "s" command from Keybus Reader sketch
the last 2 Arm Stay is initiated from the LCD Display - Function button A - configured to 0x03 Stay Arm

```
 1862.67: 00100111 0 10000001 00000001 10000001 11000111 00000000 11110001 [0x27] Status lights: Ready Backlight | Partition ready | Zones 1-8 open: none
 1862.74: 00000101 0 10000001 00000001 10000001 11000111 [0x05] Status lights: Ready Backlight | Partition ready
 1868.85: 11111111 1 10101111 11111111 11111111 11111111 [Keypad] Arm stay
 1869.02: 00000101 0 10000001 10001110 10000001 11000111 [0x05] Status lights: Ready Backlight | Invalid option
 1869.16: 01111111 0 00000001 10000000 [0x7F] Beep: long beep
 1871.10: 00000101 0 10000001 00000001 10000001 11000111 [0x05] Status lights: Ready Backlight | Partition ready
 1875.50: 11111111 1 10101111 11111111 11111111 11111111 [Keypad] Arm stay
 1875.66: 00000101 0 10000001 10001110 10000001 11000111 [0x05] Status lights: Ready Backlight | Invalid option
 1875.80: 01111111 0 00000001 10000000 [0x7F] Beep: long beep
 1878.06: 00000101 0 10000001 00000001 10000001 11000111 [0x05] Status lights: Ready Backlight | Partition ready
 1886.97: 00100111 0 10000001 00000010 10000001 11000111 00001000 11111010 [0x27] Status lights: Ready Backlight | Stay/away zones open | Zones 1-8 open: 4 
 1887.06: 00000101 0 10000001 00000010 10000001 11000111 [0x05] Status lights: Ready Backlight | Stay/away zones open
 1888.73: 00100111 0 10000001 00000010 10000001 11000111 00001100 11111110 [0x27] Status lights: Ready Backlight | Stay/away zones open | Zones 1-8 open: 3 4 
 1892.15: 00100111 0 10000001 00000010 10000001 11000111 00001110 00000000 [0x27] Status lights: Ready Backlight | Stay/away zones open | Zones 1-8 open: 2 3 4 
 1893.11: 00100111 0 10000001 00000010 10000001 11000111 00001010 11111100 [0x27] Status lights: Ready Backlight | Stay/away zones open | Zones 1-8 open: 2 4 
 1893.38: 00100111 0 10000001 00000010 10000001 11000111 00000010 11110100 [0x27] Status lights: Ready Backlight | Stay/away zones open | Zones 1-8 open: 2 
 1894.20: 00100111 0 10000001 00000010 10000001 11000111 00000011 11110101 [0x27] Status lights: Ready Backlight | Stay/away zones open | Zones 1-8 open: 1 2 
 1897.37: 00100111 0 10000001 00000010 10000001 11000111 00000010 11110100 [0x27] Status lights: Ready Backlight | Stay/away zones open | Zones 1-8 open: 2 
 1897.82: 00100111 0 10000001 00000001 10000001 11000111 00000000 11110001 [0x27] Status lights: Ready Backlight | Partition ready | Zones 1-8 open: none
 1898.01: 00000101 0 10000001 00000001 10000001 11000111 [0x05] Status lights: Ready Backlight | Partition ready
 1900.05: 11111111 1 10101111 11111111 11111111 11111111 [Keypad] Arm stay
 1900.23: 00000101 0 10000001 10001110 10000001 11000111 [0x05] Status lights: Ready Backlight | Invalid option
 1900.36: 01111111 0 00000001 10000000 [0x7F] Beep: long beep
 1903.10: 00000101 0 10000001 00000001 10000001 11000111 [0x05] Status lights: Ready Backlight | Partition ready
 1909.71: 11111111 1 10101111 11111111 11111111 11111111 [Keypad] Arm stay
 1909.96: 00000101 0 10000001 10001110 10000001 11000111 [0x05] Status lights: Ready Backlight | Invalid option
 1910.10: 01111111 0 00000001 10000000 [0x7F] Beep: long beep
 1912.06: 00000101 0 10000001 00000001 10000001 11000111 [0x05] Status lights: Ready Backlight | Partition ready
 1914.70: 00100111 0 10000001 00000010 10000001 11000111 00000001 11110011 [0x27] Status lights: Ready Backlight | Stay/away zones open | Zones 1-8 open: 1 
 1914.78: 00000101 0 10000001 00000010 10000001 11000111 [0x05] Status lights: Ready Backlight | Stay/away zones open
 1915.05: 00100111 0 10000001 00000010 10000001 11000111 00000011 11110101 [0x27] Status lights: Ready Backlight | Stay/away zones open | Zones 1-8 open: 1 2 
```

MagnusPer, 2018-05-20:
Found the setting - now the system arms and disarmed nicely from the Homey and the Keybus Reader
The solution was to enable in Sequence [015] option 3 and 4 to allow the quick-exit function.

 ```
7590.09: 00100111 0 10000001 00000010 10000001 11000111 00001000 11111010 [0x27] Status lights: Ready Backlight | Stay/away zones open | Zones 1-8 open: 4 
 7590.19: 00000101 0 10000001 00000010 10000001 11000111 [0x05] Status lights: Ready Backlight | Stay/away zones open
 7593.76: 00100111 0 10000001 00000001 10000001 11000111 00000000 11110001 [0x27] Status lights: Ready Backlight | Partition ready | Zones 1-8 open: none
 7593.83: 00000101 0 10000001 00000001 10000001 11000111 [0x05] Status lights: Ready Backlight | Partition ready
 7596.66: 11111111 1 10101111 11111111 11111111 11111111 [Keypad] Arm stay
 7596.79: 00001010 0 10000000 00000001 00000000 00000000 00000000 00000000 00000000 10001011 [0x0A] Status lights: Backlight | Partition ready | Zone lights: none
 7596.84: 00000101 0 10000000 00000001 10000001 11000111 [0x05] Status lights: Backlight | Partition ready
 7596.92: 01100100 0 00000110 01101010 [0x64] Beep: 3 beeps
 7597.15: 00000101 0 10001011 00001000 10000001 11000111 [0x05] Status lights: Ready Armed Bypass Backlight | Exit delay in progress
 7597.54: 01110101 0 00000000 01110101 [0x75] Beep pattern: off
 7600.58: 00100111 0 10001011 00001000 10000001 11000111 00001000 00001010 [0x27] Status lights: Ready Armed Bypass Backlight | Exit delay in progress | Zones 1-8 open: 4 
 7605.30: 00100111 0 10001011 00001000 10000001 11000111 00001010 00001100 [0x27] Status lights: Ready Armed Bypass Backlight | Exit delay in progress | Zones 1-8 open: 2 4 
 7605.57: 00100111 0 10001011 00001000 10000001 11000111 00001011 00001101 [0x27] Status lights: Ready Armed Bypass Backlight | Exit delay in progress | Zones 1-8 open: 1 2 4 
 7606.63: 00100111 0 10001011 00001000 10000001 11000111 00000011 00000101 [0x27] Status lights: Ready Armed Bypass Backlight | Exit delay in progress | Zones 1-8 open: 1 2 
 7616.21: 00100111 0 10001011 00001000 10000001 11000111 00001011 00001101 [0x27] Status lights: Ready Armed Bypass Backlight | Exit delay in progress | Zones 1-8 open: 1 2 4 
 7616.53: 01110101 0 00000000 01110101 [0x75] Beep pattern: off
 7616.79: 00100111 0 10001011 00001000 10000001 11000111 00001010 00001100 [0x27] Status lights: Ready Armed Bypass Backlight | Exit delay in progress | Zones 1-8 open: 2 4 
 7617.01: 00100111 0 10001011 00001000 10000001 11000111 00001000 00001010 [0x27] Status lights: Ready Armed Bypass Backlight | Exit delay in progress | Zones 1-8 open: 4 
 7622.67: 00100111 0 10001011 00001000 10000001 11000111 00000000 00000010 [0x27] Status lights: Ready Armed Bypass Backlight | Exit delay in progress | Zones 1-8 open: none
 7627.16: 00100111 0 10001010 00000100 10000001 11000111 00000000 11111101 [0x27] Status lights: Armed Bypass Backlight | Armed stay | Zones 1-8 open: none
 7627.25: 01110101 0 00000000 01110101 [0x75] Beep pattern: off
 7627.48: 10100101 0 00011000 01010110 10001101 01010010 10011010 00000000 10001100 [0xA5] 05/20/2018 13:20 | Armed: stay
 7627.53: 00000101 0 10001010 00000100 10000001 11000111 [0x05] Status lights: Armed Bypass Backlight | Armed stay
 7627.53: 11111111 1 11111111 11111111 10111111 11111111 [Zone Expander] Status notification
 7627.60: 00101000 0 11111111 11111111 11111111 11111111 11111111 [0x28] Zone expander query
 7627.60: 11111111 1 01010101 01010101 01010101 01010101 01000100 [Zone Expander] Zones open: none | Zones closed: none
 7627.81: 10100101 0 00011000 01010110 10001101 01010000 10111111 11111111 10101110 [0xA5] 05/20/2018 13:20 | Armed special: quick-arm/auto-arm/keyswitch/wireless key/DLS

```

taligentx, 2018-05-20:
Great to hear that the wiring is working!  Good point on the panel configuration, I've updated the README.md with info some pertinent panel settings (like swinger shutdown).

closed 2018-05-20 by taligentx