taligentx/dscKeybusInterface #13: [Enhancement] - Arduino (esp8266) sketch for Homey platform

MagnusPer, 2018-05-20:
Now I have a fully operational sketch running with [Homey](https://www.athom.com/en/) platform.
If you want you can add this example in your code/example. [Status-Homey.zip](https://github.com/taligentx/dscKeybusInterface/files/2020565/Status-Homey.zip)

I will hang-aroud in this repository if there are any questions using this sketch.

taligentx, 2018-06-07:
@MagnusPer Do you know if Homey has the capability to support more than one partition/security system?  I haven't looked at the Homeyduino code yet - but if it does, we can update the Status-Homey example to support multiple partitions.  For now, I've hardcoded the example in the new [develop branch](https://github.com/taligentx/dscKeybusInterface/tree/develop) to partition 1 to work as before but it'd be nice to support all partitions.

MagnusPer, 2018-06-08:
I don't think there is an stright forward solution for that using homey-arduino-library. Also there is bugg reported [Using the same capability more than once] https://github.com/athombv/homey-arduino-library/issues/41

Alternative solution is to use MQTT to Homey

taligentx, 2018-07-09:
@MagnusPer Can you verify if there are any issues with the current version of the sketch?  I'd like to verify if there are any bugs since adjusting the sketch to use the library methods that were introduced to support multiple partitions.

MagnusPer, 2018-07-21:
@taligentx 
Excuse me for the delay to verify version 1.0 of Homey sketch, I have been on vacation.
I have now verified most of the functionality - no problem at all, everything works like before :-)

The only problem I have is that I lose the connection to the router that has to be restarted ~ every other week. I have read others who have the same problem and it seems to be a known bug related to the esp8286 platform.

MagnusPer, 2018-08-20:
Feedback related to my connection problems to the router (ASUS RT-AC66U). For a month I have used static IP address and it has been stable since then.  I have not investigate why, but my feeling is it may concern the lease time.

taligentx, 2018-08-23:
Great to hear you found the issue - I've been testing with an older Asus RT-N56U and haven't seen an issue with DHCP lease renewal but I'll add a note to recommend using a static IP.  In small networks, static IPs make sense for 24/7 devices.

closed 2020-11-02 by taligentx