taligentx/dscKeybusInterface #16: Homeassistant sketch not posting status back to MQTT topics

sj-louw, 2018-05-30:
For some reason I am able to arm/disarm (write to panel) via Homeassistant and Homebridge, but nothing seems to be coming into the `dsc/Get/ZoneX/#` mqtt topics (reading from panel then posting to these zone topics), for zone status... Do you maybe have any idea...?

Further testing seems that the Arduino reads the incoming messages in the `dsc/Set` topic fine. It then successfully executes the panel "A", "S" and "D" commands, but never posts the status back to the `dsc/Get` topic so that armed status can be read by Homeassistant/Homebridge. The same goes for the `dsc/Get/ZoneX` topics. The Arduino does not publish zone updates there. I am testing with the sketch in pull request: https://github.com/taligentx/dscKeybusInterface/pull/15

taligentx, 2018-05-30:
Let's narrow this down to see if it's a decoding issue or MQTT issue.  If you can, load the Status sketch and check the output over serial - are the zone states and armed/alarm status reported correctly?

If so, you can check the MQTT communication - I've used [MQTTBox](http://workswithweb.com/mqttbox.html) to simultaneously view all of the different topics and see what is happening.

Also, thanks for the PR!  I'm catching up after holidays and will add the example - I may need to reorganize the examples to cover Arduino and esp8266 separately.

sj-louw, 2018-05-31:
Hi.

I have loaded the Status sketch and tried again, but nothing is printed to the serial terminal. I went back, disconnected my circuit and verified everything. Even made sure the transistor is still working... Please see the pics attached of my wiring. Is it possible for you to verify that my wiring is correct?

1. The panel side, full circuit not drawn in:
[image]

2. My circuit, its soldered on a PCB, reason why it looks weird :)
[image]

3. My understanding of the circuit shown in the codes' README:
[image]

I will continue to fiddle around, but I guess the wiring needs to be in order before anything will work correctly. After the wiring is confirmed, I will continue with testing MQTT, etc.

taligentx, 2018-05-31:
Thanks for the detailed info!  It's very helpful - check the DSC green/data line in your second image.  If the right side connections go to the Arduino, your image shows the 10k resistor connected to the panel/15k/ground.  It should be connected to the Arduino/15k/ground instead.

It may also help troubleshooting to temporarily remove the transistor and the connection to the Arduino write pin - you only need the `dscClockPin` and `dscReadPin` connected to read from the panel.

sj-louw, 2018-06-01:
Hi, I have modified my PCB as you mentioned, but still no go. I have from scratch build a new circuit, also, but as soon as I connect the "Write" wire to the panel, the keypad beeps and loose connection to the panel. Without the "Write" wire connected, it reads and prints to serial fine from the panel. The transistor Im using is 2N3904.

Is it not possible to reuse a pre-made and working circuit from a previous project (https://github.com/denvera/dscmod), as per screenshot below: (With that I could read and write to the panel)
[image]

taligentx, 2018-06-01:
Good to hear that the clock and read lines are working on their own - are you seeing any further invalid data (`CRC Error`, `0x00` commands, etc)?

If all of the data is valid with the write pin disconnected, verify the wiring matches this for the 2N3904:
[image]

If you're willing to troubleshoot the issue, you can wire the writing connections directly without the PCB to ensure there isn't some other interaction involved.  In the past I've plugged the base resistor directly into the Arduino, soldered the resistor directly to the transistor,  and soldered the ground and data wires directly to the transistor hanging in the air.

You can also check voltages to make sure there isn't some problem with the components themselves:
Arduino dscWritePin to ground: 0v
2N3904 pin 3 to ground: 5v-13.6v (depends on how the multimeter reads the signal, one of mine reads ~8.5v)

I've considered using a level shifter - it is possible, but would at the very least require altering the code to invert the write signal.  dscWritePin is normally low and goes high to write, which turns on the transistor and brings the data line low.

Reducing down to two pins is also possible with some rewriting.  I'll add it to the todo list - at this point, the higher priorities are protocol decoding and integration with home automation.  I like the idea of reducing the number of pins used, but I also like that the current implementation uses very common parts that anyone can find, even salvaging from scrap equipment.

sj-louw, 2018-06-02:
Hi, thanks for helping to troubleshoot this. I have verified the transistor connections, and it is indeed correct. When reading, there is no more CRC errors. The last attempt from my side was a brand new circuit with new parts, without a PCB, Just loose wires hanging all over the show :)

Regarding to make use on one single PIN, I do understand your point, and I agree with you to keep it at two PINS and using easy to find parts.

On a side note, I think I might be on to something. It seems like when you use an Arduino ethernet shield the interrupt PIN is disconnected. Here is an explanation I found of what needs to be done to make it work. (http://garagebox.org/arduino/342-complete-soft-of-explaination-on-interrupt-imr-usage-of-ethernet-shield.html) Then also what seems to be important, is to make sure the sketch is written according to the "rules to write a sketch in IMR function", at the bottom of that post. Can you verify that your code follows that rules?

Then the sketch must be written as ISR function example: `void ISR_function() { ..................... }`, whatever this means... :)

So in short, it seems there must be two things in place for the Ethernet shield if your code depends on using hardware interrupts:
1. The hardware mod on the ethernet shield
2. The sketch must be written to make use of the hardware interrupts (`#include <utility/w5100.h>`)

taligentx, 2018-06-02:
I'd like to narrow this down to see if there's a conflict with the W5100:

* Which Arduino board are you using?
* Which pins are you using?
* Are the voltages I listed earlier correct when using the Arduino board by itself with no other add ons and only USB, ground, and pins 3,4,5 connected?  You could also try removing the USB and powering the Arduino directly with the panel in case there's some strange ground loop issue.

I suspect that when you are connecting the write pin, the transistor is turning on and pulling the data line low for some reason. Checking the voltages would help determine if this is the case.

The W5100 shield uses SPI alone by default and doesn't use the interrupt mode unless you've configured the board for it.  I would start with SPI-only without the interrupt mode to get the basic connectivity working.  I've previously successfully tested the Uno with the ENC26J60 ethernet module over SPI-only and the UIPEthernet library - the MQTT examples worked well and didn't show any problems over the few days I had it setup.

One possible conflict I saw is that the SD card on the shield uses pin 4 - the documentation states that pin 4 should be set high to disable the SD card on the shield.  If so, you would need to use a different pin for dscReadPin.

sj-louw, 2018-06-02:
@taligentx 
- Which Arduino board are you using? Arduino UNO clone
- Which pins are you using? 3 (CLK), 4 (Read), 5 (Write)
- Tonight when I actually wanted to test the voltages, both my Arduino's decided they want to die :( I cannot communicate with them, trying to upload new sketches or even trying to re-burn the bootloader. So Im sort of stuck at this point :( (I have ordered some NodeMCU modules that should arrive in the week)

I am very keen to get this setup working, so I will be continuing when my ESPs arrive.

sj-louw, 2018-06-03:
@taligentx 

I have build a third circuit today, with brand new parts and Im getting the same result, as soon as I connect the custom circuit (No Arduino connected), the existing keypad looses connection.

The voltage readings before the circuit connected between panel ground and various panel pins:
 - Green: 8.6v
 - Yellow: 6.7v
 - Red: 13.6v
 - Black: 0v

The voltage readings through the custom circuit (dscKeybusInterface ref) between its ground and its various pins (While connected to the panel):
 - "Arduino PIN 3" (Yellow/CLK): 0.5v
 - "Arduino PIN 4" (Green/Read): 1.7v
 - "Arduino PIN 5" (Green/Write): 0.5v

Note that no Arduino or ESP was connected to the above circuit during testing and the main keypad still lost connection and started to behave erratically.

For reference, here is a photo of my latest attempt to the circuit:
[image]

taligentx, 2018-06-03:
I'll reproduce the setup without having an Arduino connected and see what happens here.  You should connect the base resistor to ground (the write pin) - otherwise the transistor base pin is floating and would have undefined behavior.

sj-louw, 2018-06-04:
Thanks, In the meantime, you are saying I must "connect the base resistor to ground (the write pin) - otherwise the transistor base pin is floating". Does that mean that, if you look at my diagram above, I am missing a pull down to ground next to the 1K resistor? If that is the case, on what side of the 1K must it be in the above diagram? I could also not see such a pull down on the base. Im confused.

taligentx, 2018-06-07:
@sj-louw I've been testing BC547, BC549, and a few random NPN transistors connected to the PC1555MX and PC1864 to attempt to get the Keybus to react (even connecting the collector and emitter backwards) and haven't been able to replicate this - very strange.

To debug this, I recommend starting from single components (no circuit board) step by step and see exactly which step/component causes the problem - check the Keybus after every step and let's see where it is failing.

Panel connected to AC power, battery, and keypad only (no Arduino/esp8266):
1. Connect NPN collector to DSC green - check keys
2. Connect NPN emitter to DSC Aux (-) - check keybus (and so on for each following step)
3. Connect NPN base to 1k resistor
4. Connect other end of 1k resistor to DSC Aux (-)
5. Connect a 15k resistor to DSC green
6. Connect the other end of the DSC green 15k resistor to a 10k resistor
7. Connect the other end of the DSC green 10k resistor to DSC Aux (-)
8. Connect a second 15k resistor to DSC yellow
9. Connect the other end of the DSC yellow 15k resistor to a second 10k resistor
10. Connect the other end of the DSC yellow 10k resistor to DSC Aux (-)

Hopefully this will narrow down the problem!

rogueturnip, 2018-06-07:
Just a thought, if anyone out there has experience with KiCad, Eagle or some other circuit design program, it would be very beneficial to build a circuit and gerber that people could send off to some place like Seedstudio to get built.  The ultimate would be with the design to have a socket to plug the MCU into.

sj-louw, 2018-06-07:
@taligentx Thank you very much for the detailed info and effort in trying to debug. I will try again over the weekend and report back.

@rogueturnip I agree. We need to get some sort of Schematic in Gerber format for people (including myself :) ) to send off for manufacturing or to print themselves. This is one of the first things I would like to do, as soon as I can figure out the exact schematic, working with my 1864 panel.

PascalSI, 2018-06-08:
I can easily make a scheme in Gerber format without any problems)))

sj-louw, 2018-06-08:
@@taligentx I have done exactly step 1 - 10 as you described, and it did not interfere with the main keypad. I went ahead and connected it up to my nodeMCU, with the homeassistant sketch and the diagram now as seen below: Can you please verify where the part I marked as red must go?

Reading from panel now definitely works, as I see the Zone messages coming into MQTT and Homeassistant updates the separate zone statuses accordingly. If I try to arm the alarm from Homeassistant, I can see the MQTT messages going into the dsc/Set topic, but the panel does not react on it.

Note: Last night I've tested with one of my previous versions of the circuit and could again only get writing to the panel working - no reading, so at least we know the MQTT setup is working as expected and can probably focus on the electronics to get both reading and writing at the same time going. Thanks

[image]

@PascalSI That would be great if you can make a Gerber schematic with a verified version of the circuit. It would be great if this can be designed for a single-sided PCB, so that people have the option to print it themselves too. Thanks.

taligentx, 2018-06-08:
@sj-louw Great to hear!  What is different now compared to last time when you just had a circuit attached and no Arduino where you saw the keypad have problems?

You can test the following (a few extra steps as a precaution):

1. Load the KeybusReader sketch to the NodeMCU
2. Disconnect the NodeMCU from power/USB
3. Disconnect the panel battery
4. Disconnect the panel power so it is powered down
5. Disconnect the end of the 1k resistor from Aux (-)
6. Connect the end of the 1k resistor to NodeMCU D8 (so D8 should only connect to the 1k resistor)
5. Connect the NodeMCU to USB, open serial monitor and verify the interface shows as online.
6. Connect the panel power and battery and see data shows up in KeybusReader without CRC errors or invalid commands like 0x00, etc.
7. Type a few different digits into serial monitor and see if they are recognized in the KeybusReader data

If this works, try Status-MQTT-HomeAssistant and let's see what happens.  I know the step-by-step seems a bit pedantic but remote debugging is always an interesting challenge and I'm interested in seeing if there's any kind of design bug.

sj-louw, 2018-06-08:
@taligentx Hi. I must admit, I took a bit of a shortcut with the last instructions. I unplugged the nodeMCU from the USB power and then only removed the 1K resistor from ground, leaving D8 connected to the 1K. Then I put the usb power back. Everything worked after that - read and write to the panel, and my main keypad is still functioning. Essentially the exact diagram you had in the README.

I have used different 10K resistors, although I have tested the previous ones. My solder joints was twisted together before I soldered with this last approach. These was the only differences.

Anyway, below is the exact working diagram and setup I have running now, with read and write to the panel working, using the Homeassistant sketch on the master branch. I have also set the `MQTT_MAX_PACKET_SIZE` in the `PubSubClient` Library from the default `128` to `256`, as per https://github.com/knolleary/pubsubclient

[image]

Thanks again for your help.

taligentx, 2018-06-08:
@sj-louw That's great to hear!  For HomeAssistant, note that the default topic for alarm states will be changing in the next version from `dsc/Get` to `dsc/Get/Partition1` (and with separate topics per partition up to `dsc/Get/Partition8`).  Just a heads-up if you're on the master branch now and update in the future - I'll be merging develop to master once I get some feedback on how develop is working out.

closed 2018-06-08 by taligentx

bjrolfe, 2018-12-21:
Wanted to add to this since I was having similar issues, no panel status but could send commands.

TL;DR: I have a batch of 2N3904s that are CBE and not EBC (wtf!)

Symptoms:
I was able to send commands (arm, disarm) but they took upwards of 5 seconds. Also, wasn't getting zone updates. When I armed it would go from pending to disarmed (instead of armed).

Troubleshooting:
- Disconnected the DSC green from the "collector" and I now have zone updates.
- While disconnected, check voltage drop from DSC green to ground (around 8-9 V because of the signal, good).
- Reconnect and check the voltage drop from ground to green: 4-5V; that's not good, that means my read pin was only seeing 1/3 of that voltage, not enough to reliably make HIGHs.
- Swapped the 2N3904 direction and voltage drop was correct (8-9V).
- Also swapped in a 2N4400 (EBC) and worked.
- Dropped the 2N3904 into a breadboard with a 12V power supply, LED, and 1k resistor and sure enough  the LED lit one way and not the other (because it was above the ~6V EC breakdown voltage and conducted).

I've now labelled that baggie of 2N3904 as CBE!

[image]
[image]