taligentx/dscKeybusInterface #29: QR Code

SMPSMagentics, 2018-07-08:
Hi Thanks for all your work. I am not been successful cloning the QR code, I get an error message saying "Not Valid Blynk Code"  when I download or try from screen the following file you provided https://user-images.githubusercontent.com/12835671/42364287-41ca6662-80c0-11e8-85e7-d579b542568d.png
Thanks for your help.

taligentx, 2018-07-08:
Hi @SMPSMagentics - thanks for the feedback, I was able to replicate this for the 64-zone QR code and have removed that QR code.  It contained a hard link to an internal server IP address.

However, I'm not seeing the same issue with the 16-zone and 32-zone QR codes.  I've tested using a fresh local Blynk server (server-0.38.5.jar) and a fresh phone.  Which version of the Blynk server and app are you using?

SMPSMagentics, 2018-07-09:
Hi taligentx, I have the same server. I think the issue was that the resolution of the monitor was not good, once I pulled it on my laptop screen I was able to read the QR from the the screen. I am now about to start hardware testing. Thanks A Lot.

taligentx, 2018-07-09:
Good to hear you were able to retrieve the Blynk layout!  If you run across any problems feel free to open a new issue.

Also, I'm looking for feedback on the Blynk sketch and app - my focus was on setting up a basic example of how to use a Blynk project with this library.  If you create a different Blynk layout feel free to post the QR code and I'll create a collection of QR codes for users to choose.