taligentx/dscKeybusInterface #34: Next/Prev button

l33d, 2018-08-21:
I see another problem, where can i find next prev button on keyboard? I can`t scroll alarm from menu eg. *3

taligentx, 2018-08-23:
Hi @l33d, I haven't decoded the left/right arrow data because the PC5500 keypad I'm using does not send separate data for left/right.  I suspect that the PC5501 series keypads do send different data for each direction - which keypad are you using?

l33d, 2018-08-23:
Hi, 
Yest i use this keyboard LCD5500 ( have 2 pcs ). I can help you to decode this data, but i don`t know how to do this.

taligentx, 2018-08-23:
You can load the `KeybusReader` example sketch and open the serial output to view the data from the DSC panel.  You can press a few numeric keys to make sure the keypad data is being displayed, then press the left and right arrow keys a few times to capture the data for the keys.

On the PK5500, pressing the left and right arrow sends the same data so hopefully the LCD5500 is different.

l33d, 2018-08-23:
Ok, i try and send you captured data.

l33d, 2018-08-23:
I can`t upload sketch to nano ( to small) arduino uno too ( error ) i upload it to esp, and  this is output
Left/right arrow is this same data i don`t now what is unrecognized key, when pressing maybe menu *3. 

 115.65: 11111111 1 11110111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: Left/right arrow //press left
  116.02: 11111111 1 11110111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: Left/right arrow //press right
  116.11: 11111111 1 01110101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: Unrecognized key: 0x75
  116.39: 11111111 1 01110101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: Unrecognized key: 0x75

Do you need data for keys like, chime, reset, exit and programable keys?
And could you add qr code form 64 zones? It have only more leds? ( i ad this leds in blynk with 32 zones and works)

Last question is posiblle to use lcd in blynk sketch to view all inputs like orginal lcd5500/pk5500?

taligentx, 2018-08-24:
Thanks for the data, it's the same result as on the PK5500 so for now there isn't a way to distinguish between the arrow keys and send each one.  As I mentioned, I suspect that the PK5501 (the non-alphanumeric LED keypad) uses separate left/right key press commands but I haven't seen captured data for that keypad yet.

The rest of the keys you mentioned (chime, reset, etc) have already been decoded and are listed in [README.md](https://github.com/taligentx/dscKeybusInterface/blob/master/README.md).

> And could you add qr code form 64 zones? It have only more leds? ( i ad this leds in blynk with 32 zones and works)

I originally had a 64 zone Blynk project, but the project size is too large to share via QR code - the generated QR code only had a link directly to my personal Blynk server instead of embedding the project layout directly in the QR code itself.

> Last question is posiblle to use lcd in blynk sketch to view all inputs like orginal lcd5500/pk5500?

I'm not clear on what you mean, which information are you wanting to display?

l33d, 2018-08-24:
Hi,

> Thanks for the data, it's the same result as on the PK5500 so for now there isn't a way to distinguish between the arrow keys and send each one. As I mentioned, I suspect that the PK5501 (the [...]
The rest of the keys you mentioned (chime, reset, etc) have already been decoded and are listed in README.md.

Ok thanks for the info,

> I originally had a 64 zone Blynk project, but the project size is too large to share via QR code - the generated QR code only had a link directly to my personal Blynk server instead of embedding the project layout directly in the QR code itself.

Ok, understand.  This code has only next leds for 32-64 zones?

> I'm not clear on what you mean, which information are you wanting to display?

I don`t know how to scroll from which zone was alarrm when i go to *3 menu or how to change a time.

taligentx, 2018-08-25:
>Ok, understand. This code has only next leds for 32-64 zones?

Yep, the only difference from the 32 zone project is adding the extra LEDs, you can use the cloning feature in Blynk to add the extra zones based on the existing LEDs and change the virtual pin to the new zone.

> I don`t know how to scroll from which zone was alarrm when i go to *3 menu or how to change a time.

I'll look into making the alarm memory information available to the Blynk project.  You can change the time through Blynk, you just won't see the feedback in the project's LCD.

The Blynk virtual keypad behaves like the keypads without an LCD/LED display because some of the logic like scrolling and using a cursor to change the time are handled by the LCD keypad itself and has nothing to do with the DSC panel.

closed 2019-05-10 by taligentx