taligentx/dscKeybusInterface #38: DSC PC 1808 Unrecognized data

Tiaanh, 2018-08-23:
Hallo,

I've tested on a PC 1808 but get "Unrecognized data". Any pointers where I should start to debug ?

  298.46: 00000000 0 [0x00] Unrecognized data[CRC OK]
  298.69: 00000000 0 00000000 00000000 00000000 00000000 [0x00] Unrecognized data[CRC OK]
  298.73: 00000000 [0x00] Unrecognized data[CRC OK]
  298.74: 00000000 0 [0x00] Unrecognized data[CRC OK]
  298.76: 00000000 0 00000000 [0x00] Unrecognized data[CRC OK]
  298.80: 00000000 0 [0x00] Unrecognized data[CRC OK]
  298.88: 00000000 0 [0x00] Unrecognized data[CRC OK]
  298.96: 00000000 0 [0x00] Unrecognized data[CRC OK]
  299.04: 00000000 0 [0x00] Unrecognized data[CRC OK]

Thanks

taligentx, 2018-08-24:
Hi @Tiaanh, 

Which microcontroller are you using, and how is it connected?

l33d, 2018-08-24:
@Tiaanh

Hi, 

I had same problem. Bad connection from alarm or arduino. Please follow the schematic diagram.

ColinNg, 2018-08-25:
One of Nikhil's suggestions - the WeMos D1 mini - is amazing. Here's mine, with a Power shield so I can take power straight from AUX+/AUX-. I unsoldered the barrel jack and added in Resistors and used a 2N7000 FET (it does the same as the NPN transistor - but it uses 1 less resistor). Total cost is < $5 *shipped*. 

[image]

taligentx, 2018-08-25:
>I have two jumper cables linked to extent the length. The green jumper from the Uno connect to the black cable and in turn connected to the "DSC Green" connection. The yellow jumper from the Uno connect to the white cable and in turn connect to the "DSC Yellow" connection. The purple cable on the DSC Aux(-) connect to the Unu's ground via the bread board.

@Tiaanh I haven't tested the interface using jumpers and breadboards, it's possible that the multiple connection interfaces are causing an issue.  I recommend directly soldering the wiring and resistors, it's few enough components that I don't even use a perfboard and simply solder the components inline.

Tiaanh, 2018-08-25:
Soldering and connecting without breadboard solved the issue. Thank you very much for the help.

closed 2018-08-25 by Tiaanh