taligentx/dscKeybusInterface #39: KEYBUS malfunction

l33d, 2018-08-24:
Hi,

Second time i have info od my system about KEYBUS malfunction, before this text i see trouble light and after check the problem in menu *3 i see an "battery error 520x ( additional power supply?)" I don`t have an additional power supply from dsc in my alarm. Tommorow i check on dls all the problems and paste it. This is second time when i see this text, and keybus error message, after connect virtualblynk.

ColinNg, 2018-08-25:
If you run the KeybusReader - it will output almost everything it sees to the Serial console. This might be helpful in troubleshooting. If you need to interact with the system, you can send regular keypad commands via the Serial console too.

l33d, 2018-08-25:
Hi,

I think I've found a problem. By disconnecting and connecting the GND from the ESP that was connected to the KEYBUS, this problem reappeared. I need to connect VCC and GND permanently and see if the problem appears again. If it appears, I will install a sketch keybusreader on second esp and check what it will display. Is the ESP module supervised in the system? I looked in DLS and in the keyboard LCD5500, but did not see an additional module except my 4xpc 5108 1x5208 and 2xLCD5500. Why i see when error occurs battery error in pc520x (1,2,3,4) like on the screen, which i don`t have? 
![error dsc] 

(https://user-images.githubusercontent.com/42523320/44618126-4ad3d400-a870-11e8-9a0d-b4a8bdda1629.PNG)

taligentx, 2018-08-25:
I'm updating the todo list to note that connections should be soldered and hardwired.  The Keybus data isn't particularly high speed (1Kbps) but any kind of intermittent connectivity can cause problems on the entire Keybus.

For now the interface is not registered as a separate module and will not be visible in DLS.  In theory it is possible to have the interface check for an available slot and register as a keypad - is there any useful function for doing this other than being visible in DLS?

l33d, 2018-08-26:
The problem did not return, the cause was probably a loose ground wire.

closed 2018-08-26 by l33d