taligentx/dscKeybusInterface #40: Pushbullet esp8266

l33d, 2018-08-26:
Hi @taligentx ,

I upload sketch with pushbullet (change only messages to my language). It works, but i need a little more notification to add ( alarm armed, alarm disarmed, arm AWAY, arm STAY, sabotage zones. I try to add this messages, but did not work. I do not know how to add this notification properly. I try only copy paste from yours sketch and change values.

I add it like this but no working, sketch comile without problem

       if (dsc.armed[partition]) {
        dsc.armed[partition] = false;  // 

        char pushMessage[42] = "Armed partycja ";
        char partitionNumber[2];
        itoa(partition + 1, partitionNumber, 10);
        strcat(pushMessage, partitionNumber);

        if (dsc.armedChanged[partition]) sendPush(pushMessage);
        else sendPush("Dissarmed");
       }

        if (dsc.armedAway[partition]) {
        dsc.armedAway[partition] = false;  // 

        char pushMessage[44] = "AWAY ";
        char partitionNumber[2];
        itoa(partition + 1, partitionNumber, 10);
        strcat(pushMessage, partitionNumber);

        if (dsc.armedStay[partition]) sendPush(pushMessage);
        else sendPush("STAY");
       }

The next question is possible add a notification of 24 hours waterflow line, 24 hours gas line, etc.?

taligentx, 2019-05-21:
Hi @l33d - I've updated the Pushbullet example with armed status and zone alarms.  

> The next question is possible add a notification of 24 hours waterflow line, 24 hours gas line, etc.?

What are you seeing when one of these zones is violated?  I checked a 24 freeze zone and it was reported as open/closed as well as the zone alarm triggered and restored correctly.

> When i lost AC power and get it again, i don`t see pushbullet notification? 

The default DSC panel configuration has a 30 minute delay before the AC power notification is sent, you can disable or adjust this delay in the DSC panel *8 configuration (see the README.md).

l33d, 2019-05-21:
Hi @taligentx cool to see you again :)
I try new version, thanks for update.

> What are you seeing when one of these zones is violated? I checked a 24 freeze zone and it was reported as open/closed as well as the zone alarm triggered and restored correctly.

I need only different information about violated zones like 24 hours waterflow line, 24 hours gas line.

> The default DSC panel configuration has a 30 minute delay before the AC power notification is sent, you can disable or adjust this delay in the DSC panel *8 configuration (see the README.md).

I didn`t know about this time ;) Thanks for the info.

taligentx, 2019-05-21:
>  I need only different information about violated zones like 24 hours waterflow line, 24 hours gas line.

These should work like any other zone, I checked with the 24 Freeze zone and it reported correctly.  Let me know if you're seeing something different.

l33d, 2019-05-22:
Ok, i test pushbullet today evening or tomorrow. Now i testing virtualBlynk and have question about some functions.

l33d, 2019-05-25:
I have some problem when translate this sketch and when compile i have info
 

> exit status 1
> initializer-string for array of chars is too long [-fpermissive]

Old sketch 1.0 i translate without any problem.

taligentx, 2019-05-25:
> char armedState[24] = "Włączenie DOMOWE: partycja ";

You'll need to increase the size of the `armedState[]` array, you have more characters than the original text.  It should be the total number of characters in the final message + 1 for a null character:

https://www.arduino.cc/reference/en/language/variables/data-types/string/

l33d, 2019-05-26:
Ok, thanks, all works, test all notofication later. I can add  Polish translate to yours sketch if you need.

taligentx, 2019-06-13:
> I can add Polish translate to yours sketch if you need.

Great offer, thanks!  Translations are important, I'll think about how to handle them - ideally the text would be broken out to a separate file and the sketch just specifies which language to use.  Added to the todo list.

closed 2019-06-13 by taligentx