taligentx/dscKeybusInterface #42: Request: pereodical status update for Home Assistant

barsmedia, 2018-08-27:
Hello!
It's more feature request: could HASS sketch be modified to send pereodical status update to MQTT? Cause you will loose control from "alarm control panel" if HASS will be restarted after power failure (so no arm/disarm buttons).

P.S. 1:
Confirming work on PC1565

P.S. 2:
Also for Wemos D1 R1 (!) actual pins are:
D3 (5), D4 (4), D10 (15)
and board can be powered directly from AUX via Vin

P.S. 3:
what is meaning of 1kOhm resistor in connection scheme to transistor? It's also working without it (accidentally forgotten to add it when was rewiring)

sebthauvette, 2018-11-14:
For you P.S. 3, the resistor at the base of the transistor is used to limit the current that will go through the transitor. There is a maximum amount of current that can safely be drawn from each pin of you microcontroller. Without any resistor, your microcontroller will send the maximum amount of current it can and will eventually be damaged because of this. You also waste a lot of power.

To optimal value for the resitor will depend on the transistor you use. You need enough current to put the transistor in saturation mode. Anything more is wasted and too much can damage your microcontroller.

taligentx, 2019-05-21:
Hi @barsmedia,

> It's more feature request: could HASS sketch be modified to send pereodical status update to MQTT? 

This is added in the current release 1.2, thanks to PR #63 - thanks for the confirmation for the PC1565!  Added to the list of verified panels.

Hi @shafr,
>I think someone should sit down and merge this project with Tasmota project - there would be less code duplication. It would be a matter of describing new device and putting pins & protocol there.

To clarify, which parts of the code are being duplicated?

shafr, 2019-05-22:
Well, plenty of:
* OTA updates
* Secure connection to Mqtt topics
* Home-assistant discovery
* Security Fixes ( if any would be needed)
* Telemetry
* Additional Pins for extra features

bennydiamond, 2019-05-22:
I think @barsmedia was referencing periodical status refresh of alarm system elements like zone status, fire/smoke, etc. Not MQTT device availability.

I've done such thing in a modified sketch for Uno. Should be easy to reproduce across all platforms.
[Check for "refreshAlarmSysStatus()" function at line 615](https://bitbucket.org/psyko_chewbacca/unodsckeybusinterfacemqtt/src/master/src/main.cpp#615)

My stuff is a bit overcomplicated as I need to space out inbetween my message publishing because the UIPEthernet library I'm using is a bit touchy when pumping out packets like crazy. Shouldn't be an issue when using the standard Ethernet lib tho.

barsmedia, 2019-05-22:
@taligentx Exactly, I was referencing to periodical status updates and not availability, like mentioned @bennydiamond

taligentx, 2019-05-24:
@barsmedia - The MQTT status messages are published with the `retain` flag so if Home Assistant is restarted, it should receive the last known state of every component from the MQTT broker.

I do see an issue if you are using Home Assistant itself as the MQTT broker or if a separate MQTT broker itself is restarted.  I'll look into forcing re-publishing all status messages if the connection to the MQTT broker is disrupted. Added to the todo list.

This would ensure that re-publishing the status states would be event-driven and only occur as needed. As far as a periodic re-publishing, I'm not a fan of polling-type setups unless absolutely necessary - I'd rather see HA get the correct status immediately.  However you could add something like the Timer library and set something up in the sketch if desired.  @bennydiamond good to see you were able to able to make this work for your purpose!

@shafr 
> OTA updates
Secure connection to Mqtt topics
Home-assistant discovery
Security Fixes ( if any would be needed)
Telemetry
Additional Pins for extra features

This would make sense if you're only looking to replace the front-end example sketches and implementations.  However at a practical level, the core code handling the Keybus data capture, buffering, and processing uses significant resources on the esp8266 and trying to shoehorn it into a larger project may result in resource contention.  A straight merge to Tasmota/ESP Home would also require dropping Arduino AVR microcontrollers - they are quite limited but still useful and widespread.

This is also a design preference - the Arduino environment is nicely setup to allow users to mix and match different components through libraries to suit individual needs (akin to the UNIX philosophy of composability).  For example:  if you prefer having OTA, it's a simple matter to drop in the ArduinoOTA library; if you'd like secure MQTT, it's just a matter of swapping out WiFiClient for WiFiClientSecure, etc.

Of course the nice part about open source is that the code is available for interested parties to implement integrations to Tasmota, ESP Home, etc.

barsmedia, 2019-05-24:
@taligentx retain is not solution for this. For example, if everything was okay, you had disarmed alarm and then power loss, and, f.e. MQTT server was shutdown. If you put your alarm by code on alarm panel, and after that power will be back, MQTT retain message will be "disarmed", and won't reflect true situation.

taligentx, 2019-05-24:
@barsmedia Agreed, see my previous reply.

taligentx, 2019-05-24:
Added `getStatus()` to the develop branch and updated the examples - for Homebridge and Home Assistant, the example sketches show how to trigger an update if the connection to the MQTT broker is lost and reconnected.

@barsmedia Test this out if you get a chance, I'm interested in seeing if there are other situations where an update would need to be forced.

barsmedia, 2019-05-29:
@taligentx sorry for delay, just uploaded updated lib and code, for now everything looks good. After several mqtt restarts no more "unavailable". Will check with armed state bit later.
Maybe it's good idea to add mqtt autodescovery feature also?

taligentx, 2019-05-31:
Good to hear, thanks for the update.  As far as HASS auto-discovery, it is possible but I don't think it's a good fit for this library.  This library is not currently setup to change the number of partitions and zones in the sketch, so the interface would publish all 8 partitions and 64 zones to HASS.  Also, the zones would not have any customized information for the binary sensor component (door, window, etc) so setting the configuration directly in HASS would give better representation of the security system.

That said if a PR comes in with auto discovery I'll merge it to the examples.

closed 2019-05-31 by taligentx