taligentx/dscKeybusInterface #45: Arduino sketch - add time stamp

l33d, 2018-09-03:
Hi,

Is possible to add time stamp from dsc to view it on serial port?

mzac, 2018-10-08:
You could probably include a NTP library in your sketch and print the time when an event is triggered

https://github.com/arduino-libraries/NTPClient

sebthauvette, 2018-11-12:
Do you need the actual time or only something relative to when the arduino was booted ? If you need the actual time, NTP would probably be need as mzac suggested. This means the arduino needs to be connected to the internet. 

If you only need something relative to when the arduino was booted, it would be much more simple to implement using millis(). 

If you need to log the events on an other computer, you could simply sent the events as they happen and timestamp them when they are received by the logging computer.

l33d, 2018-11-12:
Thanks for the info, i add time via php in script.

closed 2019-05-21 by taligentx