taligentx/dscKeybusInterface #47: question: does adding virtual keyboard require programming a unit?

witek1308, 2018-09-17:
Hi,
I've found this subject and liked it so much that already soldered my PCB and waiting for NodeMCU to arrive. In the meantime I want to clear all potential questions re wiring and config.

My understanding is that there is no need to configure anything to add this module (section 377 sounds like optional to me). But from what I remember adding a physical keyboard required adding it in central unit. Is this something I need to do with virtual keyboard somehow or not at all? I have PC1616.

jwolansk, 2018-09-17:
Hi, no configuration is needed on DSC. If I remember correctly, the keypad needs to be assigned to a partition so the configuration is needed by the keypad itself. The device you are adding here has access to all partitions hence no additional configuration is needed.
I also have it connected on PC1616.

witek1308, 2018-09-17:
Thanks for your response. This seems to be almost plug & play :) (despite of arduino/esp8266 programming)

witek1308, 2018-09-19:
@jwolansk in which setup you use it with your PC1616? is it MQTT or else? I'm trying to make it work and read works all fine but I can't write (on the HomeAssistant-MQTT sketch at the moment). Not sure if this is somehting with electronics or sketch itself. Can you point me to working sketch or guide somehow?

jwolansk, 2018-09-19:
To see if your output line is working, you can disconnect the board from DSC and plug out the Nodemcu, then connect power supply to where DSC power lines go and put 3V on the pin used for output communication. The Data terminal which connects to DSC should read 12V if the connection and transistor switching is working fine.
I've used Homebridge-MQTT sketch. You can run mosquitto manually from console to see the debug messages. Are you sure, you are using partition 1 on your DSC?

jwolansk, 2018-09-19:
I mean data terminal will read 12V if power supply you are using is 12V

witek1308, 2018-09-19:
@jwolansk many thanks for willing to help, I forgot to update this topic - I managed to make it working. It was either power supply and/or cabling (I mentioned it in the other subject I opened). After changing the supplier and going through cables on both sides (I have them long from attic to garage) it works now fine. I integrated it with OH2 already, can control my alarm nicely now. Many thanks to all for your help!

closed 2018-09-19 by witek1308