taligentx/dscKeybusInterface #48: [Experience sharing] Integration with PC1616 over MQTT

witek1308, 2018-09-17:
Hi,
I'm going to use this great stuff with PC1616 over MQTT (mosquitto broker) to OpenHAB2, waiting for NodeMCU to arrive.
MQTT should be straightforward but maybe there is something I should know about: specific settings or issues? Is there anything to know not to reinvent the wheel?

witek1308, 2018-09-19:
What I observed is that when client was set to QoS(1) it was causing broker to retain messages (in my case disarm) which was overriding any arm. When I have some time to debug, will try to update but if I'm not too quick in my judgement this would suggest the way it is handled in the code and not sending ack for this type of message.

taligentx, 2019-05-21:
Hi @witek1308 - is OpenHAB sending commands to the 'dsc/Set' topic with retain?  It should be one-shot to make sure the interface doesn't repeat the command to the panel.

The retain flag is handy for the status topics in case the microcontroller is restarted.

witek1308, 2019-05-22:
> Data: 22 maja 2019 00:01:42
> Do: taligentx/dscKeybusInterface dscKeybusInterface@noreply.github.com
> DW: witek1308 witek_1308@poczta.onet.pl, Mention mention@noreply.github.com
> with PC1616 over MQTT (#48)

Hi it was long ago so don't remember details but definitely I had to 
configure separate mqtt profile on for dsc to get rid if the retain. I 
don't remember though if it was because of the Arduino code or maybe OH2 
configuration which for me defaults to retain as most of my devices send 
messages which need to be retained

--- Wiadomość oryginalna ---

taligentx, 2019-05-22:
Gotcha, thanks for the info - I'll keep an eye out to see if there are further issues with having the status messages set to retain.

closed 2019-05-22 by taligentx