taligentx/dscKeybusInterface #53: Zones 17-32 not working

luis1424, 2018-10-06:
At first I want to thank for this work. It's exactly what I was looking for.
My alarm has 20 active zones and I found that zones over 16 were not working. After a little debugging I found a bug in the dscKeybusProcessData.cpp file. 
In "processPanel_0x34()" where the panelData should be written to zones 17-24, the data are written to zones 9-16. Exactly the same happens with zones 25-32.
To solve this you must modify line  336
 openZones[1] = panelData[6];
with
 openZones**[2]** = panelData[6];
and line 359:
 openZones[1] = panelData[6];
with
 openZones**[3]** = panelData[6];

After that all seems to work OK.

l33d, 2018-11-12:
In development branch sketch is ok.

taligentx, 2019-05-21:
Hi @luis1424 - thanks for the the debugging, it was a terribly silly copy/paste error.

closed 2019-05-21 by taligentx