taligentx/dscKeybusInterface #55: Pushbullet  problem when armed

l33d, 2018-11-10:
Hi,

I recently used the pushbullet on esp8266. When the alarm is disarmed, everything is ok, notifications come correctly (eg. fire, etc). However, yesterday when I armed the alarm in home mode, after a while I got correctly notification 'armed partition 1 alarm' but in a few minutes I received again the same notifications, I had to block esp, so that it would not spam unnecessarily. Where is the problem?

taligentx, 2019-05-21:
Hi @l33d - I've tested and can't replicate this issue.  If you're still seeing this, please feel free to re-open this issue.  Thanks!

closed 2019-05-21 by taligentx

l33d, 2019-06-03:
Hi,

In new sketch i have this issue too. After armed system i received multiple notification like "Security system initializing"and "Security system connected". The problem isn`t connect to wifi, because when alarm is disarmed, and i disable and connect wifi, notofication isn`t send.