taligentx/dscKeybusInterface #58: Ethernet

danielswe88, 2018-11-28:
Hi!

I have an ethernet module connected to my DSC alarm already, is there any possibility to connect to that one to get the integration to my homey? It feels a little scary to fiddle around with the cables on the alarm PCB.

I am currently using the "PowerSeries Neo Go" app, but it is not working good at all, and I want to integreate my DSC alarm with the rest of my home automation.

Thanks in advance!
/Daniel

taligentx, 2019-05-21:
Hi @danielswe88 - DSC Neo series uses a different data protocol that is encrypted, so it is not supported by this interface.  Unfortunately the only option is to use the DSC provided accessories.  Thanks for taking a look!

closed 2019-05-21 by taligentx

danielswe88, 2019-05-22:
Hello @taligentx.

I did a lot of research about this and found out exactly what you are saying. I really appreciate your work on this project. Unfortunately I can't get that to work due to my hardware.

I was talking to my electrician (it's a newly built house), he said he could come and remove our PowerSeries NEO and instead install an older DSC with nothing in exchange. But I don't know... It feels so and so going back to older hardware, maybe I could wait until DSC launches some kind of API for the NEO series. Or I could try to reverse engineer the API...

/ Daniel

taligentx, 2019-05-24:
I've only taken a quick look but it seems that there are only a few options like [Control4](https://www.control4.com/solutions/whole-home) for integration with the [DSC TL280 Internet Communicator module](https://www.dsc.com/alarm-security-products/TL280%20-%20Internet%20Alarm%20Communicator/2487).

The TL280R version has an RS232 serial interface so it could be possible to reverse engineer the serial API, though I haven't seen any documentation on it.

The Neo series seems to be what DSC wants to migrate to completely, but the PowerSeries has such a large installed base that it's hard to see third parties dropping support anytime soon.  One advantage of the PowerSeries is that you would have a wider choice of third party interfaces - Envisalink, Alarm Decoder, etc.  But it's also a matter of home security so carefully considering the pros/cons is absolutely warranted.