taligentx/dscKeybusInterface #69: Resistor values are wrong for ESP8266

saikek, 2019-02-07:
I'm confused with your resistor values.
If i'm using your configuration - the voltage is around 5 volt which is deadly for ESP8266.
I've ended up using 6.6 and 3.3 Kohm and it works great.

Can anyone confirm their values ?
[TI][http://www.ti.com/download/kbase/volt/volt_div3.htm]

zzhang1, 2019-02-26:
Are you referring to the 33K / 10K shown in the diagram? I think that gets ~2.8V from a 12V input. I believe I ended up using 15K / 5K with my ESP8266, so from a source of 12V, it gets me 3V.

If you use 6.6K / 3.3K, doesn't that get you 12*3.3/(3.3+6.6) = 12V/3=4V? I have read conflicting things about the ESP8266 being 5V tolerant, so maybe it's a non-issue after all, but I wouldn't want to test that myself!

saikek, 2019-02-26:
I've been using `Witty-Cloud` programmer for my `ESP-12F` mod chips for a long time. (20+ chips programmed). Lately I received 5 chips which weren't entering programming mode at all. only after that I measured the voltage on a programmer and realized that it was giving **5V** all this time. So I can confirm that you can use 5V power supply (at least for some period of time, for programming).

As for resistors - I found values with multimeter and some experimentation. Ideally for me was opto-coupler approach, but it just wouldn't work for me.

zzhang1, 2019-02-26:
Agree, opto-couplers would be great. I got some generic ones in preparation to tap each sensor before coming across this amazing project, but similarly they wouldn't work. With no advanced tools to see why, I suspected it could be anything from lack of current to switching speed or dampening. 

I'm super inexperienced with this stuff, so when debugging my circuit, it took me a while to realize that the ~1.5V I was seeing on my voltmeter for the bus is probably a ~3V signal, blinking on and off!

taligentx, 2019-05-10:
Hi @saikek - the esp8266 GPIO pins are 5v tolerant as stated by Espressif and as tested in practice:

https://www.ba0sh1.com/blog/2016/08/03/is-esp8266-io-really-5v-tolerant/

It's a nice feature, especially when developing for multiple platforms.

closed 2019-05-10 by taligentx

jeschmidt, 2020-09-10:
I saw the schematic and noticed the same thing immediately, so I came looking for a thread and found one!  Even if the esp8266 is 5V tolerant, why feed it 5V if you don't have to (just by choosing different resistor values)?  Did you find the communication was more robust?  I see some open issues related to long term instability and resets required -- I wonder if those are due to overvoltage on the GPIO?  Note that the "12V" pin on my DSC panel is actually 13.64V!

Anyway, as I go to implement my version of this project, I will be selecting resistors to make the voltage more 3.3V GPIO-friendly.  If I remember I will report back here with my results.