taligentx/dscKeybusInterface #70: Question: Keypad input

dmachtey, 2019-03-08:
I'm using dscKeybusinterface with a PC585 with HomeAssistant-MQTT example and nodemcu esp8266, and it works great with no sw modifications. Thank you for the great job you have done.

I'm wondering, now, if there is any way to get the keypad input status?

thank you 
Damian

slipx06, 2019-04-06:
Just to add this works perfectly on my PC5005 using the HomeAssistant-MQTT example and nodemcu. I'm aslo missing the keypad but this is because the entity reports the code_format as "null" instead of a number. 

[image]

taligentx, 2019-05-21:
Hi @dmachtey - great to hear it's working with the PC585, added to the verified panels list.  I'm not clear on what you mean about the keypad input status, which info are you looking for?

@slipx06 - same, thanks for the info on the PC5005.  The 'code' option in the [Home Assistant MQTT Alarm Control Panel component ](https://www.home-assistant.io/components/alarm_control_panel.mqtt/) is to enter a code to arm/disarm the system if necessary.  In this case, it isn't necessary - the code is entered directly in the sketch and this interface will handle entering the code as needed.

UltimatPronin, 2019-06-07:
@taligentx, maybe @dmachtey mean something same as me in #86

taligentx, 2019-06-13:
Closing, feel free to re-open if the question is still pertinent.

closed 2019-06-13 by taligentx

gabter1, 2019-11-30:
I don't want to open a new issue, but I have a similar question. I am using the Home Assistant sketch without any changes on a PC5020, so you can add it to the verified list.
I would like to take out the security code from the sketch, as I believe it's a security risk, and would like to use the keypad in the alarm panel of HA. I added "code_format: number" to the configuration.yaml and now HA is showing the keypad. Can you tell me what needs to be changed in the sketch to actually read the code from the keypad and use that as the disarm code?