taligentx/dscKeybusInterface #74: Question: Virtual Blynk Zones name

l33d, 2019-05-22:
Hi @taligentx 

I have a question about zones info. How can i rename zones info on lcd? Something like from this: zone 2 Open to Kitchen Open? Only in blynk or mus change in sketch? How can i do this?
Next question. How can i made a translate the information on the lcd? Can i do this?

Cheers

taligentx, 2019-05-24:
Hi @l33d - the Blynk LCD only shows "Zones open" and doesn't specify a zone number, the zone LEDs handle that part.  You can set custom names if you change the sketch in the `setStatus()` function.

You would need to change:
`case 0x03: lcd.print(0,1, "                "); lcd.print(0,1, "Zones open"); break;`

To something like:
```
    case 0x03: {
        if (bitRead(dsc.openZones[0], 0)) {
            lcd.print(0,1, "                "); lcd.print(0,1, "Zone 1 open");
        }
        else if (bitRead(dsc.openZones[0], 1)) {
            lcd.print(0,1, "                "); lcd.print(0,1, "Zone 2 open");
        }
        else {
            lcd.print(0,1, "                "); lcd.print(0,1, "Zones open");
        }
        break;
    }
```

This is untested but that's the idea - for each zone that you would like a custom label, have the sketch check if the zone is open and then print a custom label.  See the sketch section that updates the zone LEDs for how to check if a particular zone is open.

The `setStatus()` function is also where you can translate the Blynk text - it would be nice to have a separate file with different translations, added to the todo list.

l33d, 2019-05-24:
Hi,

Big thanks, i try in this week ;)

taligentx, 2019-05-25:
That should work, though as I mentioned I haven't actually tested the code.  You'll need to create a new section for each zone in that `if... else if... else` section for open zones.  You're right about the translation for the `setStatus()` function, for the Blynk sketch all of the terms are in the sketch.

l33d, 2019-05-25:
Ok, thanks i try and test.

l33d, 2019-05-27:
Hi,
I have only translated statuses in 95%. Unfortunately, today I tried to change the names of the zones as shown in the example, but it does not work, or I do not understand something or it will not work like this :(
First metod i try like this:

> case 0x50: {
>     if (bitRead(dsc.openZones[0], 0)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 1 open");
>     }
>     else if (bitRead(dsc.openZones[0], 1)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 2 open");
>     }
>     else {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zones open");
>     }
>     
>      break; }
>      
>       case 0x51: {
>     if (bitRead(dsc.openZones[0], 2)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 3 open");
>     }
>     else if (bitRead(dsc.openZones[0], 3)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 4 open");
>     }
>     else {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zones open");
>     }
>      
>     
>        break; }
>   
> 
>      case 0x52: {
>     if (bitRead(dsc.openZones[0], 4)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 5 open");
>     }
>     else if (bitRead(dsc.openZones[0], 5)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 6 open");
>     }
>     else {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zones open");
>     }
>     
>      break; }
> 
>       case 0x53: {
>     if (bitRead(dsc.openZones[0], 6)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 7 open");
>     }
>     else if (bitRead(dsc.openZones[0], 7)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 8 open");
>     }
>     else {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zones open");
>     }
>     
>      break; }

and second 

> case 0x50: {
>     if (bitRead(dsc.openZones[0], 0)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 1 open");
>     }
>     else if (bitRead(dsc.openZones[0], 1)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 2 open");
>     }
>     else {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zones open");
>     }
>     
>      break; }
>      
>       case 0x51: {
>     if (bitRead(dsc.openZones[0], 2)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 3 open");
>     }
>     else if (bitRead(dsc.openZones[0], 3)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 4 open");
>     }
>     else {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zones open");
>     }
>      
>      
>        break; }
>   
> 
>      case 0x52: {
>     if (bitRead(dsc.openZones[0], 4)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 5 open");
>     }
>     else if (bitRead(dsc.openZones[0], 5)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 6 open");
>     }
>     else {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zones open");
>     }
>     
>      
>        break; }
> 
>       case 0x53: {
>     if (bitRead(dsc.openZones[0], 6)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 7 open");
>     }
>     else if (bitRead(dsc.openZones[0], 7)) {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zone 8 open");
>     }
>     else {
>     lcd.print(0,1, " "); lcd.print(0,1, "Zones open");
>     }
>     
>     
>      break; }

I wanted to add my translation of the pushbullet and virtualblynk sketch, but I do not know how to do it on github. I tried, but something did not work out and did not add it.

Cheers,

closed 2020-11-02 by taligentx