taligentx/dscKeybusInterface #75: ESPHome Integration

brad55590, 2019-05-23:
Hi

Would you be able to do an ESPHome Integration?

Regards

taligentx, 2019-05-24:
Hi @brad55590 - if you mean sending status to another device running ESPHome, it seems like you can do this already using the HomeAssistant-MQTT sketch to send data and the ESPHome [MQTT Subscribe Sensor](https://esphome.io/components/sensor/mqtt_subscribe.html) to receive data.

Otherwise, what sort of integration do you have in mind?

brad55590, 2019-05-26:
So what i had in mind was possibly creating a DSC Sensor for ESPHome.

taligentx, 2019-05-31:
ESPHome and Tasmota seem appropriate for simple sensors that send back a limited set of values, there's a bit more going on in this interface.   The core code handling the Keybus data capture, buffering, and processing uses significant resources on the esp8266 and trying to shoehorn it into a larger project may result in resource contention. A straight merge to Tasmota/ESP Home would also require dropping Arduino AVR microcontrollers - they are quite limited but still useful and widespread.

You can also check with the ESPHome community to see if there is anyone interested in merging the code.  I'm not familiar with ESPHome so it's not currently a target.

closed 2019-05-31 by taligentx

Dilbert66, 2020-07-21:
Works fine. Here's my implementation.  

https://github.com/Dilbert66/esphome-dsckeybus