taligentx/dscKeybusInterface #76: DSC Power 832, 5010 issue.

Vagelis1971, 2019-05-24:
Hi guys.
Amazing work , with this i was able to integrate my old DSC 5010 into Home assistant.
Tho im having an issue with writing into the panel. I can get all the status data ,alarm data but i cant arm the system from home assistant. So i thought to give it a try with a web virtual keypad, Same thing over there to. I can read all the data but i cant arm the system.
I ve checked all the wiring many times and it looks fine , i ve checked the sketch for inserting  up the data  correctly and that is fine too. Im trying to find the solution for some days now  with out any luck. Should i try to use a smaller resistor than 1K into the base of the  NPN transistor? Maybe you guys have something to suggest me ? 
Thanks in ADV.

taligentx, 2019-05-24:
Hi @Vagelis1971 - try loading the KeybusReader example sketch and view the serial output, are you seeing data from the panel?  If so, are you seeing any messages about data errors?

If that works, try typing in a few keypad keys (like 0-9) and see if the keys are recognized in the output.  If not, there is an issue with the pin, wiring, transistor, resistor, or connections (should be soldered, no breadboard, etc).

l33d, 2019-05-24:
Hi Vagelis1971 

I think the problem would be in wiring connectin or you something missed and didn`t change in sketch. Are you use arduino or esp?

Vagelis1971, 2019-05-25:
Thanks for the fast response people... im using  esp8266 NodeMcu v3  and im using the pins d1 d2 for data and clock and d8 for the virtual keypad as it is mentioned in the sketch , all connections are soldered in place only the cables from dsc are not . 
Ill try to load the keubus reader in the evening and ill report back. 
Im sure about the resistor and the transistor all are in correct position and the NPN is soldered corectly.
Home assistant/virtual keypad was receiving the data from the panel otherwise i wasnt be able to see the zones,troubles and the alarm status change in the HA/virtual K.  screen.. right?
I need to mention that the panel is kind of old . it doent have a screen on it but only bulbs not even LEDs  (DSC PC5516) could that cause the problem?
Im very greatfull for your help guys .. thanks alot

l33d, 2019-05-25:
Are you trying another sketch like VirtualKeypad-Web?

Vagelis1971, 2019-05-25:
Yeap. I have tried it. I'm getting all the read data like alarm status, trouble, etc but when I hit the numbers to arm it it does nothing or even when I hit the arm stay or arm away button its not arming.

l33d, 2019-05-25:
Hmm. Please try to change declared pins from d1 d2  d8 to 5 4 15  and try again.

Vagelis1971, 2019-05-25:
Thanks to all ... I found the solution.
I changed the transistor from 2nxx to a BC 547 and it worked.They have a different pinout so watch it there.I have no idea if the 2Nxx was fried or the supplier send the wrong one i cant read it out  but i dont think so .
So people i can fully say that it works on a DSC Power 832 ( 5010 ) for sure.
Thanks again and keep up the good work.

taligentx, 2019-05-25:
Great to hear, thanks for the update!

closed 2019-05-25 by taligentx