taligentx/dscKeybusInterface #77: [Question] - how to stop dscKeybusInterface

UltimatPronin, 2019-05-27:
Hello! How I can stop dscKeybusInterface? I use nodemcu v3 I want to use OTA, but when my writePIN konnected to the GREEN key-bus wire I can't do OTA, it starts and failed. I my sketch without dsc.start(), keeping all wires connected to panel - ota work fine. Any ideas about it? I use last SDK & Core.

l33d, 2019-05-27:
Hi,
Which sketch you use?

UltimatPronin, 2019-05-27:
I use my own sketch with a lot of code, but now main part of your great tool is used like as your sketch for receive status of dsc panel without changes. Now after OTA start wifi drop connection. With http OTA is the same problem. Your tool work like a charm, but, if i leave all wires connected to panel, ota fails.

l33d, 2019-05-28:
This is not my tool ;) Wait for @taligentx reply.

UltimatPronin, 2019-05-28:
sorry i didn't notice that answered not @taligentx ))) 

I investigated problem a little deeper: wi-fi reconnected, because stack error and reboot. I will try check stack errors.

taligentx, 2019-05-31:
Hi @UltimatPronin - which pins are you using?  For `dscWritePin` you can also try D6 or D7 if there is a conflict on the NodeMCU with D8.  I use the Wemos D1 mini, but have a NodeMCU v2 on the way for testing.

UltimatPronin, 2019-05-31:
Hi, I use D8. at the moment there is no possibility to change the pin - the module is sealed and installed in a hard-to-reach place. At the moment I solved the problem by installing a special label before the OTA update and reboot before OTA without start your's great tool.

jim77, 2019-05-31:
@UltimatPronin which esp8266 module are you using? I might be experiencing the same issue. I am using the Homie-esp8266 and its OTA implementation via MQTT. I am using Wemos D1 Mini Pro, and dscWritePin is currently wired to D8. I tried changing the writepin to D7, but that didn't make any difference.

My solution: Homie has a hook for its OTA_STARTED event, which allows me to call timer1_detachInterrupt(). This solved my problem and I am able to successfully perform OTA. After the OTA, the device will reset itself to resume normal operation.

UltimatPronin, 2019-05-31:
@jim77 , I'am using GEEKCREIT DOIT ESP-FDEVKIT V4 NODEMCU module.
 I will try your advice to call timer1_detachInterrupt() before OTA.

taligentx, 2019-06-05:
You can also apply this for all platforms (both esp8266 and esp32) by using the following to disable the clock interrupt, which will also prevent the DSC data timer interrupt from firing:

`detachInterrupt(digitalPinToInterrupt(dscClockPin));` 

For example:
```
  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_SPIFFS
      type = "filesystem";
    }
    detachInterrupt(digitalPinToInterrupt(dscClockPin));  // Disables the DSC clock interrupt and prevents the DSC data timer interrupt from interfering with OTA
    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
```

I'm not clear on why the timer interrupt is affecting OTA, flash write operations should not be affected by the timer interrupt because the DSC code is being placed in RAM - but I haven't looked into it deeply.  Disabling the interrupt seems like a simple enough solution that unless another issue comes up I'll add a note to the README.

aboulfad, 2019-06-05:
Without adding the `detachInterrupt`, I could not either get OTA to work, it would start the upload and fail right away. After I added that one liner I can OTA and run dsc routines (example status.ino). I did not investigate further and glad for that suggestion @taligentx you made above.

taligentx, 2019-06-07:
Added `dsc.stop()` to the develop branch.

aboulfad, 2019-06-08:
I cherry-picked commit 5b5a08fb6368408ed13e1b53dce9a0ec3c96db6b and re-tested OTA using dsc.stop() https://github.com/taligentx/dscKeybusInterface/blob/5b5a08fb6368408ed13e1b53dce9a0ec3c96db6b/src/dscKeybusInterface.cpp#L112

It works, thank you.

taligentx, 2019-06-08:
Great, thanks for the feedback - closing.

closed 2019-06-08 by taligentx