taligentx/dscKeybusInterface #78: Display read

otto76, 2019-05-27:
Hi, do we have any possibility to read values from display. 
One issue I am facing - 2x per year we are shifting from Summer to Winter time and vice verse. (Plus my clock is not accurate) I would be happy to get an access to current data/time setting and if they are shitted, then push an update to correct one

taligentx, 2019-05-31:
Hi @otto76 - the LCD keypads generate their own displays based on the information sent by the panel, so there isn't any data to read directly from the keypads.

For the panel time, that information is available but currently the library isn't tracking it to save on memory.  It's on the todo list, I'll have to see what the memory impact is to manage the time.

There is a way to update the time already through virtual keypad, you just need to send the appropriate keys just as you would enter on a physical keypad.  So you could add an NTP library to get the current date/time and then send the sequence of keys to update the panel time.  Automating this is also on the todo list.

jim77, 2019-05-31:
I have an LED keypad, so it doesn't show the date/time. I'm wondering what the date/time information is used for in my case?

I've just had to do the time update because when I unplugged the battery + power, the system lost its time and the Trouble LED is lit.

These are the instructions from my panel manual:
Setting the System Date and Time
To set the system time, enter [✱] [6] followed by the Master Code. Press [1]. The keypad will now accept 10 consecutive digits:
• Enter the Time in Hours and Minutes using the 24 Hour format (00:00 to 23:59).
• Enter the Date in Months, Days and Years (MM DD YY).
NOTE: If you have an LCD keypad, your installer may have programmed your system to display the time and date while the keypad is idle. If this is the case, you may have to press the [#] key to clear the date and time before entering an access code to arm the system, or before performing any other keypad function.

taligentx, 2019-06-06:
@otto76 I missed it earlier but getting the panel time has been possible since the initial release as: `dsc.year`, `dsc.month`, `dsc.day`, `dsc.hour`, `dsc.minute`

I've added a new method to set the panel time:

`dsc.setTime(unsigned int year, byte month, byte day, byte hour, byte minute, const char* accessCode)`

Example to set 2018.02.06 15:45 (year can be either 2 or 4 digits):

`dsc.setTime(18, 2, 6, 15, 45, "1234")`

You can use this with NTP and the [Timezone](https://github.com/JChristensen/Timezone) library to get NTP time, adjust for timezone and daylight savings and then manage the DSC panel time automatically.  If you or anyone else does this, a PR would be great to add to the example sketches.

otto76, 2019-06-07:
Great, thx @taligentx 
My case is a bit different, my DSC works on seperate VLAN w/o access to home or Public nrtwork. All interaction happens via OpenHab (which has access to all 3 networks- DSC.VLAN, home and public). So, idea is to write a OH rule, to query DSC on regular bases (aka once per week) and verify result w/ time retrieved from NTP. If deviation is greater then xxx, then send an update to DSC

Many thx, its verily great to have it

otto76, 2019-06-11:
It seems to hours and minutes does not work correctly 
Code I have: 
`
// =============================================================
// Publish time
// -------------------------------------------------------------
void publishDatetime() {
char buffer[50];             

    StaticJsonDocument<50> doc;
    doc["year"] = dsc.year;
    doc["month"] = dsc.month;
    doc["day"] = dsc.day;
    doc["hour"] = dsc.hour;
    doc["minute"] = dsc.minute;
    
//    serializeJson(doc, buffer);
    size_t n = serializeJson(doc, buffer);

    mqtt.publish(mqttGetDatetime, buffer, n, true);      
}
`

But result I got: 
> dsc/Get/Datetime {"year":19,"month":5,"day":11,"hour":23,"minute":

Hours are not correct and minutes missing at all

taligentx, 2019-06-13:
I've updated the `Status` sketch in the develop branch to demonstrate getting the panel timestamp, see if there are any issues with that sketch.

closed 2019-06-15 by taligentx

jimtng, 2019-06-23:
setTime doesn't seem to be working correctly. After calling it, the panel time didn't change. Example:

dsc.setTime(2019, 6, 1, 1, 1, "accesscode");

In fact, I tried this **multiple times**, and suddenly my alarm went off. Probably not the most popular house in the neighbourhood now.

When the alarm was blaring, I ran to the physical keypad and entered my access code once. It didn't stop the alarm. In panic I pressed Reset, and entered the access code again and again.... lol. Eventually the alarm noise stopped and the "Ready" LED was on.

I observed two problems:
1. I noticed that the partition 1's alarm flag is still set to 1. It doesn't seem to have gotten reset back to 0 (like the openZone for example) after the alarm has been disarmed. Is this supposed to be reset to 0 by my sketch after reading it?

2. Why did my alarm go off? Was it something to do with setTime not working for me?

Have you tested setTime() on your system, e.g. repeatedly / multiple times etc?

realthk, 2020-05-07:
Yes, it works in develop branch, though it only gets updated every 4 minutes or so.
I guess 0xA5 and 0xEB messages (as the handlers of these messages: processPanel_0xA5() and processPanel_0xEB() updates time info) are sent not too frequently. 
Not a real problem, just don't be surprised if time seems to be off by a few minutes.