taligentx/dscKeybusInterface #80: ESP32 support - testing/feedback

taligentx, 2019-05-31:
I've added experimental support for the esp32 in the `esp32` branch - testing/feedback is most welcome!

mariokc, 2019-06-04:
Hi!
Tested this on a PC 1520 with nodeMCU.
Everything is working fine.

Elektrik1, 2019-06-06:
Why do we need an esp32 ? To use with BT ? I think esp-01/esp-07s is the best mcu for this particular use, cheapest, smallest solution.

UltimatPronin, 2019-06-07:
esp32 is more powerfull and modern - you know it ))).
Not for this amaizing project, but for ESP8266 some features is limited or may not possible due to its memory and available crypto library.

In near future I will try it too.

balazsmark, 2020-03-08:
> Why do we need an esp32 ? To use with BT ? I think esp-01/esp-07s is the best mcu for this particular use, cheapest, smallest solution.

Ethernet? :)

drbubo00, 2020-03-14:
I have a DSC 1616. I try to compile \examples\esp8266\HomeAssistant-MQTT.ino with Arduino IDE, but I can't. Compile Error "'dscPartitions' was not declared in this scope". Why? Can 
 

> > Why do we need an esp32 ? To use with BT ? I think esp-01/esp-07s is the best mcu for this particular use, cheapest, smallest solution.
> 
> Ethernet? :)

I have a Nodemcu-32 s and a DSC 1616. I try to compile \examples\esp8266\HomeAssistant-MQTT.ino with Arduino IDE, but I can't. Compile Error "'dscPartitions' was not declared in this scope". Can U help me?

hblategan, 2020-09-20:
Thanks for all the great work.

Just one note: the **VirtualKeypad-Web** example fails with a core dump because of the usage of Timers + SPIFFS (WebServer)
I think it's related to this issue: https://github.com/espressif/arduino-esp32/issues/1781
But I am not certain.

Just hopefully a helpful hint for anyone starting out with that example.

This is the core dump:

PC: 0x400df580: VFSFileImpl::VFSFileImpl(VFSImpl*, char const*, char const*) at C:\Users\Herko\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src\vfs_api.cpp line 227
EXCVADDR: 0x00000000

Decoding stack results
0x400df580: VFSFileImpl::VFSFileImpl(VFSImpl*, char const*, char const*) at C:\Users\Herko\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src\vfs_api.cpp line 227
0x4008168d: dscKeybusInterface::dscDataInterrupt() at C:\Users\Herko\Documents\Arduino\libraries\dscKeybusInterface-esp32\src\dscKeybusInterface.cpp line 640
0x40081ea9: pinMatrixOutAttach at C:\Users\Herko\Documents\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-matrix.c line 25
0x40084a0d: ipc_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/ipc.c line 62
0x400e3695: String::lastIndexOf(String const&, unsigned int) const at C:\Users\Herko\Documents\Arduino\hardware\espressif\esp32\cores\esp32\WString.cpp line 693
0x400ddf4e: AsyncAbstractResponse::_fillBufferAndProcessTemplates(unsigned char*, unsigned int) at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebResponses.cpp line 403
0x400dd6a3: AsyncWebServerResponse::AsyncWebServerResponse() at c:\users\herko\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 1974
0x400dd6c7: AsyncWebServerResponse::AsyncWebServerResponse() at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h line 461
0x400d87db: requestDigestAuthentication(char const*) at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebAuthentication.cpp line 137
0x400d8a09: checkDigestAuthentication(char const*, char const*, char const*, char const*, char const*, bool, char const*, char const*, char const*) at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebAuthentication.cpp line 170
0x400d8b31: checkDigestAuthentication(char const*, char const*, char const*, char const*, char const*, bool, char const*, char const*, char const*) at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebAuthentication.cpp line 180
0x400dd316: AsyncFileResponse::_setContentType(String const&) at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebResponses.cpp line 487
0x400dab85: AsyncWebServerRequest::_parseMultipartPostByte(unsigned char, bool) at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 445
0x400dac85: AsyncWebServerRequest::_parseMultipartPostByte(unsigned char, bool) at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 451
0x400daed5: AsyncWebServerRequest::_parseMultipartPostByte(unsigned char, bool) at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 495
0x400de6c1: AsyncFileResponse::AsyncFileResponse(fs::File, String const&, String const&, bool, std::function ) at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebResponses.cpp line 553
0x400de741: AsyncFileResponse::AsyncFileResponse(fs::File, String const&, String const&, bool, std::function ) at C:\Users\Herko\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebResponses.cpp line 565
0x400ded52: std::function ::operator()(AsyncWebServerRequest*) const at c:\users\herko\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 2266
0x4008e93d: multi_heap_free_impl at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap.c line 12

taligentx, 2021-01-22:
esp32 is supported as of the newly released version 2.0, including the VirtualKeypad-Web sketch.  Note that [arduino-esp32](https://github.com/espressif/arduino-esp32) core 1.0.5-rc6 or newer is required as it includes a fix to place timer interrupt functions in IRAM.

closed 2021-01-22 by taligentx