taligentx/dscKeybusInterface #81: power supply for esp8266

jim77, 2019-05-31:
This is unrelated to the dscKeybusInterface, but I thought I'd ask here to hear others experience. In powering my Wemos D1 Mini Pro from dsc's AUX, I tried using 7805 and also AMS1117 5V (not at the same time). However they both got quite hot. Not having a heat sink handy, and not wanting to spend too much time tinkering, I ended up using an external USB adapter and pushing through a USB cable from outside the alarm box to power my Wemos board. This gives me an added benefit of being able to hard reset and also perform firmware update if needed, although I can also do OTA.

I have also tried using a NodeMCU with vin that claimed to be OK for up to 20V. Its voltage regulator is AMS1117 on board, and it too got quite hot. I read somewhere that it can only really handle no more than 12V and the AUX power from the DSC board is more like 13.6V.

What do you guys use to power your board?

UltimatPronin, 2019-05-31:
@jim77 , I use 2596 / LM2596 /Step Down Power Supply Module (Supply Voltage:DC 4.5-24V 3Amax) for NodeMCU.
It connected to panel AUX and configured to stepdown to 5V - this side connected to VIN pin.
LM2596 only a little hot, after long time of use.
Maybe later I will try to connect AUX to VIN directly.

aboulfad, 2019-05-31:
If you have space to spare, check this [5V buck converter](https://www.ebay.com/itm/Fine-6-24V-12V-24V-to-5V-3A-CAR-USB-Charger-Module-DC-Buck-step-down-Converter/401183759010?hash=item5d686a5aa2:g:afkAAOSwyHVbMvJG), I've used it in many projects and it works great. I usually desolder the USB connector and cut the board to make it smaller.

taligentx, 2019-05-31:
As @UltimatPronin and @aboulfad mentioned, a step-down buck converter will be more efficient than using a linear voltage regulator directly from the DSC Aux power rail.  Modules based on either the LM2596 or MP2307 (also labeled as Mini360 modules) are commonly available for < $1 each shipped.

The LM2596 modules seem to run more efficiently, whereas the MP2307 modules run hot and some versions of the Mini360 modules [use under-spec'd components](https://johndoe31415.github.io/mp2307calc/) causing even worse efficiency.

I prefer to set these modules to output 5v and let the ESP8266/ESP32 onboard linear voltage regulator take care of the 3.3v for the ESP modules - this helps reduce the ripple current and switching noise from the buck converter.

I'm waiting on delivery of a NodeMCU to see if the AMS1117 is staying within acceptable thermal range when powered directly from the DSC panel.  Tagging to update the documentation with power supply info.

aboulfad, 2019-05-31:
You learn something everyday! I have like a dozen of nodemcu and always thought Vin to be 5V. Checking the [Datasheet of NCP1117](https://www.onsemi.com/pub/Collateral/NCP1117-D.PDF) does indeed support Vi up to 20V. It’s internal thermal shutdown is at 175C, and heat dissipation depends on pad size. Given it’s tiny size it’s expected to be hot, best is to measure with an infrared thermometer and see if it’s within specs.

jwolansk, 2019-05-31:
Yeah, but buck converters do not work well with ESP8266, they make the ESP unstable. It probably is a matter of filtering the noise, but I was unable to make it work as stable as with linear regulators.

aboulfad, 2019-05-31:
We shouldn’t generalize, I use buck converters with all my nodemcu and esp-12e modules with no issues of stability or reboots/crashes. It depends on the model, YMMV.

taligentx, 2019-06-21:
Results (temps measured by thermocouple):
* NodeMCU v2 module with AMS1117 directly powered at 13.6v:
    * AMS1117 surface temp: 71.0°C
    * Power usage: 1.11w
* NodeMCU v2 module with MP2307 (Mini360) module regulating to 5v:
    * AMS1117 surface temp: 39.5°C
    * MP2307 surface temp: 52.5°C
    * Power usage: 0.92w
* NodeMCU v2 module with LM2596 module regulating to 5v:
    * AMS1117 surface temp: 39.5°C
    * LM2596 surface temp: 32.0°C
    * Power usage: 0.59w

So:    
* Directly powering the NodeMCU works, but the AMS1117 will run quite hot.
* The MP2307 also runs hot with an efficiency only a bit better than the AMS1117 alone.
* The LM2596-based modules are significantly more efficient and run cool.  The power usage of this module would be even lower except for its bright onboard LED.

I've stocked up on the LM2596 modules as they are quite inexpensive - [pack of two modules for $0.75USD shipped](https://www.ebay.com/itm/2XLM2596-DC-DC-Adjustable-Buck-Converter-Step-Down-Power-Supply-Module-1-23-30V/282261008912?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649).  

Updated the README.md with power supply information in the `Wiring notes` section and removed the NodeMCU direct power wiring diagrams from the example sketches.

closed 2019-06-21 by taligentx