taligentx/dscKeybusInterface #82: Implementation using Homie framework

jimtng, 2019-05-31:
Hi, I created an implementation using the Homie framework. Published here:
https://github.com/jimtng/dscKeybusInterface-Homie

taligentx, 2019-06-03:
Hi @jimtng - thanks for the post, I'll add a link to the README.  Also an obviously better way to reset the open zones and alarm zones!

Out of curiosity, are you using this with OpenHAB or other software?

jimtng, 2019-06-04:
I use OpenHAB, yes. I had fun writing a countdown rule to activate my alarm. It goes like this:

"Hey Google, Initiate Self Destruct Sequence" (to activate the alarm).

It would then say "Self Destruct sequence has been initiated. 2 Minute countdown"
... 1 minute later...
"Self destruct in 1 minute"
...50 seconds later...
"Self destruct in 10, 9, 8, 7, ..., 1, BOOM!"

This is achieved using a timer in an openhab rule and then the item is exposed to Google Home.

Regarding the way to reset, I was wondering about resetting the Zones inside the loop the same way (i.e. var[zone] = 0, thus setting all the bits for that zonegroup to 0 in one step) , but I'm not 100% sure whether bits could be set by an interrupt while I'm inside my loop, hence I'm leaving it as bitWrite.

taligentx, 2019-06-21:
> but I'm not 100% sure whether bits could be set by an interrupt while I'm inside my loop

This shouldn't happen, the interrupts capture the Keybus data but all of the status processing happens in `dsc.loop()`.  Also a heads-up, there are a few changes coming in the next release:

* `handlePanel()` is now `loop()` and `writeReady` is deprecated, they will still work in the next release but will eventually be removed.
* Power wiring diagram updated to remove NodeMCU direct power based on #81.  
* `resetStatus` and `stop()` implemented in the library

That's great to see about OpenHAB, Google integration has been on the list for a while.  Awesome countdown!

jimtng, 2019-06-22:
Thanks for the heads up. I have updated my code with the latest changes you've made in the library. It will be committed to github after I've done some tests.

As for OpenHAB, it could probably use the HomeAssistant example, but I prefer my own implementation

closed 2019-07-06 by jimtng