taligentx/dscKeybusInterface #83: Question about dsc.keypadFireAlarm, keypadAuxAlarm and keypadPanicAlarm

jimtng, 2019-06-01:
Hi,

I noticed that these three flags (keypadFireAlarm, keypadAuxAlarm and keypadPanicAlarm) don't have the corresponding "Changed" flag. When do we know that these alarms have been cleared?

taligentx, 2019-06-03:
These are one-shot commands, the panel does not repeat the command so setting each of these flags to `false` is enough to track changes.  All other commands are constantly repeated by the panel so there a second flag is needed to track changes.

jimtng, 2019-06-04:
I still don't understand it. What is the purpose/function of those three "keypad xxx alarms" ? Whey do they get set / sent by the panel, and what does it mean?

taligentx, 2019-06-04:
These are for the physical keypad panic keys so you can determine if a user has physically pressed one of the three panic buttons on one of the keypads.

otto76, 2019-06-04:
Is it possible to trigger Panic alarm?
And can we trigger 'full alarm' including sirena, etc. Panic button sends silent call to security service w/o screaming

jwolansk, 2019-06-04:
I second this, triggering the alarms would be very nice to have.

taligentx, 2019-06-04:
> Is it possible to trigger Panic alarm?
> And can we trigger 'full alarm' including sirena, etc. Panic button sends silent call to security service w/o screaming

This is already possible, check the virtual keypad section in `README.md`.  By default the panic key is silent but you can configure the DSC panel to turn on the bell (siren) when the panic key is pressed by turning on section `015` option 2 - Panic Key Audible (Bell / Beeps).

closed 2019-06-07 by taligentx