taligentx/dscKeybusInterface #84: DSC 1832: 0xE6, 0xCE - Unrecognized data

aboulfad, 2019-06-02:
I've mostly decoded 0xE6 and 0xEB and have pushed a new [develop branch](https://github.com/taligentx/dscKeybusInterface/tree/develop) for feedback and testing,  supporting partitions 2-8, zones 33-64, and writing to partitions 1 and 2.  
...
_Originally posted by @taligentx in https://github.com/taligentx/dscKeybusInterface/issues/2#issuecomment-395521182_

taligentx, 2019-06-03:
Hi @aboulfad - which panel model do you have?  I can update the `0x16` command.

You'll see several commands with unrecognized data, I've decoded most of the protocol but there are still several commands that I haven't tracked down.  The ones that remain are likely tied to RF, thermostat, and other accessory modules that I don't have.  It can be done if anyone is interested, but it's also a question about whether these commands are useful for any practical purpose.

aboulfad, 2019-06-04:
Agreed, I didn't know those may be related to RF, my keypad is a RF model.
My panel model is: **PC1832** which I see is enumerated in https://github.com/taligentx/dscKeybusInterface/blob/93e1f361565c7f806913f7f6137f1e99250265aa/src/dscKeybusPrintData.cpp#L787
but haven't gotten to the part to understand the response and why it isn't "recognized".

Thanks.

taligentx, 2019-06-05:
Thanks for the data logs, fixed in develop branch - `KeybusReader` should correctly report the PC1832 now.

taligentx, 2019-06-05:
As far as the commands that show up as unknown or unrecognized in `KeybusReader`, it's a matter of watching the data, making a single change on the DSC system, and then seeing which individual data bits in the commands changed.

If you or anyone else is able to narrow down which actions affect these commands I can update the code - eventually it would be nice to have a complete decoding of the entire protocol but as I mentioned it's a bit academic because the commonly useful commands have been decoded.

aboulfad, 2019-06-05:
I've just tested the 0x16 fix, and it works of course.
`  257.21: 00010110 0 00001110 01000001 11110101 01011010 [0x16] PC1832 | Zone wiring: NC | Interval 4m`

As for the wireless RC commands, I've captured some logs that I will parse later on, but as you said not important. Thank you.

closed 2019-06-05 by aboulfad