taligentx/dscKeybusInterface #85: Pusbullet 1.2  notification problem when armed.

l33d, 2019-06-03:
Hi,

In new sketch i have this issue too. After armed system i received multiple notification like "Security system initializing"and "Security system connected". The problem isn`t connect to wifi, because when alarm is disarmed, and i disable and connect wifi, notofication isn`t send.

_Originally posted by @l33d in https://github.com/taligentx/dscKeybusInterface/issues/55#issuecomment-498384888_

taligentx, 2019-06-07:
Hi @l33d - I can't replicate this, I just re-tested and the initialization and connected messages are sent once at startup and the stay arm and disarm notifications are being sent.

If you're seeing multiple notifications about initializing, check the Serial output to see what's happening - the esp8266 could be resetting for some reason.  You can also run the `Status` example and watch the output to verify it matches the system states.

l33d, 2019-06-07:
After this problem, when i testing pushbullet today and try use virtualblynk, virtualblynk stop responding for  button keys, like write pin is not connected( works for about 2 weeks ok when i try it sometimes). Maybe the problem is power?  I use stepdown converter on lm2557 from aliexpress. This problem is only mine?

taligentx, 2019-06-08:
Could be power, you're seeing issues that I can't replicate and I'm not seeing other similar issues being posted.

Pushbullet is one-way notification, but you could probably setup something in Blynk to restart the ESP though that's outside the scope of this project.

l33d, 2019-06-09:
Thanks man. You`re probably right. I connect to serial and arm alarm, when alarm is armed esp always reset, but voltage is  ok, about 3,43v and when i turn it a bit high about 3,63v problem is not solved. I will check cables and more. When i find the issue i write about it.

l33d, 2019-06-10:
Ok, problem is probably in my sketch. When i armed stay i don`t received information about that. I upload original sketch and now i see this notification. Strange. I upload my translate sketch but i don`t know where is the problem.

`void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  WiFi.mode(WIFI_STA);
  WiFi.begin(wifiSSID, wifiPassword);
  while (WiFi.status() != WL_CONNECTED) delay(100);
  Serial.print(F("WiFi connected: "));
  Serial.println(WiFi.localIP());

  // Sends a push notification on startup to verify connectivity
  #ifdef AXTLS_DEPRECATED
    pushClient.setInsecure();  // Sets authentication method for BearSSL in esp8266 Arduino Core 2.5.0+
  #endif
  if (sendPush("System DSC uruchamianie")) Serial.println(F("Initialization push notification sent successfully."));
  else Serial.println(F("Initialization push notification failed to send."));

  // Starts the Keybus interface
  dsc.begin();

  Serial.println(F(" System DSC magistrali Keybus jest podłączony."));
}

void loop() {
  if (dsc.handlePanel() && dsc.statusChanged) {  // Processes data only when a valid Keybus command has been read
    dsc.statusChanged = false;                   // Resets the status flag

    // If the Keybus data buffer is exceeded, the sketch is too busy to process all Keybus commands.  Call
    // handlePanel() more often, or increase dscBufferSize in the library: src/dscKeybusInterface.h
    if (dsc.bufferOverflow) Serial.println(F("Przepełnienie bufora magistrali KEYBUS"));
    dsc.bufferOverflow = false;

    // Checks if the interface is connected to the Keybus
    if (dsc.keybusChanged) {
      dsc.keybusChanged = false;  // Resets the Keybus data status flag
      if (dsc.keybusConnected) sendPush("System DSC podłączony");
      else sendPush("System DSC odłączony");
    }

    // Checks status per partition
    for (byte partition = 0; partition < dscPartitions; partition++) {

      // Checks armed status
      if (dsc.armedChanged[partition]) {
        dsc.armedChanged[partition] = false;  // Resets the partition armed status flag
        if (dsc.armed[partition]) {

          char pushMessage[40] = "System DSC ";
          if (dsc.armedAway[partition]) {
            char armedState[31] = "Włączenie DOMOWE: podsystem ";
            strcat(pushMessage, armedState);
          }
          else if (dsc.armedStay[partition]) {
            char armedState[33] = "Włączenie NORMALNE: podsystem ";
            strcat(pushMessage, armedState);
          }
          char partitionNumber[2];
          itoa(partition + 1, partitionNumber, 10);
          strcat(pushMessage, partitionNumber);
          sendPush(pushMessage);

        }
        else {
          char pushMessage[34] = "System DSC rozbrojony: podsystem ";
          char partitionNumber[2];
          itoa(partition + 1, partitionNumber, 10);
          strcat(pushMessage, partitionNumber);
[... 120 lines cut ...]
  while (pushClient.available()) {
    if (pushClient.read() == ' ') break;
  }

  // Checks the first character of the HTTP status code - the message was sent successfully if the status code
  // begins with "2"
  char statusCode = pushClient.read();

  // Successful, reads the remaining response to clear the client buffer
  if (statusCode == '2') {
    while (pushClient.available()) pushClient.read();
    pushClient.stop();
    return true;
  }

  // Unsuccessful, prints the response to serial to help debug
  else {
    Serial.println(F("Push notification error, response:"));
    Serial.print(statusCode);
    while (pushClient.available()) Serial.print((char)pushClient.read());
    Serial.println();
    pushClient.stop();
    return false;
  }
}`

l33d, 2019-06-11:
Ok i test original sketch on nodemcu connected to my alarm for few days and it seems to be ok. After arm i received only one notification about partition and arm mode. What`s wrong is my translate sketch?

taligentx, 2019-06-13:
Good to hear it's working with the original example sketch.  I recommend changing a few sections at a time to see which of your changes is causing the reset.

l33d, 2019-06-13:
The problem is same as 1.0 sketch. I translate it too, and have this same problem with this same information as i remember. Do you know how can i translate the sketch properly? Can i use my own ascii charset like ą, ę, ł without problem? First i update arduino program, maybe there is a problem when compiling sketch with my charset.

taligentx, 2019-06-13:
Unfortunately I'm not familiar with using different charsets, as far as I've seen the Arduino IDE should support UTF-8 except for the serial terminal which is ASCII only.

taligentx, 2019-06-14:
Check your `char pushMessage[]` arrays, they seem to be too short - should be total number of characters + 1 for the terminating null character.

l33d, 2019-06-14:
Hmm, ok i try thanks for help. If i add more total numbers to array there will be a problem?

taligentx, 2019-06-14:
Great, good to hear.

closed 2019-06-14 by taligentx