taligentx/dscKeybusInterface #86: Is possible to get user id from panel?

UltimatPronin, 2019-06-04:
@taligentx, can we get the user id from panel? Is it possible?
For example, 
1)In panel we program user1 have code 0001, user2 have code 0002.
2)when panel is armed/disarmed access code entered

Can we by entered code or other data determinate which user armed/disarmed panel?
On real keypad entered code 0001 - we get user1, entered code 0002 - we get user2 in sketch.

UltimatPronin, 2019-06-07:
@taligentx, I find all needed parts inside dscKeybusPrintData.cpp
in [Armed by access code] & [Disarmed by access code] sections.

The main idea is to know and maybe log which user armed/disarmed system.

Maybe in future you will add storage of last used access code in variable to simplify usage together with Admed/Disarmed events?

For now I will try to get this codes numbers in my sketch using code parts from dscKeybusPrintData.cpp.

taligentx, 2019-06-13:
As you've seen, the data is available but not currently tracked, I'll look into it.

UltimatPronin, 2019-06-13:
@taligentx ,thank you!
Now I get it like this:
```
String sUserCode_id="";
void setStatus(byte partition) {
...
switch (dsc.status[partition]) {
...
case 0x04: __getAccesCodeID(); writeLog("Armed stay by "+sUserCode_id); break;
...
}
}

void __getAccesCodeID() {
  
  sUserCode_id="";
  bool bOK=false;
  while (!bOK){
    switch (dsc.panelData[0]) {
      //case 0xA5: bOK=true; __getAccesCodeID_0xA5(); break;      
      case 0xEB: bOK=true; __getAccesCodeID_0xEB(); break; //Access Code ID always is getted here 
      default: {      
        break;
      }
    }
    
    dsc.handlePanel();  // Continues processing Keybus data
  }
}

void __getAccesCodeID_0xEB() {  
  switch (dsc.panelData[7]) {
    case 0x00: __getAccesCodeID_0(8); return;
  default: {      
      return;
    }
  }
}

void __getAccesCodeID_0(byte panelByte) {

  bool decoded = true;
  switch (dsc.panelData[panelByte]) {
    case 0x49: break;
    case 0x4A: break;
    case 0x4B: break;
    case 0x4C: break;
    case 0x4D: break;
    case 0x4E: break;
    case 0x4F: break;
    case 0x50: break;
    case 0x51: break;
    case 0x52: break;
    case 0x53: break;
    case 0x54: break;
    case 0x55: break;
    // 0x56 - 0x75: Zone tamper, zones 1-32
    // 0x76 - 0x95: Zone tamper restored, zones 1-32
    case 0x98: break;
    // 0x99 - 0xBD: Armed by access code
    case 0xBE: break;
    case 0xBF: break;
    // 0xC0 - 0xE4: Disarmed by access code
    case 0xE5: break;
    case 0xE6: break;
    case 0xE7: break;
    case 0xE8: break;
    case 0xE9: break;
    case 0xEA: break;
    case 0xEC: break;
    case 0xEF: break;
    case 0xF0: break;
    case 0xF1: break;
    case 0xF4: break;
    case 0xFF: break;
    default: decoded = false;
  }
  if (decoded) return;

  if (dsc.panelData[panelByte] >= 0x99 && dsc.panelData[panelByte] <= 0xBD) {
    byte dscCode = dsc.panelData[panelByte] - 0x98;
    if (dscCode >= 35) dscCode += 5;

  //sUserCode_id+="Armed by ";
  switch (dscCode) {
    case 33: sUserCode_id+="duress "; break;
    case 34: sUserCode_id+="duress "; break;
    case 40: sUserCode_id+="master "; break;
    case 41: sUserCode_id+="supervisor "; break;
    case 42: sUserCode_id+="supervisor "; break;
    default: sUserCode_id+="user "; break;
  }
  sUserCode_id+="code ";
  sUserCode_id+=String(dscCode);   
  
    return;
  }

 
  
  if (dsc.panelData[panelByte] >= 0xC0 && dsc.panelData[panelByte] <= 0xE4) {
    byte dscCode = dsc.panelData[panelByte] - 0xBF;
    if (dscCode >= 35) dscCode += 5;
  //sUserCode_id+="Disarmed by ";
  switch (dscCode) {
    case 33: sUserCode_id+="duress "; break;
    case 34: sUserCode_id+="duress "; break;
    case 40: sUserCode_id+="master "; break;
    case 41: sUserCode_id+="supervisor "; break;
    case 42: sUserCode_id+="supervisor "; break;
    default: sUserCode_id+="user "; break;
  }
  sUserCode_id+="code ";
  sUserCode_id+=String(dscCode); 
  
    return;
  }
}

```

I think this is not the best way, but it works in my current test sketch.

taligentx, 2019-06-13:
Added to the develop branch, check the Status examples to see how it can be used.  Note that the panel sends the access code in a separate message after the armed/disarmed status, so it's handled as a separate status instead of combined with the armed status checking.

closed 2019-06-15 by taligentx