taligentx/dscKeybusInterface #87: A way to stop dscKeybisInterface

feder-ico, 2019-06-04:
Hi there,
Is there is a way to stop dscKeybusInterface?
I mean, like dsc.begin but dsc.stop

I am integrating the code with a different Wifi manager (IoTWebconf instead of WiFiManager) and for some reason (that I don´t know) if dscKeybusInterface is started, it fails to connect to WiFi. 
So I workarounded starting dscKeybusInterface after Wifi connection is stablished.

```
if (iotWebConf.getState() == IOTWEBCONF_STATE_ONLINE) {
                if (!DSCBegined){
                    // Starts the Keybus interface and optionally specifies how to print data.
                    // begin() sets Serial by default and can accept a different stream: begin(Serial1), etc.
                    dsc.begin();
                    DSCBegined = true;
                    Serial.println(F("DSC Keybus Interface is online."));
                }
        doDSC();
      }
```

And then it´s work fine, but if Wifi connection is lost, it will retry connecting with no success because dscKeybusInterface is begined. So I need stop dscKeybusInterface until Wifi connection is restablished and then dsc.begin again.

Can anyone suggest a way to do this?

Thanks in advance for your sugestions.
Best regards.

UltimatPronin, 2019-06-04:
@feder-ico, try to call timer1_detachInterrupt() where you need to stop dsc. #77

feder-ico, 2019-06-04:
Hi, @UltimatPronin ,
Thanks for your answer, I will try.
But please tell me, if I call timer1_detachInterrupt() to stop it, when I need to restart shoul I call timer1_attachInterrupt(dscDataInterrupt) or dsc.begin ?

Thanks again!

UltimatPronin, 2019-06-04:
I think it will be enough to call timer1_attachInterrupt(**dsc.**dscDataInterrupt);

taligentx, 2019-06-05:
@feder-ico I've added info on this in #77, if you need to stop all background activity (both the clock and data interrupts), you can use:

`detachInterrupt(digitalPinToInterrupt(dscClockPin));` 

And then `dsc.begin()` to restart - though I haven't tested this, if you do then feedback would be useful.

jimtng, 2019-06-05:
please add dscKeybusInterface::end() function to reverse what's in begin(), so that we can call it. Thank you.

feder-ico, 2019-06-05:
Thanks @taligentx , I will try and come back with some feedback.
Cheers!

taligentx, 2019-06-07:
Added `dsc.stop()` to the develop branch.

closed 2019-06-08 by taligentx