taligentx/dscKeybusInterface #90: Redirect/replace dscKeybusPrintData stream->print to a buffer

aboulfad, 2019-06-08:
Hi, 
I would like to redirect the dscKeybusReader so I can send the print data to whatever output device: serial, file, IP (MQTT, syslog,...)

I know very little of C++, but I managed to get that stream inherits from print class, and dsc.begin sets stream output to Serial. Is there an easy way to redirect the stream to a buffer that can be accessed by a method to direct the print data to the appropriate tour out device?

What I have done so far is to add a new method and did a global and replace for stream->print with the new method, but realize there must be a cleaner OO way to do this. Any suggestions? Ideally the object/buffer should be accessible in the sketch so the user has control over it and use the “output” of their choice.

I think this would be interesting to add, along with OTA, one can run any example and retrieve detailed logs from the comfort of their sofa. I’d be more than happy to continue my work and create a PR with some help/guidance.

taligentx, 2019-06-08:
Interesting, I doubted this would actually be used but this possibility already exists - check `src/dscKeybusInterface.h`:

```
void begin(Stream &_stream = Serial);             // Initializes the stream output to Serial by default
```

You can pass a different stream to `begin()` - I had implemented this so the stream could be changed to Serial1, Serial2, etc in case of multiple UARTs but in theory it should work with any stream that implements `print`.

aboulfad, 2019-06-08:
Ok but the examples of output devices I mentionned such  as IP based do not implement « print » or maybe I don’t understand what that entails. 

I saw the begin method but stream class inherits only from print so not sure how that can help especially when all the code has “stream->print” everywhere?

taligentx, 2019-06-08:
> the examples of output devices I mentionned such as IP based do not implement « print »

Take a look at `EthernetClient` and `WifiClient`

aboulfad, 2019-06-09:
> Take a look at `EthernetClient` and `WifiClient`

`using Print::write` , `virtual size_t write(uint8_t) = 0`;,...
pure virtual functions, overloading, polymorphism. Much reading and learning and even with some explanation(s) at [stackoverflow](https://arduino.stackexchange.com/questions/38965/help-understanding-printwrite-it-calls-write-where-is-write-defined-h), it sounds this is intermediate to advanced OO which sadly is beyond my pay grade.

taligentx, 2019-06-13:
It's currently setup to spit out bytes to whichever stream interface, so you could initialize as `dsc.begin(Serial1)` to output to Serial1.  If you had a WiFi client as `WiFiClient client`, in theory you could pass `dsc.begin(client)` and it would directly output to the current Wifi connection, though you'd have to setup a connection in advance and setup the receiving side to receive the data.

The way you're looking to use it with the output placed in a buffer could be done, it'd probably be best to make a copy of `dscKeybusPrintData.cpp` to a new file first and have new functions to call it separately.  In the sketch you could create a buffer and pass it to whichever function.

If someone takes this on and submits a PR I'll merge it.

aboulfad, 2019-06-13:
I can't believe it's as easy as redirecting stream to client (I deserve a smack in the head) !  That makes the IP mechanism an easy one to use/develop.

The reason I mentioned the use of a buffer, is to pass that buffer to MQTT, in case one wants to MQTT the detailed logs. That is not as obvious as the generic IP solution above as the code has `stream-> print` everywhere. I already have a copy of the file with those replacements (also using PSTR() ) but then realized that I need to pass back and forth a buffer/arguments/return results from the sketch to dscKeybus*, and got stuck in C++ land ...

I have this function that I use in my other sketches that basically dumps data to either serial or MQTT depending if I am using OTA or not.
```C
void trace( PGM_P format, ... ) {
    va_list args;
    va_start( args, format );
    int len = vsnprintf_P(NULL, 0, format, args) + 1;
    char log_msg[len];
    vsnprintf_P( log_msg, len, format, args );
    va_end( args );
    
    #ifdef USE_ARDUINO_OTA
      mqtt.publish(log_topic, log_msg);
    #else
      Serial.printf_P(log_msg);
    #endif
}
```
I definitely will continue looking at this with your support, and if I succeed in writing something that works, I'll open a PR.

aboulfad, 2019-06-13:
ok, so i'll write down my thoughts/ideas/challenges.

1. [x] Setup MQTT in sketch, as this could be used by other parts of the sketch not related to dsc
2. [x] Define new trace function similar to above, or some flavor of in new version of `dscKeybusPrintData`
3. [x] Replace all instances of `stream->print` with `trace`
4. [x] Store variables in flash instead of SRAM using PSTR()
5. [ ] Pass the Pubsubclient client instance to trace() ???

5a. Stuck on Step5, and unsure if this makes sense, create an instance of Pubsuclient and how to pass it to `dscKeybusPrintData` ? 

5b. Or just create another instance of Pubsubclient in `dscKeybusPrintData` so trace() can use it. 

5c. Looking at above, then I can't see the purpose to create a buffer and pass it to `dscKeybusPrintData` in order to use MQTT ?

Am I looking at this wrong ?

taligentx, 2019-06-13:
Perhaps something like this pseudocode:

* Sketch
    * char buffer[100] (set to at least max message length + 1 for null character)
    * dsc.getPanelMessage(buffer)
    * mqtt.publish(buffer)

* Library
    * dsc.getPanelMessage(char* buffer)
      strcpy(buffer, "panel message")

The actual implementation could and probably would be different.  The library `printPanel*` functions currently return the following as separate messages:  binary with spaces, binary without spaces, command, message.  The library should do minimal formatting, it's up to the sketch to take the messages and format them as needed for the intended use.

aboulfad, 2019-06-13:
Ah I see, so `printPanel*` would fill the buffer, which would be accessible by the sketch as it's defined there. In a way I had it in my trace function, `log_msg`!

Regarding formatting, there are few stream->print in sequence in `dscKeybusPrintData .cpp` that need to be concatenated, so one would have to do a strcpy followed by strcat,... If we use my trace function, then we can use standard C printf like minimal formatting and call just once that function.

The trace function would only return a formatted buffer without doing MQTT/serial. Is that acceptable ? 
Example:
```c
void dscKeybusInterface::printPanelStatus3(byte panelByte) {
  stream->print(F("Unrecognized data"));
  stream->print(F(" :"));
  stream->print(panelByte, HEX);
}
```

becomes

```c
void dscKeybusInterface::printPanelStatus3(byte panelByte) {
  trace(PSTR("Unrecognized data : 0x%02X"), panelByte);
}
```

Now all this would live in another `dscKeybusPrintData.cpp` file (`dscKeybusSendData.cpp` ?) you'd add to your distro ? and if a user wants to use that new logging capability, they would remove the `dsc/begin()` and just call the new function and `mqtt.publish` (for mqtt logging)?

Sorry for the many questions, I have never added code into someone else's repo ! (we are not there yet but thinking ahead)

aboulfad, 2019-06-13:
> ... If you had a WiFi client as `WiFiClient client`, in theory you could pass `dsc.begin(client)` and it would directly output to the current Wifi connection, though you'd have to setup a connection in advance and setup the receiving side to receive the data.

And in practice it works, I can't believe it was that simple :smiley: Sketch [KeybusReader.ino](https://github.com/taligentx/dscKeybusInterface/blob/master/examples/esp8266/KeybusReader/KeybusReader.ino) required few changes and the proof of concept works !

```
nc dscip.local 80

nc 30.85.1.200 80
Client connected to dscKeybusReader
   18.42: 00000101 0 10000001 00000001 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Ready Backlight - Partition ready | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
   21.43: 11100110 0 00101100 00010000 00000000 00000000 00000000 00000000 00100010 [0xE6] Partition 5 | Enabled zones 33-64: none
   27.44: 00010001 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0x11] Keypad slot query
   27.44: 11111111 1 ........ ........ 11111111 11110011 11111111 11111111 ........ 
   36.45: 00100111 0 10000001 00000001 00000000 11000111 00000100 01110100 [0x27] Partition 1: Ready Backlight - Partition ready | Partition 2: disabled | Zones 1-8 open: 3 
   39.45: 00000101 0 10000001 00000010 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Ready Backlight - Stay/away zones open | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
   42.46: 00100111 0 10000001 00000010 00000000 11000111 00000000 01110001 [0x27] Partition 1: Ready Backlight - Stay/away zones open | Partition 2: disabled | Zones 1-8 open: none
   45.47: 00000101 0 10000001 00000001 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Ready Backlight - Partition ready | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled

```

aboulfad, 2019-06-13:
> ... If you had a WiFi client as `WiFiClient client`, in theory you could pass `dsc.begin(client)` and it would directly output to the current Wifi connection, though you'd have to setup a connection in advance and setup the receiving side to receive the data.

And in practice it works, I can't believe it was that simple :smiley: Sketch [KeybusReader.ino](https://github.com/taligentx/dscKeybusInterface/blob/master/examples/esp8266/KeybusReader/KeybusReader.ino) required few changes and the proof of concept works !

```
nc dscip.local 80

Client connected to dscKeybusReader
   18.42: 00000101 0 10000001 00000001 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Ready Backlight - Partition ready | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
   21.43: 11100110 0 00101100 00010000 00000000 00000000 00000000 00000000 00100010 [0xE6] Partition 5 | Enabled zones 33-64: none
   27.44: 00010001 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0x11] Keypad slot query
   27.44: 11111111 1 ........ ........ 11111111 11110011 11111111 11111111 ........ 
   36.45: 00100111 0 10000001 00000001 00000000 11000111 00000100 01110100 [0x27] Partition 1: Ready Backlight - Partition ready | Partition 2: disabled | Zones 1-8 open: 3 
   39.45: 00000101 0 10000001 00000010 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Ready Backlight - Stay/away zones open | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
   42.46: 00100111 0 10000001 00000010 00000000 11000111 00000000 01110001 [0x27] Partition 1: Ready Backlight - Stay/away zones open | Partition 2: disabled | Zones 1-8 open: none
   45.47: 00000101 0 10000001 00000001 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Ready Backlight - Partition ready | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled

I'll upload/share the modified sketch somewhere.
```

aboulfad, 2019-06-14:
ok, [KeybusReaderIP.ino](https://github.com/aboulfad/dscKeybusInterface/blob/master/examples/esp8266/KeybusReaderIP/KeybusReaderIP.ino) uploaded to a fork of your master under esp8266.

Its basically an IP clone of your KeybusReader serial sketch (based on master, so no dsc.stop() ):

1. sends all output to a TCP established connection (only when a client connects)
2. listens for commands from the client for virtual keyboard.
3. supports OTA

I've tested few commands (s, c, w,...) and some more complex ones such as *6,1234, but unsure how the panel should behave when it receive the *6 commands, I didnt see anything happen on the panel but I see stuff in the dsc logs.

Overall, i think it's pretty useful debugging tool to have both IP and Serial capabilities. Ideally I'd like to have one sketch for all stream types, but i think its ok for now (need to move on to the HAP-Node stuff)!
 If you want to include that sketch in your distro, I can create a PR.
[alarmLogs.log](https://github.com/taligentx/dscKeybusInterface/files/3290641/alarmLogs.log)

taligentx, 2019-06-14:
Nice, that's really great to see!  One possible source of buffer overflow:  `handlePanel()` isn't being called when the client is disconnected - the interface is still running in the background because it's interrupt-based and continually filling the buffer.  This will overflow if `handlePanel()` isn't being called to process the buffer.  By the way I'll likely be renaming `dsc.handlePanel()` to `dsc.loop()` to make it more clear that it should be run on every loop iteration.

Also the delay shouldn't be necessary, if there's a problem requiring a delay it usually means some kind of logic issue.  If the buffer overflows can be resolved make a PR for the develop branch and I can merge.  Thanks for digging into this!

aboulfad, 2019-06-14:
ok, its been running for 15mns without a buffer overflow. I added `dsc.handlePanel()` in the main loop and removed the delay, WIficlient is blocking and the loop is constantly polling for client connections without a delay, usually not great practice, but it works... PR coming shortly.

taligentx, 2019-06-14:
Sounds good - for the PR, please remove the OTA code.  This is great for personal use but the example sketches should be as minimal as possible to demonstrate how to use one particular feature.

aboulfad, 2019-06-14:
oops PR already done ... let me see if I can delete the PR and issue a new one without OTA, although I think the IP version is kinda a bit lonely if you can't do OTA.

taligentx, 2019-06-15:
Merged #95 - thank you!

aboulfad, 2019-06-15:
Oh I still wanted to do the MQTT version, but that’s fine, for the purposes of a debugging tool, I guess one IP mechanism suffices. I’ll focus my energy on issue #92 . Thank you for your support.

taligentx, 2019-06-15:
Ah apologies, didn't mean to cut it short.  I'll leave the issue open.

aboulfad, 2019-06-15:
Thanks if I don’t come up with something in a week I’ll close it.

taligentx, 2019-06-20:
Heads-up, there's an issue in `KeybusReaderIP` with the `if` loop handling such that the sketch will not show keypad keys until there's a panel command at the same time, take a look at the original `KeybusReader` sketch.

aboulfad, 2019-06-20:
Is it the darn `{ }`, I think I had one wrong ?

taligentx, 2019-06-20:
I believe so, I only did a quick check but that seemed to be the problem.

aboulfad, 2019-06-20:
ok, i fixed/tested it, though I didn't notice anything different before because my alarm panel is busy spewing stuff. I made sure there was no commands, and then used the physical keypad, commands were displayed in near real time. PR #97

taligentx, 2019-06-20:
Great!  Thanks for the PR, merged.

aboulfad, 2019-06-21:
ok I gave up, can't figure out one common solution (Serial, IP, MQTT), so one IP version for dscKeybusReader is enough.

closed 2019-06-21 by aboulfad