taligentx/dscKeybusInterface #91: mqttPartitionTopic receives multiple "D" events

aboulfad, 2019-06-11:
Hi,

running the esp MQTT example, I have noticed whenever the motion detectors trigger, the mqttPartitionTopic receives multiple D events (I think 4 D's per trigger).  Unsure if this is the intended behaviour of the panel, dsc software or MQTT sketch ?

PS: this is from closed issue #89

```
mosquitto_sub -t home/dsc/Get/Zone3
0
1
0

1
0
1
0

1
0
```

```
mosquitto_sub  -t home/dsc/Get/Partition1
D
D
D
D
D
D
D

D
D
D
D
D
D
D
D
D

D
D
D
D

D
D
D
D

D

D
D
D
D
```

aboulfad, 2019-06-13:
ok this is really weird, up until now the above behaviour persisted, until this morning, `mqttPartitionTopic` does not receive those spurious "D" events. Don't really know what changed...

taligentx, 2019-06-13:
I tracked this down, it wasn't an issue until adding `dsc.getStatus()` (now renamed to `dsc.resetStatus()`) - this turned on the flag for `exitDelay` which wasn't being handled in the Homebridge sketch.

The Homebridge sketch doesn't need to be handling exit delay as HomeKit doesn't seem to use that status so I've updated the example sketches in the develop branch and also refined some of the partition state processing.

aboulfad, 2019-06-13:
i'll check that I didn't cherry-pick more stuff than needed when I tested dsc.stop(). Maybe I should just pull `develop` branch and work in it (on my fork) ? 

So far, I have been working off the latest master and `cherry-pick` some commits/changes manually.

taligentx, 2019-06-13:
Feedback on the develop branch would be handy to catch any issues before I release it.

I'm also looking at including a new example sketch for HomeKit using https://github.com/hobbyquaker/homekit2mqtt because homebridge-mqttthing doesn't seem to handle changing status from physical keypads (HomeKit keeps trying to go back to whatever state was last set in HomeKit).

aboulfad, 2019-06-13:
Sure, I'll switch to develop and cross fingers ! I haven't gotten to the Homebridge stuff yet, but I quickly tried it, and found a [bug](https://github.com/arachnetech/homebridge-mqttthing/issues/126)

I may stick to using HAP-Node, as all my accessories were developed on HAP-Node without HB... it may be a bit more coding than using HB, but you have complete control.

aboulfad, 2019-06-14:
I switched to develop branch and this issue is solved. thank you!

closed 2019-06-14 by aboulfad