taligentx/dscKeybusInterface #92: HomeKit Security System Support

aboulfad, 2019-06-14:
Hi,
this is to capture our discussion and progress on how to properly update HomeKit [SecuritySystem Service](https://github.com/KhaosT/HAP-NodeJS/blob/d0f92c202e312717829b76c3803c86aa04fa8dc0/lib/gen/HomeKitTypes.js#L3277) when a change happens in the Alarm system (accessory to Homekit)

To prove and discover how it is supposed to work, I coded a simple HAP-Node script, that adds the SS service/acc and listens on MQTT for commands.

In order to get HomeKit SS icon to go from `OFF` to `Arming` then to `Home`, the sketches in this library have to send over to MQTT:

**Solution 1**:  The "exit in progress" (not done) and then "SA" (done) as they happen on the Security Panel  to `mqttPartitionTopic`. (In the included [short video](https://www.dropbox.com/s/4ghgd3hzdhzy9yg/HomeKit%20SecuritySystem.mov?dl=0), I simulated this library sketches by sending `EX` for `Exit delay in progress` and `SA` for `STAY ARM` using mosquitto_pub.)

THEN the corresponding HomeKit script (HAP-Node, HomeBridge, MQTTThing, ...) has to handle the exit and different arming types by updating the respective SecuritySystem service characteristics `[SecuritySystemCurrentState, SecuritySystemTargetState]` (Issue [#126](https://github.com/arachnetech/homebridge-mqttthing/issues/126) )

**Solution 2** OR this library sketch publishes `SA` to two different MQTT topics `getTargetState` then to `getCurrentState`, first one corresponding to `exit delay in progress`, the second to indicate system is armed. The HomeKit script then updates the appropriate characteristics.

Depending on the chosen solution, the https://github.com/taligentx/dscKeybusInterface/blob/master/examples/esp8266/Homebridge-MQTT/Homebridge-MQTT.ino needs to send some payload to indicate there is an `exit in progress`.  Or publish SA to those two homekit topics...

taligentx, 2019-06-15:
At the moment until this is handled correctly in the HomeKit interfaces there's not much else for the sketch to do.  The actual implementation would depend on how each HomeKit interface chooses to address this, if at all.

As it stands right now, [homekit2mqtt](https://github.com/hobbyquaker/homekit2mqtt) is closest to the desired behavior - the HomeKit state is tracked whether inside HomeKit or changed externally.  Downsides are the DSC sketch having to subscribe to multiple topics to get the desired state for each partition (hobbyquaker/homekit2mqtt#152) and lack of exit delay notice, but seems to work well for core functionality and ease of installation (compared to installing homebridge and then homebridge-mqttthing separately).

homebridge-mqttthing seems it's setup to work well (being able to customize the MQTT payloads, separate topics for target and current state, etc) if the current bugs can be worked out.

Custom scripts are also valuable but it can also be a challenge to keep these up to date as the HomeKit framework evolves.  The nice part is that updating the DSC example sketches is very simple so it's just a matter of having something to work with.

aboulfad, 2019-06-15:
Not sure if I follow, but the current dsc MQTT HB sketch send `SA` once to mqttPartitionTopic, and as a consequence there is no way for any HomeKit framework to handle proper transition from Off->Arming->Armed in conjunction with the current sketches ? It's possible we are saying the same thing.

aboulfad, 2019-06-15:
Great progress and thanks to @tobekas in issue [#126](https://github.com/arachnetech/homebridge-mqttthing/issues/126#issuecomment-502399314), I was able to get the proper Homekit icons to update accordingly.

Though an update is required here in the Homebridge-MQTT sketch, basically when `exit delay` happens to send payload `1S` to mqttPartitionTopic, followed by `SA` when system is armed. The Status sketch has some details: `dsc.exitDelayChanged` and `dsc.exitDelay` that should help.

aboulfad, 2019-06-16:
ok knowing now how to put HomeKit SS service in the right state, I just added the exitDelay checking to Homebridge-MQTT sketch, and now I have the expected transition `Off->Arming->Armed` when state changes are triggered from the Alarm system. Cool ;-)

I have never used the Night Arm, as such I don't understand that code segment in the sketch, in my custom version, I've removed the NA until I get it. There are other scenarios to check, entry delay, Night Arm, but its just a matter of understanding the workflow initially, then modifying the code is very easy.

I am very impressed how well this library works, kudos to the great work you have done @taligentx 🙇 

```
      // Checks exit delay status
      if (dsc.exitDelayChanged[partition]) {
        dsc.exitDelayChanged[partition] = false;
        if (dsc.exitDelay[partition]) {
          mqtt.publish(publishTopic, "1S", true);
        }
        else if (!dsc.armed[partition]) // Checks for disarm during exit delay
          mqtt.publish(publishTopic, "D", true);
      }
```

taligentx, 2019-06-17:
Updated `Homebridge-MQTT`, this should retain the existing Homebridge configuration and implement target states.

Also, thanks!  Good to hear the library is working out.

aboulfad, 2019-06-17:
Great, I’ll spend my time testing your latest commit to develop, I was struggling with the logic to address some uses cases for targetState. I couldn’t figure out a clean way to know the system is disarmed. 

The way it’s implemented in the sketch, is with “else” without checking for a condition. I was searching for when the system is disarmed if there is some Boolean like the other states (ArmedAway, ...) but I couldn’t find any. From the surface, when the disarm msg is received a bunch of other states are updated. Maybe not needed as I don’t understand much the library.

aboulfad, 2019-06-17:
Quick question, do you know if there’s a code to disable the sirene? If not I’ll just remove the wires.

aboulfad, 2019-06-17:
Bug #1: 'SA' on RC or keypad, 1A published instead of 1S to topic
Maybe has to do with this ?
https://github.com/taligentx/dscKeybusInterface/blob/d53a042a37cde83241a7b77b96387c28165d2ceb/examples/esp8266/Homebridge-MQTT/Homebridge-MQTT.ino#L288-L292

aboulfad, 2019-06-17:
Unsupported Use Cases: (will add to this list)

UUC-1: Zone Open, User arms Security System (SS) in Homekit, Icon stuck in `Arming` as there is no feedback from SS that it did not arm because of an open Zone.

aboulfad, 2019-06-17:
**Tested Use Cases**:

RC/Panel triggers:
Check that Homekit reports correct state
- [X] Away Arm
- [X] Disarm during exit delay 
- [x] Disarm after system armed

taligentx, 2019-06-17:
> Bug-1: 'SA' on RC or keypad, 1A published instead of 1S to topic

This is effectively a placeholder to let HomeKit know the system is about to be armed, HomeKit doesn't differentiate between "Arming..." for away or stay, so this this be inconsequential.  If the user is arming by pressing the access code at the keypad, the panel also does not know if the armed state will be away or state until the exit delay is complete and if an away zone is opened during the delay.

> Zone Open, User arms Security System (SS) in Homekit, Icon stuck in Arming as there is no feedback from SS that it did not arm because of an open Zone.

Good catch, might be able to set something up so if the panel isn't ready and a request comes in to arm, the sketch changes the HomeKit target state to disarmed.

taligentx, 2019-06-17:
> This is effectively a placeholder to let HomeKit know the system is about to be armed, HomeKit doesn't differentiate between "Arming..." for away or stay, so this this be inconsequential.

I'll take a second look at this later since HomeKit does show which state it is expecting to end up in.

aboulfad, 2019-06-17:
I am confused ... when I press either the RC or panel dedicated buttons for Stay or Away, check the logs... the panel does know, or my panel !!!

```
 8508.25: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
 8508.25: 11111111 1 10101111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Arm stay
 8508.39: 11100110 0 00011000 00000001 00000000 00000000 00000000 00000000 00000000 11111111 [0xE6] Partition 1 | Status lights flashing: none | Zones 33-64 flashing: none
 8508.43: 10111011 0 00000000 00000000 10111011 [0xBB] Bell: off
 8508.52: 11100110 0 00011010 01000000 00000000 00000000 00000001 00000001 00000000 00000000 01000010 [0xE6] 0x1A: Unrecognized data
 8508.53: 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 10000000 [Keypad] Unrecognized key: 0xFF
 8508.55: 01110101 0 00000000 01110101 [0x75] Partition 1 | Beep pattern: off
 8508.62: 11001110 0 00000001 00010000 00000000 00000000 00000000 11011111 [0xCE] Unrecognized data
 8508.68: 11001110 0 01000000 11111111 11111111 11111111 11111111 00001010 [0xCE] Partition 1,2 activity
 8508.76: 00000101 0 10001011 00001000 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Ready Armed Bypass Backlight - Exit delay in progress | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
```

```
8530.00: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
 8530.01: 11111111 1 10110001 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Arm away
 8530.15: 11100110 0 00011000 00000001 00000000 00000000 00000000 00000000 00000000 11111111 [0xE6] Partition 1 | Status lights flashing: none | Zones 33-64 flashing: none
 8530.19: 10111011 0 00000000 00000000 10111011 [0xBB] Bell: off
 8530.27: 11100110 0 00011010 01000000 00000000 00000000 00000001 00000001 00000000 00000000 01000010 [0xE6] 0x1A: Unrecognized data
 8530.28: 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 10000000 [Keypad] Unrecognized key: 0xFF
 8530.31: 01110101 0 00000000 01110101 [0x75] Partition 1 | Beep pattern: off
 8530.37: 11001110 0 00000001 00010000 00000000 00000000 00000000 11011111 [0xCE] Unrecognized data
 8530.43: 11001110 0 01000000 11111111 11111111 11111111 11111111 00001010 [0xCE] Partition 1,2 activity
 8530.46: 01110101 0 00010001 10000110 [0x75] Partition 1 | Beep pattern: single beep (exit delay)
 8530.54: 00000101 0 10000011 00001000 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Ready Armed Backlight - Exit delay in progress | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
```

taligentx, 2019-06-19:
Reworked the sketch a bit to handle the target states during exit delay and when the panel isn't ready.

If the system is armed externally (keypad), HomeKit will still default to showing "Away" during the exit delay and will show the correct state when the panel arms.  The interface can see if a keypad uses the "Stay" or "Away" function key, but the panel doesn't send any messages during the exit delay indicating the arming mode.

At some point it's probably possible to support the `*1` function to switch between "Stay" and "Away" armed while armed, added to the todo list.

aboulfad, 2019-06-19:
Thanks, I'll give it a try. So the interface clearly shows what was pressed and actually displays the right state, wether Stay or Away when arming the system... So I am at a loss why that information is not relevant during exit delay? clearly i don't understand some basic notion.

taligentx, 2019-06-19:
> So the interface clearly shows what was pressed and actually displays the right state, wether Stay or Away when arming the system

Check KeybusReader, this interface can only see that a physical keypad pressed a function key but there are no messages from the panel stating how it will arm.  It would require more memory for the library to process status states based on keypad keys and I don't think it's worth it right now - the only impact right now is cosmetic when the user arms the system from a physical keypad stay/away function key and then opens the Home app and watches the status during the exit delay.

aboulfad, 2019-06-19:
If I follow the logic implemented/described, the Security System(SS) is only deemed to be in a specific state ONLY when the SS send a message indicating the state ? (obviously after the exit delay)

So the user initiated action is kinda irrelevant in the implementation ?

I am seeking clarification, not trying to influence anything. In my simple mind, if a user pressed a key to arm `State x`, I expected the library to take account of that information and relay that to the "application. 

Currently, an invalid transient event is being sent, which could render other integration more troublesome. I am definitely not sitting and watching what happens to my Home app icons ;-)

Anyways, enough on this item, I got my answer.

taligentx, 2019-06-20:
> Security System(SS) is only deemed to be in a specific state ONLY when the SS send a message indicating the state ? (obviously after the exit delay)

Correct, any physical keypad can send any keys at any time, but only the panel itself can definitively indicate that the key it received is valid and that it is in a particular state.  As far as I've seen, the panel doesn't send any feedback during the exit delay about which armed state it will end up in. The LCD PK5500 keypad also does not show which armed state it will enter if you press a function key - it will only show "Exit delay in progress".

In theory the library could process keypad function key data and then compare it to the panel state to predict the armed state, but processing keypad data would double the processing code and require its own data buffer - so it's question of how to spend the limited resources on the microcontrollers to provide information that the panel and keypads themselves doesn't bother to provide.

>So the user initiated action is kinda irrelevant in the implementation ?

Most user actions are confirmed by the panel - for example pressing any of the panic keys.  Apparently the keypad function key stay/away keys are not.

> Currently, an invalid transient event is being sent, which could render other integration more troublesome.

In general I agree it's better to commit errors of omission, but in this particular case is there a scenario where placing HomeKit in the "Arming..." state for `Away` instead of `Home` or `Night` and then setting to the actual state when armed causes an issue?

taligentx, 2019-06-20:
Also keep in mind the example sketches are just that - examples of what the interface can do and a starting point for users to customize to their needs.  So if the exit delay does cause an issue, it's one line to comment out.

As far as things to check out in HomeKit, I've been curious to see if there's a way to utilize the Keybus connected status to indicate in HomeKit if the security system is offline or has a trouble status.  So far it seems that only the `Tamper` characteristic has any effect (and only shows the tamper status when you go into the SecuritySystem object's settings.  Same with zone faults/tamper, etc.

aboulfad, 2019-06-20:
> Also keep in mind the example sketches are just that - examples ...

Indeed, take all my questions/comments as attempts to understand things :)

> As far as things to check out in HomeKit, I've been curious to see if there's a way to utilize the Keybus connected status to indicate in HomeKit if the security system is offline or has a trouble status. ...

Oh, there is a way, that I've been using from my lengthy discussion in HAP-Node issue [556](https://github.com/KhaosT/HAP-NodeJS/pull/556), basically it will display "No Response" on the icon.

The SecuritySystem [characteristics](https://github.com/KhaosT/HAP-NodeJS/blob/a42102f5db0388b67f28974ca5fa489933ca1be6/lib/gen/HomeKitTypes.js#L3278-L3293) you mentioned do not show anything on the icon, with the exception of `StatusLowBattery` which could be used to indicate low backup battery for the SS BUT for some reason Apple did not include that for SS... 

Homekit/Home app is fun but its lacking in many aspects... Eve app is one that supports history and all other charac (fault, tampered,...)

aboulfad, 2019-06-20:
> In general I agree it's better to commit errors of omission, but in this particular case is there a scenario where placing HomeKit in the "Arming..." state for `Away` instead of `Home` or `Night` and then setting to the actual state when armed causes an issue?

One user facing issue I can think of: `Notifications`. You click on keypad for one state, and you get on your phone two notifications, first is the transient invalid one, then after exit delay the correct one. 

Though admittedly, if we have all this working, it's to avoid using the keypad 😃 so maybe we can put this issue to bed ? (unless there's an easy solution...)

Ok, I have pulled 0a129af and will test the new MQTT HB sketch and get back.

aboulfad, 2019-06-20:
- [X] UUC-1: HomeKit stuck in Arming 👍 
Though I notice MQTTThing is sending `1D` to `dsc/Set` which in theory is harmless, can't understand why ...

```
 EventedHTTPServer [iPhone] HTTP request: /characteristics +7s
  HAPServer [AA:BB:11:FF:AA:DD] HAP Request: PUT /characteristics +0ms
  Accessory [Homebridge] Processing characteristic set: [{"aid":2,"iid":11,"value":1}] +0ms
  Accessory [Homebridge] Setting Characteristic "Security System Target State" to value 1 +0ms
[2019-06-20, 2:22:13 p.m.] [Alarm Homebridge] Publishing MQTT: home/dsc/Set = 1A
  EventedHTTPServer [Apple TV] Sending HTTP event '2.11' with data: {"characteristics":[{"aid":2,"iid":11,"value":1}]} +0ms
  EventedHTTPServer [iPhone] HTTP Response is finished +1ms
[2019-06-20, 2:22:13 p.m.] [Alarm Homebridge] Received MQTT: home/dsc/Get/Partition1 = 1D
[2019-06-20, 2:22:13 p.m.] [Alarm Homebridge] State sectar is now: 3
  EventedHTTPServer [Apple TV] Sending HTTP event '2.11' with data: {"characteristics":[{"aid":2,"iid":11,"value":3}]} +0ms
[2019-06-20, 2:22:13 p.m.] [Alarm Homebridge] Received MQTT: home/dsc/Get/Partition1 = D
  EventedHTTPServer [iPhone] HTTP request: /characteristics +300ms
  HAPServer [AA:BB:11:FF:AA:DD] HAP Request: PUT /characteristics +0ms
  Accessory [Homebridge] Processing characteristic set: [{"aid":2,"iid":11,"value":3}] +0ms
  Accessory [Homebridge] Setting Characteristic "Security System Target State" to value 3 +0ms
[2019-06-20, 2:22:13 p.m.] [Alarm Homebridge] Publishing MQTT: home/dsc/Set = 1D
  EventedHTTPServer [iPhone] HTTP Response is finished +1ms
```

aboulfad, 2019-06-20:
> Bug List:
> 
> * **Bug-1**: 'SA' on RC or keypad, 1A published instead of 1S to topic
ok w the new commit, although we agree about those transient events, the sketch sends after the exit delay another random event, then the actual armed state.

The good news, it seems notifications are only for the final state update. But there are too many state changes being sent than is needed. I didn't look at the sketch, too many changes.
```
mosquitto_sub -t home/dsc/Get/Partition1
1S
... Exit delay in progress
1S
SA

1A
... Exit delay in progress
1S
SA
```

aboulfad, 2019-06-21:
**HomeKit Security System behaviour**
(quick summary to optimize the HB MQTT sketch publish actions)

Basically, `SecuritySystemCurrentState` **and/or** `SecuritySystemTargetState` are updated based on the following conditions:

1. IF an action (ARM, DISARM) originated from HK, HK sets `SecuritySystemTargetState` and expects SS to set `SecuritySystemCurrentState`
2. IF an action (ARM, DISARM) originated from SS, SS should set both states

NOTE: _Given that MQTT HB sketch uses same topic for `getTargetState` and `getCurrentState`, different values for Target state [1S,1A,...] are needed to set the proper HK states._

---
**Homekit to SecuritySystem**
```
Legend: 
- 1S/T to Set: Update `SecuritySystemTargetState` with value `1S` and send to topic `setTargetState`
- (Home): Indicates Homekit current state `Home`
- Home: Homekit action to set SS to `Home`
```
[image]
---
**SecuritySystem to Homekit**
[image]

taligentx, 2019-06-21:
> Oh, there is a way, that I've been using from my lengthy discussion in HAP-Node issue 556, basically it will display "No Response" on the icon.

Sounds good, that could tie into `KeybusConnected`.

>  on a sidenote, I am almost done with V1 of a HAP-Node script for Security System. very crude but works, and have complete control.

Great!  That'd be interesting to check out, there are a few aspects of HomeKit support that could use refinement.  Off the top of my head:

* Siri currently times out when arming the security system
* Disarming through Siri should ideally have an authentication option - biometric/passcode required.  I've seen official HomeKit accessories with this, though I'm not sure if the publicly available APIs support this behavior.

> The good news, HomeKit notifies only when Armed and Disarmed and only when both Current and Target states are correct. But there are too many state changes being sent than is needed.

This is because there is a split between the MQTT request (example: arm away will set Away target state) and the actual state being reported by the panel (armed away will set both target and current state).  From the sketch's point of view, even if it knows that it set the initial target state, it can't know absolutely that some other target state wasn't set externally in the time between the initial request and when the panel finally arms.  So whenever the panel definitively reaches an armed state, it will set both the target and current state.

> HomeKit Security System behavior
> (quick summary to optimize the HB MQTT sketch publish actions)

Thanks, that looks like a good overview of the state processing.  Handling `*1` to switch between stay and away arming is the remaining refinement for the Home app, while Siri could still use some refinement as I previously mentioned.

It's great to see that you're digging into this, my focus is normally on the library itself and the front-end sketches are out there to give a starting point for others to come up with interesting uses.

aboulfad, 2019-06-21:
> Sounds good, that could tie into `KeybusConnected`.
>
You could tie to anything, I saw from the comments that  `KeybusConnected` is meant to detect if data is detected on the keybus, how do you check ? (this can also tie into MQTTThing `getOnline` )

> Great! That'd be interesting to check out, there are a few aspects of HomeKit support that could use refinement. Off the top of my head:
> 
> * Siri currently times out when arming the security system

Yep, check the [issue](https://github.com/arachnetech/homebridge-mqttthing/issues/131) I opened in the other repo. Not sure if there's a solution, I feel it's a bug in Apple's Siri's HK SS implementation. A nice workaround is to setup scenes with the target state desired (which works). I could share a gist but it is currently barebone, no error checking, mqtt connection checks, ... it works well so far, I haven't been using HB/MQTTthing.

> * Disarming through Siri should ideally have an authentication option - biometric/passcode required.  I've seen official HomeKit accessories with this, though I'm not sure if the publicly available APIs support this behavior.

I see nothing in HAP-Node to that effect, it's possible the commercial and public support for HAP differ. BUT, when my iphoneX is locked, if I ask Siri "Turn SS off", she replies that I need to unlock the phone first. So there is some authentication mechanism !

> ... So whenever the panel definitively reaches an armed state, it will set both the target and current state.

Gotcha.
 
> It's great to see that you're digging into this, my focus is normally on the library itself and the front-end sketches are out there to give a starting point for others to come up with interesting uses.

I wanted to spend time on the MQTT HB sketches but you did, so Thank you 😃

aboulfad, 2019-06-21:
**UUC-2**: SS Automation - Leave Home, Arm Away.
[image]

Seriously Apple, I need to confirm to run this automation when I am leaving, and ditto for arriving. That implies the user has to look/handle the phone. How useful is that if someone is leaving their Home in their car ???

That makes the automation close to useless, at least for my usage ...

aboulfad, 2019-06-22:
Ah the ingenuity of the collective comes to the rescue. A not so elegant workaround to the automation confirmation for SS, Door Lock and Garage Door openers is to create a dummy switch in HAP-Node/HB, and use that to trigger the SS without any confirmation to run the automation. 🚀 

Two lines in HAP-Node and we are in business:
```javascript
// add dummy switch as workaround for SS Automation confirmation
var dswitch = new Service.Switch("Dummy Switch");
acc.addService(dswitch);
```

It clutters the automation rules, as instead of two rules (Leave, Arrive), now there's four:
```
R1: Leave, Dummy switch ON
R1: If Switch ON, Set SS to (HOME, ...)
R3: Arrive, Dummy switch OFF
R4: If switch is OFF, Turn SS OFF
```

aboulfad, 2019-06-22:
Since we are on the topic of workarounds, I thought why not extend the usage of on/off HAP accessories (switch) to notify Homekit of various trouble in the alarm system!

- General trouble (Siren disconnected), AC power, Low battery,...

Basically, the MQTT sketch would publish an true/false to additional zones (outside the supported range?) , and in HB you add those zones as switches.

In my HAP-Node implementation, I'll also use that concept to implement panic/fire switches, so I can trigger those actions from Homekit (requires modification to `mqttCallback`) and could also be done with MQTTThing.

aboulfad, 2019-06-23:
I think we managed to tackle some HomeKit use cases. Thanks for your support!

taligentx, 2019-06-23:
> Basically, the MQTT sketch would publish an true/false to additional zones (outside the supported range?) or any topic and in HB you add those zones as switches.

Sure, for example the `HomeAssistant-MQTT` sketch publishes to the `dsc/Get/Trouble` for panel trouble and `dsc/Status` for online/offline status, this could certainly be done in the sketch for HomeKit.

It'd be great if there was an Arduino library that replicates [maximkulkin/esp-homekit](https://github.com/maximkulkin/esp-homekit) to be able to create standalone HomeKit devices without an intermediary like HAP-Node.  There seems to be quite bit of interest though the developer is focused on `esp-open-rtos` so it'll need to be a new project.

Though it could be interesting to see what would be involved in porting dscKeybusInterface to esp-open-rtos - far less generally useful than porting esp-homekit to Arduino but an interesting learning exercise.

aboulfad, 2019-06-23:
> It'd be great if there was an Arduino library that replicates [maximkulkin/esp-homekit](https://github.com/maximkulkin/esp-homekit) to be able to create standalone HomeKit devices without an intermediary like HAP-Node. 

I don't think there's another one but maybe it will change in the future! Just tested today live my automation for my alarm with the tricks mentioned above, and it works nice minus the fact that geofencing is minimum 100m !

aboulfad, 2019-06-24:
> The SecuritySystem [characteristics](https://github.com/KhaosT/HAP-NodeJS/blob/a42102f5db0388b67f28974ca5fa489933ca1be6/lib/gen/HomeKitTypes.js#L3278-L3293) you mentioned do not show anything on the [...]

Learning never stops ... LinkedServices !!! So you can link the `Service.BatteryService` to  other services!!! It supports `BatteryLevel`, `ChargingState` & `StatusLowBattery`.

taligentx, 2019-06-28:
> Learning never stops ... LinkedServices !!! So you can link the Service.BatteryService to other services!!! It supports BatteryLevel, ChargingState & StatusLowBattery.

Nice!  If you're interested in poking around with HomeKit characteristics further, keep eye out for a new repo.  I was curious to see exactly what it would take to port the library to RTOS and surprisingly it's been going well, I have data capture working and was able to successfully test out [esp-homekit](https://github.com/maximkulkin/esp-homekit) using a couple of contact sensors for two zones.  So a fully integrated HomeKit interface on esp8266/esp32 may be possible.

There currently isn't an example in `esp-homekit` for the security system object but for now I'm focusing on getting the library itself up to basic functionality (working on handling writes to the panel at the moment).

taligentx, 2019-07-06:
> ...this interface can only see that a physical keypad pressed a function key but there are no messages from the panel stating how it will arm.

While working on the RTOS port, I happened to glance over at the keypads and noticed that there is a very obvious way to tell how the panel will arm at the end of the exit delay.  The `Bypass` light!

I totally missed this basic indicator - I've added this into the RTOS HomeKit example and the exit states are now correctly reported during the "Arming..." phase.  The exit state will even change from "Home" to "Away" if a zone is opened during the exit delay as the panel updates its state from stay arm to away arm.  I'll add this to the Arduino library as well after seeing how it works out on RTOS.

aboulfad, 2019-07-06:
> 
> While working on the RTOS port, I happened to glance over at the keypads and noticed that there is a very obvious way to tell how the panel will arm at the end of the exit delay. The `Bypass` light!
> 

Cool, except what is this Bypass light? I’ll check the RTOS sketch examples for curiousety. FYI, the library has been up and running for weeks now (develop branch) 👍

taligentx, 2019-07-25:
> Cool, except what is this Bypass light? 

It's an indicator on non-LCD keypads indicating if there will be zones bypassed.  The library now uses it to check if the panel is going to arm stay or away.

> FYI, the library has been up and running for weeks now (develop branch) 👍

Great to hear!  Getting closer to the next release.

>>The exit state will even change from "Home" to "Away" if a zone is opened during the exit delay as the panel updates its state from stay arm to away arm.

> Not sure I follow, so this is relayed to the MQTT updates before the final state update to HomeKit? 

The HomeKit target state is updated each time the exit state changes - so arming on a physical keypad by access code will start with a target to stay arm, and if a non-stay zone is opened during the exit delay the target will change to away arm.

aboulfad, 2019-08-15:
> > Cool, except what is this Bypass light?
> 
> It's an indicator on non-LCD keypads indicating if there will be zones bypassed. The library now uses it to check if the panel is going to arm stay or away.

Alas, even after I pulled your latest 'develop' branch changes and made manual updates to my homebridge.ino script, the library still sends '1A' initially when I press the panel or RC 'Stay Arm' key.

Also, I notice that since my latest syn of your develop branch, the library is now less responsive and now exhibit some weird behaviour, but I will open a separate issue or a separate post.

neohusky, 2021-02-19:
Hi There

Thank you for you contribution with this great library.  Id like to create a standalone device/accessory using an esp32 dev board using the homekit Protocol without a 3rd party eg mqtt, home assistant or OpenHAB.

Is there any chance you'd consider adding an example platform.io (or otherwise) example using something like the ESP32 Homekit SDK?

taligentx, 2022-03-08:
Hi @neohusky - I've added an esp32 example sketch to the develop sketch that uses the [HomeSpan library](https://github.com/HomeSpan/HomeSpan), this enables a standalone HomeKit device.

It looks like an excellent library and setup in a way that makes it easy to configure the HomeKit accessories for partitions, zones, PGMs, smoke sensors, etc.  It seems to be working well in testing but feedback is always appreciated.  I'll close out this issue and feel free to open new issues for the HomeSpan sketch.

For esp8266, I've [ported the library to RTOS](https://github.com/taligentx/dscKeybusInterface-RTOS), but it's not as easy to configure as the HomeSpan library.  I think the faster processor of the esp32 also makes it much more suited to this purpose.

closed 2022-03-08 by taligentx