taligentx/dscKeybusInterface #96: System Unavailable

aboulfad, 2019-06-17:
Hi,
when I send *2 using the virtual keyboard, the system displays `System Unavailable`, and after 10-15sec it comes back. During this time, the keypad is non responsive, almost like the PC1832 rebooted ? It's no biggie, i just won't use the virtual keypad or that command !

I am using the IP version of keybusreader, and can't for sure rule out if something else is upsetting the panel. I am able to repeat this everytime I send `*2`.

Here, I enter `*2` on the keypad, and all is OK.
``` 9310.15: 11100110 0 00011000 00000001 00000000 00000000 00000000 00000000 00000000 11111111 [0xE6] Partition 1 | Status lights flashing: none | Zones 33-64 flashing: none
 9310.16: 11111111 1 00101000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] *
 9310.23: 00001010 0 10000000 10011110 00000000 00000000 00000000 00000000 00000000 00101000 [0x0A] Backlight - Enter * function code | Zone lights: none
 9310.32: 11100110 0 00100000 10000000 10011110 00000000 00000000 00000000 00000000 10000000 10100100 [0xE6] Status lights: Backlight - Enter * function code | Zone lights: none
 9310.40: 00000101 0 10000000 10011110 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Backlight - Enter * function code | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
 9310.41: 11111111 1 ........ ........ 11111111 11111111 11111111 11111111 ........ ........ [Keypad] Partition 1 | Key: [Digit]
 9310.48: 00001010 0 10000000 10100001 00000000 00000000 00000000 00000000 00000000 00101011 [0x0A] Backlight - *2: Trouble menu | Zone lights: none
 9310.54: 01011101 0 00010000 00000000 00000000 00000000 00000000 01101101 [0x5D] Partition 1 | Status lights flashing: Trouble | Zones 1-32 flashing: none
 9310.62: 11100110 0 00011000 00000001 00010000 00000000 00000000 00000000 00000000 00001111 [0xE6] Partition 1 | Status lights flashing: Trouble | Zones 33-64 flashing: none
 9310.79: 11100110 0 00100000 10000000 10100001 00000000 00000000 00000000 00000000 10000000 10100111 [0xE6] Status lights: Backlight - *2: Trouble menu | Zone lights: none
 9310.86: 00000101 0 10000000 10100001 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Backlight - *2: Trouble menu | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
 9313.61: 01011101 0 00000000 00000000 00000000 00000000 00000000 01011101 [0x5D] Partition 1 | Status lights flashing: none | Zones 1-32 flashing: none
 9313.62: 11111111 1 00101101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] # 
```

Here, I send `*2` using the virtual keypad:
```
9313.69: 11100110 0 00011000 00000001 00000000 00000000 00000000 00000000 00000000 11111111 [0xE6] Partition 1 | Status lights flashing: none | Zones 33-64 flashing: none
 9313.77: 00000101 0 10000001 00000001 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Ready Backlight - Partition ready | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
 9321.33: 01011101 0 00010000 00000000 00000000 00000000 00000000 01101101 [0x5D] Partition 1 | Status lights flashing: Trouble | Zones 1-32 flashing: none
 9321.34: 11111111 1 ........ ........ 11111111 11111111 11111111 11111111 ........ ........ [Keypad] [Digit]
 9321.41: 11100110 0 00011000 00000001 00010000 00000000 00000000 00000000 00000000 00001111 [0xE6] Partition 1 | Status lights flashing: Trouble | Zones 33-64 flashing: none
 9321.65: 00000101 0 10000000 10100001 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Backlight - *2: Trouble menu | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
 9329.55: 10100101 0 00011001 00011010 00110010 00100000 00000000 00000000 00101010 [0xA5] 2019.06.17 18:08 | Timestamp
 9348.73: 11100110 0 00101100 00000001 00000000 00000000 00000000 00000000 00010011 [0xE6] Partition 1 | Enabled zones 33-64: none
 9348.74: 11111111 1 11111111 11111111 11111110 11111111 11111111 11111111 11111111 11111111 [Keypad] Unknown Keybus notification
 9348.91: 11100110 0 00011010 01000000 00000000 00000000 00000001 00000001 00000000 00000000 01000010 [0xE6] 0x1A: Unrecognized data
 9348.94: 01110101 0 00000000 01110101 [0x75] Partition 1 | Beep pattern: off
 9349.55: 01011101 0 00000000 00000000 00000000 00000000 00000000 01011101 [0x5D] Partition 1 | Status lights flashing: none | Zones 1-32 flashing: none
 9349.63: 11100110 0 00011000 00000001 00000000 00000000 00000000 00000000 00000000 11111111 [0xE6] Partition 1 | Status lights flashing: none | Zones 33-64 flashing: none
 9349.71: 00000101 0 10000001 00000001 00000000 11000111 00000000 11000111 00000000 11000111 [0x05] Partition 1: Ready Backlight - Partition ready | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
```

taligentx, 2019-06-19:
This seems like normal panel behavior.  Are you navigating the trouble menu after pressing `*2` in virtual keypad?  If not, the panel will automatically leave the trouble menu after a delay.

Other keypads recognize that the panel is in a programming mode from some other keypad and will disable themselves until the panel is ready.

aboulfad, 2019-06-19:
> This seems like normal panel behavior. Are you navigating the trouble menu after pressing `*2` in virtual keypad? If not, the panel will automatically leave the trouble menu after a delay.

No, I don't. 
> Other keypads recognize that the panel is in a programming mode from some other keypad and will disable themselves until the panel is ready.

Ok gotcha!

closed 2019-06-19 by aboulfad