taligentx/dscKeybusInterface #98: esp8266 pushbullet not connect

marioveyna, 2019-06-21:
hi there, im newbie on this and i make my esp8266 for a pc1555 dsc board, everything fine, on monitor gets connect to wifi, and keybus, but not pushbullet, i got the ""Initialization push notification failed to send." i put the access token, i think is the only thing, hope anyone can help me with a working example of the code. thanks

taligentx, 2019-06-23:
Hi @marioveyna - which version of the Pushbullet sketch are you using?

If you are using esp8266 Arduino core 2.5.0 or above, try using the Pushbullet sketch from the `develop` branch.  The esp8266 core switched SSL libraries and there is a change in the sketch to account for this.

marioveyna, 2019-06-25:
thanks for answer, i am new on this and i don't know how to do pushbullet sketch from develop branch that you say, can you please help me, thanks in advance

marioveyna, 2019-06-26:
branch develop works for me i got the "security system initializing" in my phone, but is the only msg i got... any idea what its happened? thanks

taligentx, 2019-06-28:
> branch develop works for me i got the "security system initializing" in my phone, but is the only msg i got... any idea what its happened? thanks

Good to hear that the Pushbullet part is working.  Try flashing the `KeybusReader` sketch - what are you seeing as the output?  You can let it run for 5 minutes or so and post the output here.

marioveyna, 2019-07-02:
hi again, i ran the keybusreader and this is the output i got:

```
DSC Keybus Interface is online.
    3.00:  [0x00] Unrecognized data[CRC OK]
```
when i compile sketch, i got an error saying that D2 and D8 variables are not declared, i declared it as a byte, maybe is this an error?

is there somewhere a forum with people that make this work and interchange tips? i know you are the developer and maybe you are busy

taligentx, 2019-07-02:
> when i compile sketch, i got an error saying that D2 and D8 variables are not declared, i declared it as a byte, maybe is this an error?

You should be able to use the KeybusReader example sketch without making any changes if you wire to GPIO 5 and 4 (D1 and D2 if you have a NodeMCU/Wemos board).  As you're not seeing any data from the DSC panel, there's something going on with this wiring or the pins you're using.

> is there somewhere a forum with people that make this work and interchange tips? i

It's a good idea to have a support thread or forum, I'm open to suggestions on where would be a easily accessible place for something like this (Arduino forums, Reddit, etc).  I do have a thread on the Arduino forums:

https://forum.arduino.cc/index.php?topic=557475.0

marioveyna, 2019-07-03:
thanks i will check the wiring and forum, but what do you think about error when compiling about D1 and the others not declared? what is the data type of this variables? thanks

taligentx, 2019-07-06:
There's some problem with your esp8266 setup if you are getting errors about the D1/D2/D8 pins.  I recommended trying to flash some of the esp8266 specific examples to your board (Files > Examples > ESP8266 > Blink, etc).  Also double check the esp8266 Arduino installation steps to make sure you have the correct board selected for whichever development board you are using:

https://github.com/esp8266/Arduino

taligentx, 2019-07-16:
@marioveyna Are you still seeing this issue?

marioveyna, 2019-07-16:
sorry friend, have some personal problems.... let me test that you mentioned and on weekend tell if work, thanks for your interest in my problem with this....

taligentx, 2019-07-26:
Feel free to re-open if you're still seeing this issue.

closed 2019-07-26 by taligentx