taligentx/dscKeybusInterface #99: How to determine the actual number of partitions?

jimtng, 2019-06-22:
Hi, is it possible to find out how many partitions are actually in use? For example, my home alarm system only has/uses 1 partition. If this can be determined from code, I can skip broadcasting other partition status.

taligentx, 2019-06-23:
This isn't currently tracked in the library but in theory you could check the status commands (0x05, 0x27, etc) to see which partitions are disabled (status = 0xC7).

In general, it is possible to obtain all panel and module data in the sketch - the library exposes the panel data as `panelData[]`, this is populated with the last received panel command each time `dsc.loop()` returns true.

At that point it's a matter of processing the data similar to what happens in `dscKeybusProcessData.cpp` to see which partitions are disabled.

For now selecting which data to use resources to track is arbitrary - I've thought about having a set of extended status commands that are only available to esp8266/esp32, though for now it's made more sense to setup new status tracking as actual use cases come up.

jimtng, 2019-06-24:
> This isn't currently tracked in the library but in theory you could check the status commands (0x05, 0x27, etc) to see which partitions are disabled (status = 0xC7).

I'm unfamiliar with the inner workings of the panel protocols. Could you elaborate please? What are the "status commands" and how can I access them?

taligentx, 2019-06-28:
> I'm unfamiliar with the inner workings of the panel protocols. Could you elaborate please? What are the "status commands" and how can I access them?

`dscKeybusPrintData.cpp` demonstrates this - it first checks `panelData[0]` (the first byte of the panel data) to see which command it is, and then hands off to a separate function for each command to decode and print.  Several of the panel commands share the same value for messages so the messages are all in `printPanelMessages()` to save program space (instead of duplicating all of the messages for each status command) and the functions for each status command send which byte contains the message to decode.

For example: To see if partition 2 is disabled, in the sketch you could check on each run of `dsc.loop()` to see `if panelData[0] == 0x05` (status command for partitions 1-4) and `if panelData[7] == 0xC7`, then partition 2 is disabled.  Partition 3 status is in `panelData[7]`, partition 4 is `panelData[9]` as listed in `printPanel_0x05`.

If I get around to writing an example of unlocking a panel it would demonstrate this technique in the sketch because it would need some of the panel data that I don't think the library needs to track in general.

taligentx, 2019-06-28:
Actually, I forgot that I had already broken out the status per partition so it should be straightforward:

```
<dscKeybusInterface.h>
// These contain the current status message and LED state for each partition.  See dscPrintPanelLights() and dscPrintPanelMessages()
// in dscKeybusPrintData.cpp to see how this data translates to the status message and LED status.
byte status[dscPartitions];
byte lights[dscPartitions];
```

Loop through the partitions but as an example for partition 2 you should be able to check `if (dsc.status[1] == 0xC7)` then the partition is disabled.

jimtng, 2019-06-29:
In my loop, I got these values:
dsc.status[0]: 1
dsc.status[1]: 0
dsc.status[2]: 0
dsc.status[3]: 0
dsc.status[4]: 0
dsc.status[5]: 0
dsc.status[6]: 0
dsc.status[7]: 0

taligentx, 2019-06-29:
Which panel model is this?  The PC1864 will use `0xC7` for disabled partitions 1-4, and `0` for partitions 4-8 if none of 4-8 are enabled.  I haven't checked but they would probably report `0xC7` if one of that group was enabled.

PC1864, partition 1 enabled, 2-8 disabled:
```
Status: 1
Status: C7
Status: C7
Status: C7
Status: 0
Status: 0
Status: 0
Status: 0
```

When the sketch first starts up this will not be valid until one of the status commands is processed.

jimtng, 2019-06-29:
It's PC1565. At the time I reported, the sketch has been running in the main loop() for a while.

jimtng, 2019-06-29:
> When the sketch first starts up this will not be valid until one of the status commands is processed.

I'd like to know the partitions inside setup(). What's the best way to wait until the status commands are processed? Sorry, I understand that this may sound like a stupid question to you. I'm still working on better understanding how the library works.

jimtng, 2019-06-29:
I tried the following inside setup() but it didn't seem to work:
```c++
  dsc.begin();
  delay(1000);
  while(!dsc.keybusConnected && !dsc.loop()) delay(100); 
```

I'm using Wemos D1 Mini Pro, and it's not connected to the keybus (or anything other than the usb to my laptop to power it), but the above code simply proceeded past, so afterwards, I still don't have a valid dsc.status[] data.

closed 2019-06-30 by jimtng