taligentx/dscKeybusInterface #100: setTime not working, triggered an alarm

jimtng, 2019-06-23:
setTime doesn't seem to be working correctly. After calling it, the panel time didn't change. Example:

dsc.setTime(2019, 6, 1, 1, 1, "accesscode");

In fact, I tried this multiple times, and suddenly my alarm went off. Probably not the most popular house in the neighbourhood now.

When the alarm was blaring, I ran to the physical keypad and entered my access code once. It didn't stop the alarm. In panic I pressed Reset, and entered the access code again and again.... lol. Eventually the alarm noise stopped and the "Ready" LED was on.

I observed two problems:

I noticed that the partition 1's alarm flag is still set to 1. It doesn't seem to have gotten reset back to 0 (like the openZone for example) after the alarm has been disarmed. Is this supposed to be reset to 0 by my sketch after reading it?

Why did my alarm go off? Was it something to do with setTime not working for me?

Have you tested setTime() on your system, e.g. repeatedly / multiple times etc?

jimtng, 2019-06-23:
More info, my alarm is PC1565
Arming with dsc.write("w") and disarming with dsc.write("NNNN") (the access code) work fine. 

It seems however, that the following don't work:
- in the ready state, dsc.write("accescode") which I presume should set the alarm into away mode
- in the ready state, dsc.write("*105#") to bypass zone 05

Observed behaviour when trying to bypass zone 05 step by step:
- dsc.write("*1") - physical keypad went into program mode 
- dsc.write("05") - physical keypad returned to ready state - this is not supposed to happen

I suspect some sort of incompatibility with my old alarm model?

jimtng, 2019-06-23:
@taligentx, in troubleshooting this issue, I **came across a bug** in dscKeybusProcessData.cpp 

```
void dscKeybusInterface::setTime(unsigned int year, byte month, byte day, byte hour, byte minute, const char* accessCode) {
  if (!ready[0]) return;  // Skips if partition 1 is not ready
  if (hour > 23 || minute > 59 || month > 12 || day > 31 || year > 2099 || (year > 99 && year < 1900)) return;  // Skips if input date/time is invalid
  static char timeEntry[21]; // THIS NEEDS TO BE STATIC
```

Essentially, what is being passed on to dsc.write(char *) needs to remain accessible since dsc.write() continues to access it even after it has returned to the caller.

Perhaps a more elegant solution is for dsc.write(char *) to copy the string into an internal buffer managed by the dsc class itself. This would remove the need to make the "char timeEntry" variable static.

I was also affected by this bug in other parts of my code that use dsc.write.

taligentx, 2019-06-23:
@jimtng That's not a fun situation, I'll see if I can reproduce the same issue - my original testing didn't show this kind of behavior so I'll need to try something else.  Originally I used a physical button switch to trigger `setTime()` and ran it consecutively while a separate interface ran `KeybusReader` but didn't see the same problem.

Great catch for `timeEntry`, I've pushed a commit to set this to static as it should be.  The only issue with creating a separate buffer is the extra memory usage and not knowing in advance how large a buffer will be necessary.  The advantage of having it static in `setTime()` is that the memory will only be allocated if a sketch actually uses `setTime()`.  On the todo list to take a closer look at this.

jimtng, 2019-06-23:
@taligentx thanks for your prompt response. The issue occurred because dsc::write and dsc::writeKeys (and ultimately the interrupt processing) are simply working with a pointer to the string given by the caller. The memory in the caller can become invalidated, for example a local / stack variable that goes out of scope, or multiple successive calls to dsc::write(char *)  in a loop before the interrupt finished writing all the data from the previous string.

Perhaps it has just been lucky (because esp8266 is a single threaded/task system) that the invalidated memory still contained the same data hence this bug hasn't been detected previously.

You could set an arbitrary buffer size managed like a fifo linked list, and fail the call to dsc::write if the buffer is full. 

Alternatively, and perhaps more simply, block the call to dsc::write and return when the interrupt has finished sending the given string.

If you want me to write a code that reliably demonstrates the issue, let me know. Meanwhile I will need to figure out how to trick my alarm that the sirens are still connected after I've unplugged them :D I think some sort of resistors in place of the sirens would do the trick?

jimtng, 2019-06-23:
@taligentx could you tell me who would clear the dsc.alarm[partition] flag once the alarm status is reset by entering the access code? In my case, it remained set after the above incident until I've power cycled my alarm (including unplugging its battery).

jimtng, 2019-06-26:
I see that dsc.writeReady will be deprecated. There needs to be a way for the library user to know that writeKeys has finished writing all the data, before calling dsc.write() again. Otherwise, successive calls to dsc.write(const char*) will overwrite the writeKeysArray pointer.

jimtng, 2019-06-26:
I see that dsc.writeReady will be deprecated. There still needs to be a way for the library user to know that writeKeys has finished writing all the data, before calling dsc.write() again. 

Imagine calling dsc.setTime() in such manner:
for (int i = 1; i< 100; i++) dsc.setTime(2019, i % 12 + 1, i % 31 + 1, 0, 0, "xxxx");

The timeEntry static variable will have gotten overwritten before dsc.writeKeys could finish writing the previous complete set of string.

I have a similar situation as above, i.e. my implementation allows for users to write to the keypad via mqtt. So my mqtt handler receives the data, writes it to a static char[] variable, and pass it on to dsc.write(). If someone were to send mqtt messages in rapid succession, it would in turn call dsc.write() repeatedly similar to the above loop, thus overwriting my static char[] variable before writeKeys can finish.

This can be avoided by using the writeReady flag like so (in pseudocode, greatly simplified, error handling / checking omitted):

```
static char staticStringBuffer[50];
while (!dsc.writeReady) dsc.loop();
strcpy(staticStringBuffer, receivedmqttmessage); // this needs to be done after writeReady = true
dsc.write(staticStringBuffer);

```
Furthermore, it should be implemented as a function to avoid people modifying its value.

taligentx, 2019-06-28:
> could you tell me who would clear the dsc.alarm[partition] flag once the alarm status is reset by entering the access code?

If the alarm was triggered and kept going until the siren stopped, then the `dsc.alarm[]` status is cleared if the `0xA5` command reported that the panel was disarmed (either by access code, wireless key, DLS, etc).  The `0xA5` command has a CRC so it's a bit more trustworthy - the panel also sends a constant stream of `0x05` status commands (literally nonstop), but these do not have a CRC so I originally didn't trust them to flag the alarm as being disarmed.

For now I've added the `0x05` status commands as valid for flagging the alarm off, but I may revert this because I really would prefer to keep the critical status changes to CRC checked commands.  

When your alarm was tripped and the siren finally went silent, did the alarm go back to ready status or was it still armed?  You can use a 10k resistor instead of the siren to give the panel a dummy (and silent!) siren.  If possible I'd really like to see the output of `KeybusReader` during this event if you are able to replicate this scenario.

taligentx, 2019-06-28:
> The timeEntry static variable will have gotten overwritten before dsc.writeKeys could finish writing the previous complete set of string.

This issue may be limited to `setWrite()`, as it currently doesn't block if a write in progress and probably should, I'll take a look.

Normally in the case of a single character, `dsc.write()` as called from the sketch will not block and will return immediately because the character is converted in the library `setWriteKey()` to the panel format in a separate char that is handed off to the interrupts to write at the next opportunity.  If the sketch immediately sends another `dsc.write()`, it will block and run `dsc.loop()` until the previous write is complete.

In the case of multiple characters in an array, `dsc.write()` currently does not block, the sketch needs to keep the char array intact - in which case, yes, the sketch should be able to check if the write is complete.  `writeReady` is currently only for the single character, `bool writeKeysPending` is actually how the library checks internally if there is a char array write in progress.  For a value that the sketch can check, the library will need to be updated so that `writeReady` handles both cases.  Added to the todo list.

 Design-wise, I'll look into this to see if it's preferable to block until the char array is written or if the sketch should handle keeping the char array around.  Writing in the library at the moment is based on the sketch behaving as a human would interact with the panel,  it'll need added robustness to handle being hammered in a `for()` loop.  I'm leaning towards blocking until the write is complete but input is always welcome.

jimtng, 2019-06-29:
I think making it block until the write is complete is the simplest most straight forward approach, so I'm voting for that. If you like, you can add an additional parameter with a default value, e.g `dsc::write(const char* str, bool block=true);`

jimtng, 2019-06-30:
setTime now works with the change to static buffer, so I'll close this issue. However, I'd still like to see a blocking write().

closed 2019-06-30 by jimtng

taligentx, 2019-07-22:
> If you like, you can add an additional parameter with a default value, e.g dsc::write(const char* str, bool block=true);

This is the direction I was leaning in as well, the above commit implements optional blocking and also splits off `writeReady`so it's purely to provide status and not used by the library itself.  Thanks for the feedback!