taligentx/dscKeybusInterface #101: Installed Code unlocking

aboulfad, 2019-06-24:
Ok someone had to request this! I’ve been reading in the reference manual about keypad lockout and I can’t figure out if there’s a way to find out if it is enabled, obviously without the installer code.

So I am curious about the suggestion you make about the workaround:

> ...possibly automatically power-cycling the panel with a relay to skip the lockout time.

That would require knowing that a keypad lockout happened, and an event will be registered, but why do you think that it isn’t saved even if the unit power cycles?

taligentx, 2019-06-28:
>I can’t figure out if there’s a way to find out if it is enabled, obviously without the installer code.

If I recall correctly (though it's been a while since I looked at this), the keypad lockout status is only sent after the number of invalid access codes has been reached.  At that point, the library could determine that it is locked out and either: 
* Wait for the lockout to expire by testing the next access code at a periodic interval until it the panel does not sent a keypad locked out status
* Power cycle the panel

> but why do you think that it isn’t saved even if the unit power cycles?

I don't recall the panel saving the lockout status, but I haven't tested this recently to confirm.

aboulfad, 2019-06-28:
Sadly, my system is in-use and connected to monitoring station, so I'd be wary of getting it locked and un-usable for the maximum of 255mns... I wanted to change some settings and realized my installer changed his code again ! (i got it once by glancing over his shoulder ;-) )

taligentx, 2019-07-16:
I ran across a panel with an unknown installer code so it was time to look into this.  The keypad lockout can be bypassed if the power is cycled while the panel is jumpered to reset to factory defaults, otherwise a power cycle does not affect the PC1864, older panels may be different.

I've added an example sketch to unlock the installer code, it can check a PC1864 for all possible 4-digit codes in ~8.5 hours and checks the well known codes first.  I'll leave this issue open for folks to respond if they'd like to add their discovered installer codes to help speed up future searches.

aboulfad, 2019-07-16:
You are too good! Thanks for the follow-up, so sadly my unfounded suspicions were correct about power cycle.

So by resetting to factory defaults, I guess all programming is lost and given my panel is connected to a monitoring company, and contains much programming I am not aware of, that is no go for me :(

But for other folks, is this [procedure](https://www.alarmsystemstore.com/pages/dsc-powerseries-1616-1832-1864-how-do-i-default-panel-if-i-don-t-have-the-installer-code) correct to reset to factory defaults?

So the unlocks steps are:
- reset panel to factory defaults
- run unlock script after panel has been reset and keep jumper cable?

taligentx, 2019-07-16:
> So by resetting to factory defaults, I guess all programming is lost and given my panel is connected to a monitoring company, and contains much programming I am not aware of, that is no go for me :(

There's no need to factory default or use a relay, that's only to speed things up if your panel happens to have keypad lockout enabled - is it enabled on your panel?  The lockout is typically 5-15min but you could set aside a 4.5h window to manually test to see if it is enabled (in case the lockout period is maxed out).

The minimum setup to unlock:
- DSC panel - only power and the microcontroller need to be connected
- Arduino or esp8266 with virtual keypad wiring and power

That's pretty much it, the serial connection isn't even needed because the microcontroller also uses its onboard LED to blink out the code when complete.

For security systems in active use, it'd be useful to be able to pause the code search so it only runs while the security system is not being used.  I'll add it to the todo list but for now I'll hold off pending feedback.

aboulfad, 2019-07-16:
> There's no need to factory default or use a relay, that's only to speed things up if your panel happens to have keypad lockout enabled - is it enabled on your panel? 

That is the million $ question and that I originally asked when I opened this "issue", if there is a way to find out without brute force/installer code.

I could try manually, but not sure how many tries before it times out! Last time I tried about 5 or 6.

taligentx, 2019-07-16:
> I tried 30 in a row and nothing locked.

Seems unlikely that keypad lockout is enabled.  For reference, an ADT-branded PC1864 default configuration permits 6 code tries before locking out for 160 seconds.  Which is interesting because the user-facing configuration only permits setting a lockout time in whole minutes.

aboulfad, 2019-07-16:
ok maybe i'll bite the bullet and try. I will IP'ify your script and try! my panel is in my basement and can't dedicate a laptop for hours connected to it :-D

aboulfad, 2019-07-17:
OK, `UnlockerIP` script running from the comfort of my office/iMac 😄 (it makes me laugh and think about all these movies where they hack stuck remotely)

I gotta say, I would not have been able to write Unlocker script. It requires good knowledge of your library, but now with the example and excellent description it serves as good foundation. So big thanks for your effort.

I was surprised that it takes 3s per code try! I chose to start at 1000 instead of 0. We shall see if I pass the 255 mark.

aboulfad, 2019-07-17:
Script found the code after roughly one hour :-) Super easy to modify to fit one needs, and the IP/OTA version makes it all worthwhile for those whose panels are in "inhabitable" places.

@taligentx , I can PR an IP/OTA version should you want. I find IP without OTA kinda defeats the purpose. It was really a breeze to upload over IP the new Unlocker script, then revert back to the dsc sketch without ever visiting my basement !!!

taligentx, 2019-07-18:
> Script found the code after roughly one hour :-) 

Hooray!  I suspect quite a few panels end up in landfills or recycling due to unknown installer codes, hopefully this sketch reduces the waste by a few panels.

> I was surprised that it takes 3s per code try! 

The panel is rate limited, otherwise the Keybus protocol would support testing a code every ~320ms.

> I chose to start at 1000 instead of 0.

I've setup the esp8266 version of the sketch to use a list of codes [sorted by general frequency](https://www.datagenetics.com/blog/september32012/).  It does eat up resources but could reduce the code search time.

> I can PR an IP/OTA version should you want. I find IP without OTA kinda defeats the purpose.

Sure, though take a look at the last commit, there are quite a few changes.  As far as OTA, it's not in the examples for a couple of reasons:
* It doesn't add value to the examples as far as helping demonstrate a particular library feature or integration (it's an add-on component based on personal preference).
* It's easy to add if a user would like to use it.
* Personally I'm not a fan of having OTA code on a device that can disrupt the Keybus if it is compromised in some way.  For this type of application, OTA examples should probably use [signed updates](https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#advanced-security-signed-updates).

aboulfad, 2019-07-18:
That’s some huge known test code list! I can’t debate what you wrote above, it’s all sound and in any case it’s your library :-) I haven’t tried signed OTA, I am curious how well it works. Thanks again. I can close this issue or leave it open, let me know.

taligentx, 2019-07-18:
> That’s some huge known test code list!

Yep, I wasn't sure how well a 10k element array would work but it doesn't seem to be an issue.  

I'd like to leave the issue open for additional feedback from all users to see how this is working out.

BriceDet, 2019-08-17:
May be it's a stupide question : is the installer code can be less than 4 digits ?
I saw in the list some test line 118, 119, 120...
`13,69,99,103,114,209,212,258,301,319,331,405,408,419,501,509,
513,517,521,523,622,721,813,818,913,914,926,929,1144,1311,1590,1911,1937,2469,3579,4711,4860,5757,5888,116,205,208,
217,228,317,330,414,415,417,486,507,524,610,612,613,624,666,807,812,820,915,919,1199,1345,1452,1800,1914,2110,
2311,.....`

Are this Installer code possible ?
Thanks
Brice

szokesandor, 2019-10-31:
Hello, I was successfully used your sketch on Arduino UNO to guess my installer code. It was working well without relay. My system is PC5020. 
Also the keybusreader seems to be working. I have only problem, when sending keypresses. It is not always accepted. Now I can access to my system completely. Thanks for the whole library.

jgrosso2171, 2019-10-31:
Hi, how did you do to see the installer code?  As for the failures to send commands, it can be a problem in the intensity of the interrupt signal, check the resistance value placed on the transistor base and the distance between the arduino interface and the data bus terminal of the  license plate.

taligentx, 2020-10-01:
@szokesandor - late response, thanks for the feedback and good to hear you were able to retrieve the installer code on the PC5020.  For issues with writes, check wiring - connections should be soldered, etc.

Closing, `Unlocked` sketch will be in the next release, thanks all for testing!

closed 2020-10-01 by taligentx