taligentx/dscKeybusInterface #103: how to see triggered zones using *3 command?

connected201, 2019-06-27:
hi, i have DSC PC 585H, i want to see earlier triggered zones using *3 command, dsc panel showing all triggered zones, but in blynk i cannot use this option, what problem is?
i recorded short video:
[video_2019-06-27_23-13-02.zip](https://github.com/taligentx/dscKeybusInterface/files/3336456/video_2019-06-27_23-13-02.zip)
or link: https://mega.nz/#!zLgznQTQ!Xzg2BpokUwbx3Bfl6CICvpI61P3mmWkPqjwGZnMO6D4

taligentx, 2019-06-28:
Hi @connected201 - the Blynk example doesn't display this information as it is currently only setup to input keys and display the panel status, not perform any programming or * functions.

If you're interested in setting it up for programming, you could check the `panelData[]` for the `0x5D` and `0x63` commands, this would give you which zone lights are flashing, then you could pass those zones to Blynk.  I'll tag this issue if you or anyone else wishes to develop these types of features.

connected201, 2019-06-28:
Hi @taligentx , i want to make this option to work, but i does not have enough knowledge for implementation :(, so i can donate to developer team to implement it, it's possible? 
best regards.

zlazar, 2020-01-21:
Dear connected201, please, can you help me to configure the nodemcu connect to a local blynk server?
Thank you

taligentx, 2020-12-29:
I've updated the `Virtualkeypad-Blynk` sketch in the `develop` branch to handle displaying the alarm memory zones in the *3 alarm memory menu - this also demonstrates how sketches can be updated to handle any panel data that the library doesn't natively handle.  I'll leave this issue open for feedback in case there are issues.

taligentx, 2021-01-22:
The `VirtualKeypad-Blynk` and `VirtualKeypad-Web` sketches in newly release version 2.0 both display zone lights for *3 alarm memory, as well as in the other programming modes to match the display of the physical LED keypads.  If there are issues, feel free to re-open this issue.

closed 2021-01-22 by taligentx