taligentx/dscKeybusInterface #105: How to determine the actual used zones?

jimtng, 2019-06-29:
Similar to my other question about partitions, how can I determine which zones are actually in use? My alarm only uses 5 or maybe 6 zones. It would be good not to have to send the status of all 64 zones especially after a dsc.resetStatus().

taligentx, 2019-07-06:
I'm not sure this information is directly available from the panel.  As far as I've seen, the panel reports enabled zones per partition but does not provide information on whether the zones are set to NULL (00) or are setup as instant, delay, etc.

`[0xB1] Enabled zones 1-32 | Partition 1: 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32`

The `0xB1` command shows which zones are assigned to each partition, but in this case I only have zones 1 and 2 configured while the rest are NULL.  However, the panel still reports that the zones are available to configure.

Theoretically you could implement checking on the sketch side by monitoring for open zones and building up a list of zones that have been changed to form a list of zones that are in active use, though this seems like it'd be easy to miss zones.

You can edit `dscKeybusInterface.h` to change dscZones - this is counted in groups of 8 zones so you could limit to 8, 16, 24, etc zones at a time.  Because of the way that Arduino handles header files, this can't be changed in the sketch.

taligentx, 2019-07-16:
Feel free to re-open if there is more to look into this.

closed 2019-07-16 by taligentx