taligentx/dscKeybusInterface #106: Announcement - dscKeybusInterface now ported to RTOS, including native HomeKit support

taligentx, 2019-07-06:
Hi folks,

As discussed in #27 and #92, I've been taking a look at [esp-open-rtos](https://github.com/SuperHouse/esp-open-rtos) to see if it would be feasible to port the library.  This would enable native HomeKit integration (without using other intermediary devices with MQTT, Homebridge, HAP-NodeJS, etc) using [esp-homekit](https://github.com/maximkulkin/esp-homekit).

Quite a learning curve - the library is up and running and is ready for testing and feedback:

https://github.com/taligentx/dscKeybusInterface-RTOS

I've included ports of the `KeybusReader` and `Status` sketches, as well as an example HomeKit integration that demonstrates using the esp8266 on its own to provide the security system accessory, as well as a couple of zones and fire sensor.  For now, adding additional zones and partitions is a manual process, as the library gets tested I'll look at simplifying new additions.

As compared to running within the Arduino framework, the initial setup may look daunting, but once installed it's quite straightforward to edit code and flash the esp8266.

Please post issues, feedback, and PRs (yay!) in the RTOS library:
https://github.com/taligentx/dscKeybusInterface-RTOS/issues

Thanks,
Nikhil

aboulfad, 2019-07-06:
Hi @taligentx , congrats on the port, that is awesome. I’ll give it a try when I am done with my Irrigation project. Curious how well esp HomeKit works.

JohnMaree, 2019-07-06:
Amazing!!!! Congratulations! I know I bailed on this a while ago and I regret that — turns out studying and anything else doesn’t go well together! Well done! I will test it as soon as I can!!!

aboulfad, 2019-07-06:
@taligentx , Since you have the sketch running, can you get the free heap,... I am curious if esp-open-RTOS is similar in heap/SRAM/... utilization as nonos. (ESP.getFreeHeap(), or the equivalent on open RTOS)

taligentx, 2019-07-07:
> Since you have the sketch running, can you get the free heap,... I am curious if esp-open-RTOS is similar in heap/SRAM/... utilization as nonos. (ESP.getFreeHeap(), or the equivalent on open RTOS)

With one iOS client connected, xPortGetFreeHeapSize() reports: 32188

closed 2020-11-02 by taligentx